openLRSng/openLRSng #86: Compile for Broversity fails

JohnieBraaf, 2015-03-05:
Hi trying to compile with the Arduino IDE I got the following message:

/serialPPM.h: In function 'void sendSBUSFrame(uint8_t, uint8_t)':
serialPPM.h:75: error: call of overloaded 'write(int)' is ambiguous
note: candidates are: virtual size_t HardwareSerial::write(uint8_t)
note:                 size_t Print::write(const char*)

I have resolved it with a type cast: 

Serial.write(<b>(uint8_t)</b>SBUS_TAIL);

kh4, 2015-03-08:
I will add the cast.  Notably compilation with arduino is not verified anymore as project is built with make. 

Arduino support is deprecated and will be removed with 4.0 version (all arduino arduino references removed so that build obly needs avr-gcc.

kh4, 2015-03-14:
Fixed in 42600a2

closed 2015-03-14 by kh4