openLRSng/openLRSng #87: GPIO swap

stormjohn, 2015-03-09:
Forgive me I made this comment on a closed issue (#70). But if you will, I will repeat myself! 
How common is the GPIO swap an issue? I have a Flytron M3 (1W) its struggling to make 100mW on my power meter with a 50 Ohm "dummy load" no matter what I try. I compiled with the latest 3.8 firmware.
Any help would be greatly appreciated.
Thank you.
Lyn.

kh4, 2015-03-09:
With correct hw type (2) selected the GPIO:s are proper the setting is only used for certain hardware types and is handled internally.

What powermeter you are using? Notably you should measure the 'peak' power not average since the transmission is 'pulsed'. For example with default settings the average power from Flytron M3 would be around 150mW. You can use the 'openLRSng_tester' to transmit continously.

stormjohn, 2015-03-10:
Thanks for the reply Kari.
 
I tested both with "Average" & "PEP" on my cheap SWR / PWR meter, ( my finances could not stretch to a Bird 43). Anyhow the Average is as I said 100mW and Peak is only slightly higher at 120mW! I don't think there is anything wrong with the meter as I tested my "Dragon Link" in the same manner and it's showing the expected 500mW output. 
I did noticed the spectrum analyzer was giving a peak RSSI output of only 81 at 432MHz. This did surprise me as I adjusted the frequency in the configurator for the UK legal 458MHz band.
What I have also found even with a Diamond SRH771 antenna on the Tx I cannot use it. The Rx gives me an intermittent "blue" LED changing almost straight away to a solid "red" LED at just a few Cm's away. If the Tx was giving even a small output I surely should have had a reasonable signal at a few Cm's. I did find the best response was to fit a small omni pole to the Rx and the "dummy load" (the one that came with the "Dragon Link" for range testing) to the Tx. I could get >50 meters with full servo control and solid "blue" LED and no lost "packets". If I reverse the antennas i.e. put the "dummy load" on the Rx and the small antenna on the Tx I again loose any control at few Cm's with all "packets" lost (solid "red" LED).  
Again any advice would be appreciated.

Lyn

fofi1, 2015-04-29:
Flytron's 1W transmitter uses only 3.3V supply for the RFM23BP. So the maximum Output Power is 350mW for this module. 

Only at 6V it would deliver about 900mW of output power. 

The RFM Modules only work great with Dipole antennas due to internal impedance matching to 50 ohm.

kh4, 2015-04-30:
You might have burned the RFM23 modules output stage, thus is quite common.

It can happen by no or bad antenna while it is transmitting or by static electricity charge to the antenna port. You can get RFM23BP modules for about 10 to 15USD.

stormjohn, 2015-06-29:
Hi Kari.

 

Sorry for the late response.

 

I would like to agree with you on this if it wasn’t for the fact that I have both the original Flytron and the HK orange “look alike” unit both rated at 1W. Both are giving me a similar output!

 

I have some experience using RF units and would like to think I have not powered them without the antenna fitted!

 

I now know that because of the on-board 3V3 supply I am at best only going to see 27dbm (Hope RF datasheet).

 

Both units are showing a similar reading on the power meter! That is nowhere near my “Dragon link” unit.

 

I understand if you are unable to devote time to this.

 

In the mean time I am completing a unit with a ATMega 328 and Hope RF RFM 23BP with the full 5V5 supply at the Vcc pin.

 

Kind regards

 

Lyn.

Timvrakas, 2015-06-30:
Just to let you know, you are not the only one having these problems... I
just finished a prototype run of USB dongle RFM23bp devices, and I cannot
get any range out of them! I can't tell if it's my inexperience or bad
luck, but they sure are finicky.

stormjohn, 2015-06-30:
Hi Tim.

 

Thanks for sharing.

 

I will let you know how I get on with the custom units.

 

Kind regards

 

Lyn

linuxslate, 2018-01-02:
Sorry;  I commented on the closed issue too.

I am building my own DIY hardware, and I am using BOARD_TYPE == 4, which has #define GPIO_SWAP.

I have GPIO0 connected to RX ON, and GPIO1 connected to TX ON. In other words, they are NOT Crossed. It is wired like I see everyone wiring RFM22B's.

Therefor I thought I did NOT need #define SWAP_GPIOS.

But then I looked at the code. In common.h, I see:

```
#ifdef SWAP_GPIOS
  spiWriteRegister(0x0b, 0x15);    // gpio0 RX State
  spiWriteRegister(0x0c, 0x12);    // gpio1 TX State
#else
  spiWriteRegister(0x0b, 0x12);    // gpio0 TX State
  spiWriteRegister(0x0c, 0x15);    // gpio1 RX State
```

The comments seem to indicate that if GPIO0 connected to RX ON, then I DO need SWAP_GPIO's.

Is the comment wrong or mis-leading?

Is 0x12 on or off? I have read the RFM23BP data sheet, and I still don't get it. Also, RX ON and TX ON are apparently active low.

Please help, as I have been messing with these both in software and hardware, and probably fried my module.  I have more on the way, but I do not want to install them without understanding this better.

--- Very confused.

linuxslate, 2018-01-02:
This is also unresolved in (Closed) issue 81.
https://github.com/openLRSng/openLRSng/issues/81

They are telling him to use BOARD_TYPE == 2, which does not have #define GPIO_SWAP, but both the schematic in that issue, and the photo show that the GPIO's are wired crossed (look at the feed throughs).

In the hopes of making this easy, I have even done some ascii art to try to make this clear.

```
If you have a RFM23BP, and the GPIO connections
are wired like this:

Looking at top (component side) of RFM23BP

 ______________  
|   _______    |
|  |       |   |
|  |   |   |   |
G  G   V   R   T
P  P   C   X   X
I  I   C
O  O       O   O
1  0       N   N

Then DO/DO NOT use #define SWAP_GPIOS

If it is wired like this:

 __________
|   _______)___
|  |       |   |
|  |   |   |   |
G  G   V   R   T  
P  P   C   X   X
I  I   C
O  O       O   O
1  0       N   N

Then DO NOT/DO use #define SWAP_GPIOS

```

If someone "in the know" could just edit it DO's/DO NOT's appropriately, it would really help.

This could even go at the beginning of hardware.h

csurf, 2018-01-02:
@linuxslate,
normal: GPIO0 ==>> TX pin
swapped: GPIO0 ==>> RX pin
so, add/remove the '#define SWAP_GPIOS' statement accordingly based on your connections.

the '0x12' & '0x15' values are binary/bitmap values meant to configure the GPIO registers in order to have them output the TX and RX states. These are fed into the TX_ANT and RX_ANT pins on the RFM module in order to put the antenna into either TX mode or RX mode.

See this datasheet for more info on register config:
https://www.silabs.com/documents/public/data-sheets/Si4430-31-32.pdf

csurf, 2018-01-05:
@mikeller  
let's close this back out once @linuxslate responds with an 'ACK' for the above explanation...

linuxslate, 2018-01-05:
My DIY transmitter seems to be working with it wired like this:

```
______________  
|   _______    |
|  |       |   |
|  |   |   |   |
G  G   V   R   T
P  P   C   X   X
I  I   C
O  O       O   O
1  0       N   N

```
... and #define SWAP_GPIOS  defined.

This seems to agree with csurf's explanation above.

So I am OK with closing this issue.  However; I think that the comments in the #ifdef SWAP_GPIOS section of common.h should be replaced with csurf's explanation:
```
normal: GPIO0 ==>> TX ON, GPIO1 ==> RX ON
swapped: GPIO0 ==>> RX ON, GPIO1 ==>TX ON
```
(Note that I changed the nomenclature to match the pin callouts in the RFM23BP datasheet.)

They should also be included as a comment at the top of hardware.h
The comment should probably include a line explaining that this is only for some boards with a RFM23BP. 
 
Just suggestions.  I'm just a just a dumb user that likes to solder stuff and fly radio control.

csurf, 2018-01-05:
@linuxslate 
ok, agreed, the comments will be added to hardware.h. I'll have to set it up on a separate 'hardware cleanup' branch where I'll start breaking up the hardware profile definitions into separate files.

BTW, @linuxslate, are you on IRC chat? You should come check out the  openlrs channel on freenode. It's pretty dead nowadays, but a few of us are still active on there sometimes, and it's easier to ask questions & discuss ideas there.

@mikeller 
please grant me write access so I can start actively cleaning stuff up here on the repo...

closed 2018-01-09 by mikeller

linuxslate, 2018-01-30:
Believe it or not, there is still confusion on this.  See edited comment above.

Just updating for any DIY'ers or product designers that may still want to use this module.

jonathan2018, 2018-01-31:
#define SWAP_GPIOS. Wherecan   i can add it ?
#if (BOARD_TYPE == 3)
#if (__AVR_ATmega328P__ != 1) || (F_CPU != 16000000)
#warning Possibly wrong board selected, select Arduino Pro/Pro Mini 5V/16MHz w/ ATMega328
#endif

#if (COMPILE_TX == 1)
HardwareSerial *rcSerial = &Serial;

#define USE_ICP1 // use ICP1 for PPM input for less jitter

#define PPM_IN 8 // ICP1

#define TX_AIN0 A4 // SDA
#define TX_AIN1 A5 // SCL
#define TX_MODE1 11
#define TX_MODE2 12

#define BUZZER_ACT 6
#define BUZZER_PAS 3
#define BTN 7

linuxslate, 2018-01-31:
I am not sure *why* you would want to add it.
CAUTION:  Adding it can cause hardware damage.

If you are a DIY'er (Building or modifying your own hardware) you should correct the GPIO wiring, *not* add #define SWAP_GPIOS

If you are using a supported board, you should not have to add this, and if you do, hardware damage may result.

If you still feel you need to add it, it can be anyplace in the board definition, just make sure it is not inside of the #if (COMPILE_TX != 1)  #else structure, other wise you could end up with it defined for RX, and not for TX or vise-versa.  In this case, hardware damage would almost become a certainty because it will definitely be wrong for one.

The only board that uses it (== 4) puts it at the very end.