openLRSng/openLRSng #88: Question: where is the frequency hopping spread spectrum algorithm contained in the code

ElectricRCAircraftGuy, 2015-03-09:
I'm trying to write my own custom Frequency Hopping Spread Spectrum (FHSS) code for another project I'm working on, and I think I have a pretty good idea how I might go about doing it, but it would be very helpful to see how it's done on other existing systems already too, such as openLRS. Does anyone know where to direct me in the code so I can decipher the FHSS algorithm used by openLRS?

ElectricRCAircraftGuy, 2015-03-09:
Or....better yet, is there any documentation on the FHSS algorithm used here?

kh4, 2015-03-09:
Well it is really simple, there is an array of channel numbers, the transmitter will hop to next channel prior to each transmission and begin from first on the end of the list.

On receiving side the RX will start with slow hop mode listening for full cycle on one channel before going to next. When a packet is received the RX syncs up and enters fast hop i.e. hops after each packet.

If telemetry is enabled the RX will transmit 'reply' on the same channel that the tx->rx packet was received prior to hopping to next channel. 

If a packet is not received by rx it hops to next channel by time (for a full hop cycle before entering the slow mode) so it is able to handle quite a few lost packets in row before losing sync.

ElectricRCAircraftGuy, 2015-03-09:
Thanks! Is there any calibration between the Tx/Rx clocks to help them sync better?  Ex: during initial bind, do the Tx and Rx send each other timestamps repeatedly for some period of time to come up with a correction factor for one of the clocks? This would help ensure they don't lose sync as easily when packets are lost.

Do you know where in the code what you describe is located?  

What you describe makes sense, but I'd like to make sure I understand the gory details.

How long is a single channel hop period?  ie: they hop channels every ____ms or us?

I understand that you can have up to 25 hopping channels. Is that the standard number to use, or is another number more common?
-I assume these are selected randomly during binding, by the Tx perhaps, and then sent to the Rx for both to store them in EEPROM?  

I know BlueTooth uses active hopping technology. What this means is that it marks channels as bad as it uses them. ie: it "learns" which work and don't work, and won't waste time on channels that don't work. Is this implemented, or worth implementing? I envision an array of 1 byte, for instance,  per 4 channels, to keep track of channel status: 2 bits can have 4 states, so you can have 0b00 = bad channel (no packet received for last 2 hops), 0b01 = unknown channel (no packet received for last 1 hop), and 0b10 = good channel. Hops will only occur to good or unknown channels. Perhaps another array of this size would keep track of times skipping a bad channel: ex: you can skip a bad channel 4 times before attempting it again to see if its status has changed. Upon trying a bad channel again, if it works, change its status from bad to unknown, and unknown to good. 
--this technique would make fewer lost packets
--is anything like this implemented?

kh4, 2015-03-09:
There is no clock syncing as such, basically the accuracy from the CPU xtals is more than enough and there is about 1ms gap to allow them getting slightly drifted, the drift wont accumulate since the timing is always based from last received packet.

The code itself is divided in few parts common.h TX.h RX.h maybe bits somewhere else. You may want to check openLRSngDL code instead as it has quite a bit less stuff around the same hopping system.

The hopping period depends on datarate and size of the packet.see getInterval() routine in common.h the interval varies from few ms to ~60ms.

The table is for 24 hops, default is 6.

The channels may be selected randomly, by RSSI scan and manually set by user. The list is sent from TX to RX during binding and stored on EEPROM.

There is no learning since the system is in many cases used without telemetry i.e. there is no backlink that we could use, when telemetry is enabled we could do channel swapping but I do not see much benefit from that atm. As the RSSI based clear channel selection can be triggered without any extra tools this can be used to select clear channels on the field.

closed 2015-03-14 by kh4