openLRSng/openLRSng #91: Garbage in the 6th channel of PPM output frame

Mick59, 2015-04-15:
Hello,
I use an Orange 1W 433MHz set (TX with JR style box, RX with 8 chan + RSSI).
Instead of legacy 8 PWM output for servos, I have set the PPM on CH6, possibility which is offered by OpenLRSng.
I use a 9XR TX with OpenTX and i have set a 16CH PPM on IT, then indeed on the OPenLRSng device via the google coonfigurator too.
Almost all  is working fine. But the 6th channel in the PPM frame have garbage (random values between the three one's : MIN / MAX / TX expected data).

If I switch off the TX, the garbage continue, so this is due to the RX module.

I could search in the source code if you have not time.
Many thanks for your work indeed !

kh4, 2015-04-15:
What are you feeding the PPM into (and seeing the values)?

Does 8ch mode work as expected?

Does 16ch mode work correctly  if you enable "limit ppm to 8ch" in rxconf?

Mick59, 2015-04-15:
After some more research, I found that was a data corruption due to an array overflow in my flight controller (MultiWii modified). 
So the issue can be closed.
I apologize for the inconvenience.
Have a nice day.
Mick

closed 2015-04-15 by Mick59