openLRSng/openLRSng #92: Diversity doesn't switch

fofi1, 2015-04-18:
Hi,

After building and flashing my 2 Broversitys I recognized that the ANT_Detect Pin (PD7) stays low all the time. I have never seen it switching. Even if I disconnect the active Antenna it doesn't want to switch. It is the same with both Diversitys :( .

Is there any way to look up the rssi values of the two antennas? Is there any debug information i can use to look if the HF switch works correctly?

P.S: I'm very happy that Openlrsng now supports diversity (VERY VERY important when we fly down montains. Multipathing is extreme).

Have a nice day,
Florian from Germany

readerror67, 2015-04-19:
Hello Florian,

PD7 was actually put in place but is not actually used. I added it incase code was added to monitor the active antenna in OpenLRSng but this is not needed. R5 should not be populated as it will pull things down to ground.

There are also a few other troll parts that should be left unpopulated:
R9 - Resistor by LED (pulldown)
R18/R4 (Right under PPM)
R5 as covered above.

If you flash the BroversityRX firmware and have it bound the LEDs should both be blinking very quickly, it may look like they are both on but its actually switching between antennas during the pre-amble.

Also make sure you have diversity enabled on the TX, this will generate a slightly longer pre-amble which is used for the algorithm to select the best antenna.

fofi1, 2015-04-19:
Hi,

thanks for the fast response.  R5 is currently populated on my board. I will remove it and test it again next weekend. 

Have a nice day :)

fofi1, 2015-04-29:
Hi,

With the removed Resistor R5 everything is working perfect. Thanks for your great support.

I hope i can make a rangetest tomorrow :)

closed 2015-04-29 by fofi1