openLRSng/openLRSng #95: not binding without ppm

universam1, 2015-05-10:
Because I have build the transmitter by myself, I had it only connected to power and expected it will bind.
But with PPM input of my control unit it binds every power up.
The missing thing was the PPM input, without this signal the transmitter starts not the binding process.

This is IMHO a bug because on multi module TX or for other testing scenarios you will not always have the ppm input available at start time.
Would it be possible to remove that?

readerror67, 2015-05-10:
Signal monitor tab -> Min PPM = 0

kh4, 2015-05-10:
It binds but normally TX will not start transmitting in normal mode until PPM is present, Setting the Min PPM to zero will omit that check.

closed 2015-05-10 by kh4