openLRSng/openLRSng #97: Taranis TX low power causing cleanflight receiver bar fluctuations

Ninjarev, 2015-06-10:
Dont know if anyone is having this problem..  When taranis transmitter drops to 7.8V using a LRS deluxe UHF transmitter module cleanflight has fluctuations in the channel bars in reciever tab. It gets worse as voltage drop gets lower. When it drops to 7.2V the UHF will not bind.  If I charge the radio back up to 8.4 volts it works fine..  Also note using a Sparky and the reviever is a mini DTF..

kh4, 2015-06-14:
Please check by having the TX module connected to PC while taranis is low power, open up the signal monitor tab an see if the bars are stable there.

Ninjarev, 2015-06-24:
Thanks for response I will check it next time battery is low 

 

JIM

csurf, 2018-01-09:
@Ninjarev,
did you ever get this tested and resolved? if so, please post back and close this issue out.

csurf, 2018-01-23:
no response, closing...

closed 2018-01-23 by csurf