openLRSng/openLRSng #98: Orange TX + full recomplie no longer talks to the configurator tool

directgumby, 2015-06-25:
Hello,
I just did a full recompile of the source and flashed via the USBasp but now the TX will not connect to via the Google configurator app.  I get connecting and then a 10 second timeout and it closes.  I've tried all baud rates, and I've ensured that these flags are set:
#define COMPILE_TX 1 // compile TX code
#define BOARD_TYPE 2 // 2 = Flytron OpenLRS M2/M3 Tx Board / OrangeRx UHF TX

//### Enabled Features (some features can be enabled / disabled with compile flag)
#define CLI // Command-line interface
#define CONFIGURATOR // Phoenix Serial Protocol (required for Configurator to work)

Any help here would be appreciated.

As a note: the bottom line for me is that i'm trying to setup the RX unit to use PPM off of port 6.

Thanks

kh4, 2015-06-26:
There is no need to recompile anything to get PPM working. Just use the provided binaries.

directgumby, 2015-06-26:
The problem I have currently is that the configurator tool no longer connects to the hardware, even for flashing.  So I'm not sure how to load just the bin.

Thanks.

Timvrakas, 2015-06-26:
Is compiling from source no longer an option? I really think it should be,
how else can users test their own modifications?

readerror67, 2015-06-26:
I've compiled from source just fine.It even works in Arduino...

directgumby, 2015-06-26:
What I can't quite fixture out is why the serial connection to the TX module does not se3em to work... I can flash it via USBasp, but the FTDI controller does not seem to connect.  Would this be a need for the custom bootloader? (maybe I put on a different one?). If not that, I'm open to other ideas here.

Thanks.

Sent from Windows Mail

readerror67, 2015-06-26:
Blow out the Atmega, full erase. Then flash only the bootloader, then try to flash w/ configurator. Also, check the fuses are correct for the bootloader you are using.

DTFUHF, 2015-06-26:
You might want to close this and start a new issue that addresses the
actual problem you're having, instead of possibly unrelated issues. Also
provide as much info as possible. We don't know what hardware you have, how
you're connecting it, etc.

kh4, 2015-06-26:
If you flash with USBASP from arduino, there will be no bootloader. Connecting should still work but flashing wont.

You can use avrdude (or some of various GUIs of it) to flash firmware binary directly.
OrangeTX TX should be flashed with TX-2-bl.hex from here.

http://openlrsng.org/releases/v3.8.5/433/

Fuses should be set to:
Low: FF High: DE Ext: 05

directgumby, 2015-06-26:
How do I set the fuses?

http://openlrsng.org/releases/v3.8.5/433/

kh4, 2015-06-27:
https://gist.github.com/kh4/3201e21b8d04681434ca

closed 2015-08-05 by kh4