openLRSng/openLRSng #101: openLRSng OLED

eslanek, 2015-12-15:
I would like to implement an OLED by the I2C port on the transmitter.
My question is, which part of the code can cause problems to add the LCD control code ?
I would like to create a similar funtion to the existing one in the sketch dialog.h  to manipulate parameters from the LCD and some switch. What you think is the best option, rewrite the code , or use a mini Arduino to communicate over the serial port whit dialog.h .

kh4, 2016-02-14:
It may be hard to fit code for handling a OLED display (with fonts etc) into the available flash. Especially the type 6 flash is really tight,

csurf, 2018-01-10:
@mikeller, another good candidate for closure...

closed 2018-01-10 by mikeller