openLRSng/openLRSng #102: The "RF Power" setting on the transmitter

kcleung, 2015-12-23:
In the wiki:

https://github.com/openLRSng/openLRSngWiki/wiki/Settings-Guide

the "RF Power" is said to be from a scale of 0-7

Then the wiki says that the scale is "not linear", 

So is it exponential?

and then the wiki says "for 1W device should not be set to 0"

I have the following questions:

* Suppose if I buy a OrangeRx 1W transmitter, and a OrangeRx 100mW receiver, then will the transmission power be automatically limited to 100mW?
* On a 1W transmitter, if I set the "RF power" to "0", then how much power will this translate to?  
* If I want the output power to be 25mW or 100mW, how should I set the "RF power" accordingly in each case?

mlyle, 2015-12-23:
> So is it exponential?

It's poorly defined.  There's saturation effects / it can be sigmoid.  E.g. not much difference between levels 6 and 7 on a 1W transmitter.

> Suppose if I buy a OrangeRx 1W transmitter, and a OrangeRx 100mW receiver, then will the transmission power be automatically limited to 100mW?

No.  It's advantageous to use more power on the ground than the receiver uses-- that way the model can still hear you even when return telemetry starts to drop out.

> On a 1W transmitter, if I set the "RF power" to "0", then how much power will this translate to?
>If I want the output power to be 25mW or 100mW, how should I set the "RF power" accordingly in each case?

Don't know.  Measure it. :/

csurf, 2018-01-05:
@mikeller 
let's close this out...

BTW, can you grant me RW access to this repo?
That way, I can just start cleaning out all of these old issues...

closed 2018-01-09 by mikeller