openLRSng/openLRSng #104: Orange RX - flashing issue

Prabhu-369, 2016-01-02:
Hi all,
I have recently bought  a  orange rx 1W module for 9xr pro.

When flashed with openLRSng code the warning led keeps blinking at a constant rate. No beep sound nothing. I tried to hold down and fail safe button and see if anything changes. But nothing changed. 

I am completely new to orange rx. It would be a great help if someone could guide me to get the module working.

Note: 
I flashed orange rx using turnigy USBasp.
Tried loading the code several times and replaced boot loader also but what i get is a flashing led status(warning led).

Someone please help me.

Thanks in advance,
Prabhu

Nredor, 2016-01-02:
OrangeRX 1W module needs more current than your USBasp can provide. You need to power it from external supply when  you flash.

Prabhu-369, 2016-01-03:
Could you please tell me how much power i have to supply?.

I have tried programming the board by providing power supply using 9XR transmitter. still no use.

After flashing what i get is the warning led blinking at constant rate.. any other suggestion ?

Thanks 
Prabhu

closed 2016-01-20 by Prabhu-369