openLRSng/openLRSng #105: Beacon is out of frequency

villamany, 2016-01-03:
Im using the beacon function but real frequency is one channel down. I E: if i want beacon at PMR ch1 i need to selct PMR ch2 on the clic interface.

kh4, 2016-02-14:
Which frequencies are you referring to, (EU) PMR or (US) FRS.

You can also enter beacon frequency manually.

villamany, 2016-02-15:
thanks. I use PMR channels (EU). 
I already got it working by setting one channel plus than wanted.

csurf, 2018-01-05:
@villamany
I tested this feature for multiple PMR channels (configured from command line), and it works as expected; no issues found...

FYI, I referenced the above PR to note that there's a fix in the queue for extending the channel range to 16 channels for PMR446 (Configurator will also need to be updated eventually).

@mikeller (@kh4 ?) 
Please close this issue out...

PR #151 (2018-01-10): Extend PMR446 channel range to 16 channels
Per ECC Decision (15)05, effective January 2016, PMR446 channel range
was extended from 8 to 16 channels for analog operation. This fix
reflects that change. This also references/partially fixes #105.

side-note, this will also require a small fix to configurator as well...
commit 07aca28b (2018-01-10): Extend PMR446 channel range to 16 channels

Per ECC Decision (15)05, effective January 2016, PMR446 channel range
was extended from 8 to 16 channels for analog operation. This fix
reflects that change. This also references/partially fixes #105
closed 2018-01-09 by mikeller