openLRSng/openLRSng #106: Futaba MicroPPM not working

kolins-cz, 2016-01-09:
[image]
[8 MHz, 16 M Samples [12].zip](https://github.com/openLRSng/openLRSng/files/83839/8.MHz.16.M.Samples.12.zip)

Futaba's compressed 12ch MicroPPM does not work on 3.8.4 version. Firmware sees some movement of channels, but there are wrong channels moving or two at the time in configurator.
Setting "Inverted" makes no change.
Attached screenshot from logic analyzer and raw data. Can be opened with Saleae Logis software.

kh4, 2016-01-10:
Darn, completely forgot to test this, thanks for reminding me ;)

kh4, 2016-01-25:
I tested this by generating microppm with a spare arduino, the only bug I found out is that the invert flag is not handled properly and neeeds a reboot, commit 82888a3 fixes that.

kh4, 2016-01-25:
You can test with firmware files from here http://openlrsng.org/releases/v3.8.7/ until configurator is updated (latest configurator is fully compatible)

kolins-cz, 2016-01-25:
Hello, thanks for this. As a matter of fact I am not sure if I did 
reboot after setting invert. I will try new version and let you know. 
Give me some time, I have to do it remotely with my friend.
Thank You
Mirek

kolins-cz, 2016-01-31:
Hello. I tested this and it does work. When setting Inverted, save and restart it works. With new firmware, it works immediately.  There is only minor bug, on github it is version 3.8.7 but hardware is saying 3.8.6 Can you please post Arduino code to generate microppm somewhere? It might be useful in future. Thanks, Mirek

closed 2016-01-31 by kolins-cz

kh4, 2016-01-31:
https://gist.github.com/kh4/f28a5f181d2c80bbeb00