openLRSng/openLRSng #107: speedup update freq by transmitting only changing channels

universam1, 2016-01-13:
AFAIK every channel is transmitted in order regardless whether its value has changed or not. Could we increase speed and/or range by transmitting only changing channels and submitting 'full' frames only from time to time?

The nature of RC is that during one cycle not all channels are going to change at once. Mostly maybe 2-3 channels only change, while other ch like AUX only in few seconds.

The idea is, regularly sent full frames, between only the changing ones.
I see following advantages: 
- The active channels get automatically priority.
- Less channels altered the more bandwidth, up to almost full bandwidth for a single active channel
- it would be possible to transmit always all 16 channels, regardless if used or not, since they would take almost no air time!

For instance, every channel gets an ID, say 2-byte word with first 4 bits the address 0-15 and remaining 12 bit the value. The TX keeps track of last values and skips channels not changed. Of course on CPPM input there would be some trigger needed against jitter, but SBUS & co would be fine. 
Every 500ms or so a full frame is sent to sync. 
On RX side it would be a simple assigning of channels to array that do arrive.

If worst comes to worst, and the RX doesn't receive, probably this is not an issue, since as a pilot you would add more input if you are missing response which does implicitelly call a update fire of that channel.

What do you think about that?

Thank you!

mlyle, 2016-01-13:
There's a lot of delicate things that come into making a protocol which does this.  This isn't to say it can't be a win, but that it's nontrivial.

First, the receiver needs to be able to deterministically predict where the transmitter will be transmitting at a given time.  So the channel hop time needs to be predictable to the receiver no matter what the transmitter is trying to do.

Second, shorter frames are not always as much of a win as one might think, because of channel hop time, the preamble detector, etc.

Third, the effect of loss needs to be considered.  E.g. if someone clicks a "flight mode" switch which changes a FC from acro to attitude mode, and then immediately moves the pitch stick full forward, what are the consequences if the frame containing the "mode" change is lost but the pitch input is maintained?

ElectricRCAircraftGuy, 2016-01-13:
That's what I see too. The only way one can successfully implement what is
being proposed I think is to have two-way communication, where the receiver
responds to the sender with an acknowledgement packet, and the sender keeps
sending repeat information until the receiver acknowledges receipt of this
information. Otherwise, packet losses can be catastrophic. For one way
communication I don't see a way to avoid this hazard without sending every
channel every frame. This way,  lost packets have essentially no effect, as
the RC vehicle continues operating as commanded so long as the occasional
packet does get through.

Sincerely,

Gabriel Staples
www.ElectricRCAircraftGuy.com

(sent from my Android)

universam1, 2016-01-13:
Well, I dont think it is needed to change the hop time or change the frame lenght at all. I am thinking about to add a channel ID that helps the receiver determine which channel was sent. So no change needed to current transmission, just changing the data.
 
Also, the 'full frame' the one that contains all channels could be deterministically discovered or maybe even an adjustable setting. Or a fallback to 'syncronous' mode by the emergency button click...

Or, if we have 2-way communication, then sent back the CRC. If not correct, fall back to "syncronous" mode.

mlyle, 2016-01-13:
> Well, I dont think it is needed to change the hop time or change the frame lenght at all. I am thinking about to add a channel ID that helps the receiver determine which channel was sent. So no change needed to current transmission, just changing the data.

If you don't change the frame length, you'll be sending data at the present rate.  So what is the benefit from not sending all the channels?  Also the frame that contains all channels can be presumed to be longer...

universam1, 2016-01-14:
Indeed, the 5 bytes chunks made me think it is possible without altering the frame length which is obviously necessary...

I came accross this while working on FRSKY air protocol. Interestingly, that is actually what they're doing, sending active channels on a higher update rate than idle ones. Even more, they are also prioritizing more important telemetry frames. Obviously, it works.
They introduce a frame by a start and end byte + CRC, and channels by an ID. This is smart implemented, since it gives the TX full domain over what to send and how often, thus not requirering the RX to any special timing agreement.

Dont get me wrong, I dont want to propose FRSKY protocol nor mark OpenLRSng as not well done - I love to use OLRSng wherever possible, but I currently still prefer FRSKY for acro planes because of responsiveness. 

Another benefit hence could be, that free airtime bandwidth could be granted between the serial data and the channels updates, dependent of needs.

I think it would be possible to lift OpenLRSng to a higher level with some airtime management like this.

DTFUHF, 2016-01-14:
That's interesting about the frsky protocol, I've looked at D8 and knew
they did telemetry scheduling due to the limited down link, but full frames
are sent on every uplink packet - same with PXX. Also - the frsky uplink is
only about 80Hz iirc, which makes PPM the bottleneck for system latency and
refresh - and openlrsng can easily beat

universam1, 2016-01-14:
Not talking about the D8 or PXX  which is to the TX, but on the air code which is different.
You beat 80HZ update rate with the 19200 baud setting? I guess you mean 115200 baud that I would not consider stable.

DTFUHF, 2016-01-14:
I am referring to the over-the-air d8/16 protocol, working source can be
found in DeviationTX firmware.

From memory, openlrs at 19.2k, the 8 channel link is 71Hz, the 4+4 channel
link is higher

universam1, 2016-01-14:
I checked, at 19200 with 8ch it is at 39hz, with 4+4 it is 42hz.
80hz would be a nice goal.

DTFUHF, 2016-01-14:
Ah, right, I typically don't use telemetry :)

With telemetry off uplink rate is doubled.

kh4, 2016-02-01:
I will keep this in mind if I do drastic changes to protocol.

As for now I am not going to do differential since it would mean differentiating the telemetry/nontelemetry modes significally.

closed 2016-02-01 by kh4