openLRSng/openLRSng #109: No option to set telemetry baud to 38400

norem, 2016-03-31:
KH4,

Is there a reason for not including 38400 as an experimental telemetry speed ?
I've had good luck using 57600  and requesting small chunks of info...

Changing (settings as an example issue) would be when trying to get information from a naze32 flight controller via **SET**... only the beginning of the data is received. Thinking that pulling back from 57600 a bit might help or at least be able to get telemetry at a slightly lower rate than 57600? 

tnx

csurf, 2018-01-05:
@norem 
do you still have a need for this? 

@mikeller 
Setting this up in code shouldn't be too hard, but there are some additional registers for the RFM that need to be configured & tested.

@kh4 
not sure if you're still around and/or able to work on this, but I'd appreciate the help if you could give me some tips on how to best setup the `bind_data.modem_params` values for the 38400 datarate.

csurf, 2018-01-10:
there's a small CLI tool in the "utils" folder on the repo that can be used to generate the RFM configuration register values . I'll see if I can get it to produce some starting values for 38400.