openLRSng/openLRSng #112: ppm puls length issue

alexFab, 2016-07-04:
i try to get openLRS with PX4 to work, but no success so far. I found a [post ](http://diydrones.com/forum/topics/apm-ppm-from-openlrsng?commentId=705844%3AComment%3A1737913)where someone describes the problem.

> Actually OpenLRS produces a very short 'low' pulse on the serial PPM, like 50 µs. For some systems this is not an issue, as the important data is the time between two pulses, not the pulses themselves.
> But for APM (and Pixhawk), this is too short. [....] normal 300 µs low pulses and this works fine
> Is it possible to fix this in the current version? or is this issue known?

DTFUHF, 2016-07-05:
PPM pulse width has been 300us since at least 2012...

the hardware pictured in that post is Orange RX, which comes with original OpenLRS firmware.

The original poster never follows up, so it's impossible to know what he was actually running.

alexFab, 2016-07-05:
okay, my initial problem is: px4 doesn't recognize my openLRS RX (submicroLRS). But [this](https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/generate-ppm-signal/Generate_PPM_signal_nodelay.ino) simple ppm simulator works fine with px4. I'm investigating the differences at the moment.

alexFab, 2016-07-05:
the ppm signal from openLRS is looks pretty much the same like the ppm from the [simulator](https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/generate-ppm-signal/Generate_PPM_signal_nodelay.ino) but inverted. Any Idea?

alexFab, 2016-07-05:
since the issue the different from the titel, i close it here and open a new one.

closed 2016-07-05 by alexFab