openLRSng/openLRSng #113: PX4/APM compatibility

alexFab, 2016-07-05:
px4 doesn't recognize my openLRS RX (submicroLRS). But [this](https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/generate-ppm-signal/Generate_PPM_signal_nodelay.ino) simple ppm simulator works fine with px4. 
ppm signal from openLRS looks pretty much the same like ppm from simulator but inverted. 

Any Ideas were the problem is?

alexFab, 2016-07-05:
the signal from the ppm simulator looks like [this](http://openrcforums.com/forum/viewtopic.php?f=115&t=4277&p=63341&hilit=inverted#p63890) Signal.

kh4, 2016-08-23:
Both polarities of PPM signal are widely used, I can implement a flag to allow swapping polarity.

However, in case you are not using telemetry you should consider switching to SUMD or SBUS.

gregd72002, 2016-10-30:
I would be interested to see this too

AlexFa, 2017-08-11:
> Both polarities of PPM signal are widely used, I can implement a flag to allow swapping polarity.

I would really like to have the possibility to swap the polarity by software. I found a hardware solution but that means more wiring..

csurf, 2018-01-05:
@mikeller 
we should probably close this out, as the guys with the PX4 issue seem to have figured it out, unless there's still a desire to add new logic on RX side to invert the PPM signal?

mikeller, 2018-01-09:
@csurf: Not sure about this one - inverted PPM does not seem to be that common. But adding the option to output inverted PPM would be trivial, so if it's solving somebody's problem then why not?

csurf, 2018-01-10:
@mikeller 
done. let's close this out...

mikeller, 2018-01-10:
@csurf: Yep. If you add `Fixes #<issue number>` to a pull request, the referenced issue will be closed by the system when the pull request is merged.

closed 2018-01-10 by mikeller