openLRSng/openLRSng #116: Feature request: Support serial input for TX module

mikeller, 2016-08-28:
As discussed in https://github.com/opentx/opentx/issues/2894#issuecomment-242977504 , there is interest in getting low-latency serial input for opnLRS TX modules to work.

This should be possible by reusing the serial port of the hardware, if not even by reusing the ppm pin.

DTFUHF, 2016-08-28:
OpenLRSng already supports SUMD, SBUS, DSM, and some hardware has inverters to ensure correct polarity regardless of the controller used, additional protocols can be supported ( pull requests welcome) but there are already three well documented and well supported ones.

mikeller, 2016-08-28:
I may be looking in the wrong place then, I could not find any documentation on how to configure the openLRSng TX module to use serial input. Where is the documentation for this?

DTFUHF, 2016-08-28:
It should be on this page: https://github.com/openLRSng/openLRSngWiki/wiki/Settings-Guide

But since nobody has written it yet, these protocols are available in the 'serial baudrate' drop down menu.

universam1, 2016-08-29:
Could it be possible to implement on TX module side the Multiprotocol (https://github.com/pascallanger/DIY-Multiprotocol-TX-Module) since this becomes kind of the standard (Er9X,Ersky9x, Opentx 2.2) for TX modules?

I know there is little interest to speed up the refresh rate with I-frames, but at least the signal quality aka PPM noise would be eliminated which becomes a problem with these days copter FC. 
Thanks

mikeller, 2016-08-29:
@DTFUHF :

> But since nobody has written it yet, these protocols are available in the 'serial baudrate' drop down menu.

Found them, thanks. I tried to get them to work with my 9XR Pro, running er9x, but no luck. I connected  PPM out from the module bay to the serial rx on the module, but I couldn't get the module to work. Is there a signal level conversion or an inverter needed? (I used Spektrum 2048 / DSMX, as this seems to be the only protocol supported on both sides at the moment.)

marcmerlin, 2017-05-25:
Howdy, do I understand correctly from this issue that this branch allows serial data (mavlink) to flow down from RX to TX module, but does not allow sending mavlink data back up from TX to RX?
In other words, does it mean that one cannot use this as a way to send configuration commands to the aircraft?

mikeller, 2017-05-25:
@marcmerlin: Sorry, but which branch are you talking about?

marcmerlin, 2017-05-25:
@mikeller gitsly branch. That's the only supported way to send mavlink via openlrs modules, is it not?

mikeller, 2017-05-25:
I honestly would not know. There was no activity on that branch since I started working on openLRSng, and  I wasn't aware that this branch is about MavLink support until you said so, the branch name isn't exactly descriptive. ;-)

marcmerlin, 2017-05-25:
@mikeller first, great to hear that you're working on openlrsng :)
Can one (I) use the master branch to pass bidirectional 19200bps serial data?
I think at the time the gitsly branch was made because master did not support this properly.

mikeller, 2017-05-26:
Well, 'working on openLRSng' is a stretch. I submitted a bunch of pull requests that have been open for almost a year now, and despite being well received and successfully tested, they have not yet been merged...

There seems to be some provision for data received on the serial port of the TX to be sent to the RX:

https://github.com/openLRSng/openLRSng/blob/master/TX.h#L731

But I have no idea how well this works.

KnuckleUpFPV, 2017-11-21:
I did some testing with the wolfbox and  multimodule settings. This was with the 3.8.8 chrome gui firmware. Multi module had no effect. If i chose frsky or the xjt, even the dsm for external module I had control. I set telemetry to smart port. On the xjt setting i got rssi and 8 other sensors. Some labled with what they were while others all had the same name. On frsky i got only labled sensors. I had telemetry set to multi telemetry on the tx and smart port on the module.

mikeller, 2017-11-21:
@KnuckleUpFPV: You need to set the 'Serial Port Baudrate' dropdown to 'MULTI' in the configurator. This will only work if you download and install the configurator from GitHub at the moment. I still have not found the time to make the other changes that are needed for the configurator, and then push for a release.

KnuckleUpFPV, 2017-11-21:
Oh i know i talked to you somewhere else about this. I just wanted it to be known that you can connect over serial without any updates and its much faster than standard ppm setup. I have no logical way of showing or proving this. But i can feel it. Its like flying with an x4r  or xm+, but crazy far away. I havent had any dropped packet beeps. Oh i forgot i set the baudrate to sbus as well. Instead of 3 sensors i got many. Rx voltage, battery voltage, no sat under gps, but thats because this quad doesnt have a gps. I wont go back to using ppm and 115200 anymore. I feel much more confident in the link and speed. My rssi also updates very fast now and doesnt drop randomly to 64 or lower from packet loss. It just slowly goes down the further I go. It doesn't work on crsf though. Only works on the others. Give it a try if you get a chance. Still have to test my Orangerx modules to see if they will do telemetry without the mod now. Never got the mod to work so this would be great if it works. Ill try to test tonight and see what happens.

mikerc2017, 2017-11-24:
hello
anyone test this , please make a video to everyone can see .
i want to use it for Taranis , but i do not see output SBUS  on taranis in external modul ?

KnuckleUpFPV, 2017-11-24:
You set the openlrs baudrate to sbus. Then you set external rf on opentx to xjt or frsky

mikeller, 2017-11-24:
Fixed by #122.

closed 2017-11-24 by mikeller