openLRSng/openLRSng #118: WIP: Serial PPM-in over native pins (PPMin)

universam1, 2016-09-06:
This is to give a heads up and request for guidance.

@mikeller and I are working successfully on adding Serial input with Multiprotocol, since the existing SUMD or SBUS protocols are facing dropping support on the controller side opentx/opentx/issues/2894

#### Multiprotocol via Hardware Serial ####
There is already a nice working and tested Multiprotocol implementation via the **HardwareSerial** port from @mikeller: openLRSng/openLRSng/pull/117

### Serial on native pins ###
The logical conclusion was whether it is possible to parse the native PPM-in pin via Software Serial which will bring benefits like
+ __no soldering__ required so plug and play
+ configuration still accessible as before on the hardware serial
+ no additional work then enabling digital transmission

It works! :smile: 
After a lot of tuning and improvements as parity checks we are now up to 99.9% frame integrity!!
That means on 1/1000 frames there is one value "0" thus corrupt, this happens ~20sec. So far this is pretty awesome result given that it's Software Serial. (should be possible to filter this one frame out).

A working softserial implementation on its native pins you are welcome to test it here: __https://github.com/universam1/openLRSng/tree/Multi_with_SoftSerial__

Now I would appreciate some feedback where the stakeholder would like to go.

### custom "OLRS" protocol or Multiprotocol ###
As being said Multi over Softserial does already work 99.9% but there are some downsides of using Multiprotocol (changing standard to come, no ID for OLRS, no CRC). As of opentx/opentx/issues/2894#issuecomment-244875639 we could get integration in OpenTX with a custom protocol.

##### OLRS custom benefits: #####
+ adding CRC gives SoftSerial 100% integrity.
+ adding configuration data to **change values like Power or Telemetry on the field** without a PC hooked up.
+ more efficient telemetry or data transmission
+ not dependent to Multiprotocol changes (that are to come)

##### Multiprotocol benefits:#####
+ already available on some firmwares like er9x
+ reuse some config fields

Now I would appreciate some feedback where the stakeholder @kh4 @DTFUHF would like to go and what MR would be accepted.

Thanks

mikeller, 2016-09-06:
> adding configuration data to change values like Power or Telemetry on the field without a PC hooked up.

Already supported by openLRSng. Invention of a new protocol not necessary for this.

> more efficient telemetry or data transmission

Please explain how you plan to achieve this.

> not dependent to Multiprotocol changes (that are to come)

universam1, 2016-09-07:
> Please explain how you plan to achieve this.

Sure!
#### Multiprotocol #####
+ 8E2 = start + 8bit + parity + 2x stop => 12bit/byte
+ Frame : [sync][3x config][SBUS start][16x 11bit ch] => 27byte / frame

27byte x 12bit = 324bit/frame

#### custom Olrs prot ####
+ 8N1 = 10bit/byte
+ Frame:  [sync][2x config][ 16x 10bit/ch][CRC8] => 24byte / frame
( with 10bit we avoid unnecessary overhead in scaling 1000-2000 == 10bit)

24byte x 10bit = 240bit/frame

Therefore, custom protocol would be 324/240== __35% more efficient__

mikeller, 2016-09-07:
> Therefore, custom protocol would be 324/240== 35% more efficient

What exactly do we get from being 'more efficient' in terms of using less bandwidth on the part of the transmission that already has 5 times more capacity than the radio interface can transmit?

universam1, 2016-09-07:
Time! :)

Not true. Currently the update interval of Mulit is 22,5ms == 44Hz. Therefore, we are already missing and skipping on a transmission rate of >50 Hz that you are above when on 19200 w/o telemetry.

mikeller, 2016-09-07:
Not sure how you are getting to the 22.5 ms update interval, but it sure ain't caused by the bandwidth on the serial link, which allows for 187 bytes to be transmitted in 8E2 during this interval.

universam1, 2016-09-07:
Of course - the question is was how Multi is currently acting and what benefits a new protocol could enable.

csurf, 2018-01-10:
@mikeller,
is this discussion ongoing or resolved?

mikeller, 2018-01-10:
@csurf: Yes. Multiportocol support has been implemented and is supported by most TX firmwares.

closed 2018-01-10 by mikeller