openLRSng/openLRSng #124: 1W module & no PPM -> destroyed module

danielsmd, 2016-11-13:
Problem occurs when using RFM23BP (1W) and PPM signal is missing . After sending a package RFM23BP remains configured in TX mode, and it gets destroyed (Q5  and U4)(RFM23BP doesn't operate ok at 100% TX ON)  . Problem also occurs when using the Configuration Tool, and Tx module is communicating with RX module.
If the receiver is not used, the problem does not occur.

Possible fix : in TX.h, after line 821, add rx_reset();

mikeller, 2016-11-13:
Hmm, the line number you are referring to seems to be odd. Is this the line 821 that you are referring to:

https://github.com/openLRSng/openLRSng/blob/master/TX.h#L821

danielsmd, 2016-11-13:
Sorry. Line 900
[https://github.com/openLRSng/openLRSng/blob/master/TX.h#L900](url)

mikeller, 2016-11-13:
That looks way more like it should be. Will test your proposal.

csurf, 2018-01-05:
@mikeller  @danielsmd 
what's the status on this?
Was this ever fixed?
can this be closed out?

mikeller, 2018-01-09:
Sorry, this dropped off my radar.

@danielsmd: Is this still a problem?

linuxslate, 2018-02-01:
I would also like to know more about this.

Q5 is getting very hot.  I am sure I already destroyed one or 2 RFM23BP's.  I only have 1 left.

It may be normal for the power amp to get warm/hot, but Q5 is not a power part, and shouldn't ever get hot.

It is either this, or I am still not understanding the #define SWAP_GPIO's thing.  Either way, it should not destroy hardware.

I think @danielsmd original report is correct.
Looking at the code (I was working with 3.8.8) for just a few minutes, It looks like if we loose (or possibly never had) PPM,  we don't keep sending packets, so tx_done() is never going to return 1, so we never execute the:
RF_Mode = Receive;
      rx_reset();

I notice that there have been changes in rx_reset().  This has made the 
RF_Mode = Receive;
in the relevant part of TX.h redundant, and it has been removed. the functionality that was in rx_reset() has been moved to the new RFM.h. Perhaps this is intended as a fix, as well as a general clean up.

Note, this is around line 1150 of TX.h in master as of this post.

It definitely seems to me that we have to stop transmitting someplace other than in the if (tx_done() == 1) { } because I don't think we can be assured of finishing sending packets gracefully.

I'll try newer code, and see if there is any difference.

csurf, 2018-02-14:
Although I can't confirm that this was a problem before with the older code, it should no longer be an issue. The recent changes that I made to the RFM code should have taken care of this by automatically putting the module in "READY" mode after transmitting and/or receiving a packet. 

You can see this in the code for putting the RFM into TX mode. There's a READY mode flag that gets set simultaneously to signal to the RFM that should immediately go to READY mode after the packet has been sent (same applies to RX mode). READY mode is a lower-current, semi-standby mode that's meant for the purpose of quick switching between TX and RX modes (see datasheet for more info).

That being said, I've never run into an issue of overheating and/or damaging a module during testing (where I've sometimes also disconnected the PPM input). So, perhaps this was related to some other issue (defective module, sending too much voltage to the RFM, etc.).

In any case, it shouldn't be a problem anymore.
closing this out.

closed 2018-02-14 by csurf