openLRSng/openLRSng #128: Serial conexion no possible without DTR signal

jpino1, 2017-03-16:
Hi,
From https://github.com/openLRSng/openLRSngWiki/wiki/Connecting-to-a-computer

"The following connections must be made between the device and USB-Serial adapter:

    Ground
    TX
    RX

If you are flashing the device, an additional pin is required: **(optional if simply changing settings)**

    DTR
"
I was perform several test and eve the DTR signa is neede for communication, this is a problem for use BT module.
Any chance of disable the mandatory DTR signal for read/write settings with configtool?,It would be very interesting to use the BTmodule to change the settings.
I do not know if this affects sending the telemetry through the BTmodule I have to try it.
Thank you!

csurf, 2018-01-05:
@jpino1 
The DTR pin is used to reset the ATMega328 in order to enter OpenLRSng's "BIND"/Configuration mode (this happens prior to booting up into 'normal mode'). This isn't something that's done by OpenLRSng code; it's a function of how the USB-to-serial connection is normally setup to the AVR. The reset signal is automatically sent to the DTR pin (for a standard FTDI USB-to-TTL serial adapter) upon connecting to the  USB COM port.

For connecting with a BT module, you'd have to do the following:
- connect your bluetooth module to the TX module via the serial connection
- manually power cycle the TX module while briefly holding down the 'BIND' button on the TX module in order to get it into 'bind mode' (continuous beeping, LED's flashing)
- try to connect from your PC to the TX module over bluetooth to the CLI with a terminal emulator such
as putty, or try to connect via the Configurator app selecting the correct COM port for your bluetooth connection

The key is to manually reboot the TX module and place it into BIND mode using the button on the TX since the bluetooth connection isn't able to automatically reset the TX module.

Please try this and post back your results.
If it works, please close out the issue, and consider adding a 'how-to guide' to the wiki so that others can  benefit as well.

jpino1, 2018-01-05:
Hi Csurf,
Thank you very much for the detailed information.
I need a little time to try what you tell me .
I post the results.
Thank you very much for taking the time to answer me.
regards
Jose A.

csurf, 2018-01-09:
@jpino1
please close this issue after you've tested and are satisfied with results.

jpino1, 2018-01-23:
Hi, csurf.
Work as you described, I will close the issue.
Thanks for the help and the great work.
Best  regards.

closed 2018-01-23 by jpino1