openLRSng/openLRSng #129: Inject RSSI on servo channel

ArielBaravalle, 2017-05-01:
hello

when Inject RSSI on servo channel ,  the port output and the channel lost your function adn convert into a pwm proportionale to the RSSI,  ok.

is posible ? add a function to select if only want inject rssi in the ppm  and can use the channel to a servo normal controled from the tx channel selected.

thanks.
regards.
Ariel

mlyle, 2017-05-01:
Can't you just configure the PPM RSSI channel to be outside the range of normal channels, and not send it to a PWM port?  e.g. if you have an RX with 8 outputs, set 8 to PPM, 1-7 to PWM 1 through 7... and inject RSSI on channel 8.

ArielBaravalle, 2017-05-02:
thanks mlyle for your fast reply.  
I use 12 channels in my tx and inject the rssi in the channel 13 , this work fine.
I can asign any channel to the rssi in the INAV config.

thanks again.
Best regards.
Ariel.

closed 2017-05-02 by ArielBaravalle