openLRSng/openLRSng #130: Documentation for state of the gitsly branch?  51 commits behind master ?

marcmerlin, 2017-05-25:
I was looking at gitsly, and got quite worried when I saw this:
This branch is 2 commits ahead, 51 commits behind master.
as well as the last merge in 2014 by @kh4 
master itself is now fairly old and mostly frozen in time it seems, but gitsly looks even older, and by quite a bit.

I was interested in doing bi directional mavlink, but I have brotronics modules and now I'm worried that the mavlink branch is old enough that it's going to be missing code I actually need.

Before I waste a lot of time looking into it, could someone tell me if it's going to work at all, or if I need to throw away my @DTFUHF TX JR module deluxe and other RX modules I have, none of them are supported by uLRS?
Thanks.

podhrmic, 2017-06-06:
Sadly it looks like openLRS development is mostly dead:-/ It might make more sense to fork master and go from there...

marcmerlin, 2017-06-07:
@podhrmic forking whatever doesn't fix anything if no one had the time/knowledge to work on it :)
Some amount of people went to uLRS which is closed source and will become a commercial product, but I guess the main problem is that no one has the time/inclination to work on openlrsng, it's indeed going to stay in the state it's in.
On the plus side, it's not in a bad state, but the mavlink situation is indeed non ideal, mostly not documented, and I guess gitsly has removed himself from the repo and notifications (from what I can tell), so is not interested in continuing work either.
Not ideal, but those things happen. I was just wondering if someone knew more about the last state of things, than me.

podhrmic, 2017-06-07:
Indeed. I briefly looked at uLRS a while ago, but openLRS just seemed to be in better shape. @marcmerlin can you be more specific about the mavlink feature? Is openlrs supposed to transmit mavlink packets? I am curious, because I use it only for radio control, no telemetry.

On the other side, @DTFUHF I am interested and happy to help maintaining this repo if you need a hand.

marcmerlin, 2017-06-08:
@podhrmic the gitsly branch adds a way to pass mavlink data to the TX module like uLRS would, but in a more efficient way that requires less radio bandwidth (that's my understanding).

marcmerlin, 2017-07-21:
@podhrmic BTW about uLRS, it's closed source, maintained by a single person who can fall off the side of the earth one day, and all existing versions have a timebomb in them, so if the author stops releasing new ones on interval, all existing configurators will stop working.
And then the hardware support is poor and it doesn't work on many modules supported by openlrsng.
So, it's not for me.

csurf, 2018-01-10:
@mikeller, we should probably close this and just start a new, fresh issue regarding ongoing mavlink support&development. This issue was more of a user rant anyway...

mikeller, 2018-01-10:
@csurf: Agreed.

closed 2018-01-10 by mikeller

marcmerlin, 2018-01-10:
howdy.
I sure did not try to rant here. Is my original message coming out wrong or unreasonable?
If you feel a new issue is better, great, please open it with the wording you'd like and would you mind referencing it here? (or leaving this one open until there is some state of affairs posted somewhere?)
Thanks much

csurf, 2018-01-10:
@marcmerlin 
no offense meant.
We're just trying to clean up old issues here on the repo that are stagnant and don't really serve much of a purpose.

We've moved the MAVLINK support discussion to issue #152, so you can keep track of the progress there...

marcmerlin, 2018-01-10:
awesome, just what I needed. I've just subscribed to the new one now.