openLRSng/openLRSng #131: 1W Tx used as Rx without hardware modifications ?

mric3412, 2017-05-30:
Using a Tx module as Rx without hardware modifications should be great.
Buzzer input is used for the PPM output and PPM input (connection pin of the Tx module) is used for RSSI. Is it possible to use the PPM input pin for the PPM output instead of RSSI ? (I tried to modify hardware.h but it didn't work)
This configuration works in UltimateLRS (but it's not opensource)

csurf, 2018-01-05:
@mric3412 
I will take a look at it as soon as I can...

csurf, 2018-01-05:
@mric3412 
The reason for the buzzer pin being used for PPM output is due to how the pins are mapped on the OrangeRX 1W TX. The buzzer pin is the only available pin that's tied to the ATMega328's TIMER1. TIMER1 is used in OpenLRSng for both PPM and PWM generation. The PPM out pin is tied to TIMER2, which is used for RSSI and/or LBEEP signal generation within OpenLRSng. These things are hardwired into the MCU and cannot be changed. The PPM generation code would need to be completely redesigned in order to handle using TIMER2 which is an 8-bit timer (lower resolution than TIMER1 which is 16-bit).

In conclusion, sorry, but this isn't really feasible at this point without a major code rewrite. There are other, easier options for setting up a 1W RX. However, if you really must use an Orange TX module, you'll just have to deal with removing the buzzer and soldering in some jumper wires for the PPM connection.

Please close this issue out if you're satisfied...

mric3412, 2018-01-07:
Thank for your reply. I understand it would need to completely rewrite a lot of code. For those who don't want to risk to deteriorate their module, it's interesting to know that UltimateLRS software implements this feature. I have to precise that v1.x of ULRS is open source (I tested it successfully with two 500mW wolfbox modules) Moreover UART is used for MAVLINK link (Mavlink successfully tested with PC/Android GCS as missionplanner) Version 2.x of ULRS is not open source (but free)

closed 2018-01-07 by mric3412