openLRSng/openLRSng #132: SBUS into 1W Rx as Tx

geekness, 2017-07-08:
I have a custom ground station that has a 1W Rx used as a Tx. 
Does the current firmware allow the 1W Rx used as Tx to take an SBUS signal from the ground station to send to my aircraft? 
Does it require a hardware mod? 
Which pin should the sbus wire be connected to? 
Is it configured in the configurator?

mikeller, 2017-07-09:
SBus goes into the serial port Rx pin (the same that's used to flash the firmware). And yes, setting serial speed to `SBUS` will enable it.

geekness, 2017-07-09:
@mikeller thanks.
I assume that the pin labelled "PORT7/RX" is the serial in then?
Do I still power the 1W module with 5V into the BATT & GND port? Or does it need to be fed by the servo pins?

mikeller, 2017-07-09:
Not knowing what hardware you are using, I would think so. Also, the way power is supplied should not change.

csurf, 2018-01-05:
@geekness 
Please close this out if you've managed to get it sorted...

closed 2018-01-09 by mikeller