openLRSng/openLRSng #139: Clean up / re-organize hardware definitions

csurf, 2017-12-27:
@mikeller,
I'm thinking it might be a good idea to do a general reorganization of the hardware defines. IOW, instead of having all of the hardware profiles stacked into a single 'hardware.h', perhaps it might be better to send each profile to its own source/header file, and then include it based on target. 

This would simply help to keep things a bit cleaner and easier for users to read & work with. I'd even like to go a step further and setup a simple folder structure:
- AVR
-- TX
--- (src files for TX hardware profiles)
--RX
--- (src files for RX hardware profiles)

It initially might seem like overkill/over-complicated to setup, but, in the long run, I think it wouldn't be a big deal to keep things more organized by function & platform. 

Specifying the 'AVR' platform might not be necessary at this point, but I figured maybe this same code base could eventually be expanded to handle other platforms, so it might not hurt to separate the files by platform. However, this is just a thought and can simply be scrapped.

Does something like this sound reasonable?

mikeller, 2017-12-27:
Sounds like a good plan to me. The way that `hardware.h` currently is it certainly does not do us any favours.

I have got a couple of things that are more urgent than this, not least of them is a problem with current builds based on #122 causing the TX to reset when trying to connect to an RX in configurator (stack overflow?) with the build configuration that I am currently using.

Another point is that there are no safeguards in place to at least warn if the build toolchain / libraries are not what is expected / confirmed working.

csurf, 2017-12-27:
@mikeller 
ok, cool, I'll try to put this together in a PR.

In general, I think the plan was to get away from Arduino & its dependencies, and instead go to pure libAVR/gcc for builds (I think you've been kinda working on this bit by bit, yes?). This could probably help with the issue you're describing, as we'd have more control over the compiler/library version dependencies (via a makefile/custom config). 

I guess we could probably open up a separate issue to "Remove Arduino Deps." and then keep track of the transition there...

mikeller, 2017-12-27:
Thanks.

Yes, I have always been building the firmware through the makefile. This works, but a lot of the code still relies on the arduino libraries, so it will require a major rewrite to get away from this. In the mean time, I think adding a check for the correct version of the libraries to the makefile will help us get consistent releases and better testing.

linuxslate, 2018-01-04:
In hardware.h

Under:  (BOARD_TYPE == 4) // kha:s openLRSngTX & clones

`#define RF_OUT_INDICATOR A3 // only used for Futaba`

appears twice.

Currently line 569 and line 610 in master.

csurf, 2018-01-04:
@linuxslate 
ok, thanks for pointing that out.
We'll need to fix & clean everything up as we move ahead to refactoring the hardware definitions into separate files...

mikeller, 2018-01-10:
@linuxslate: Thanks for reporting this. Can you please open a separate issue for this? That way it's less likely to get overlooked and forgotten.

csurf, 2018-01-22:
@mikeller @linuxslate
let's get a conversation going on how to best clean this up.
I'll throw out some ideas to start with, and then we can discuss & decide...

method for managing HW type definitions:
- single file for each HW type, with the RX an TX code mixed up together: messy but simplest (this is how it's currently laid out, but in the single `hardware.h` file
- separate files for RX and TX: this will duplicate code that's common to each HW type, but it's the cleanest & easiest option to work with (I personally like this idea)
- a 'base' hardware definition file for each hardware type, which includes the common code that applies to both TX and RX, and then a separate file for TX and RX that gets included based on module function (TX or RX): way more complicated, probably not a good idea
- some other way of cleaning this up??? ideas welcome...

storage method:
- we add a separate folder (named 'hardware', 'HW' or whatever) to store the hardware definition files

Within the above folder, we can:
- add another pair of `RX` and `TX` sub-folders within the main `hardware` for each type
- simply indicate the module type in the filename, such as `rx_3.h`, `tx_6.h`, etc., and place them all in the main `hardware` folder

Please don't think that want to over-think or over-complicate this, but I figured I'd throw out the main possibilities that I could think of  :-) 

Please provide some feedback so that we can get this started & cleaned up...

csurf, 2018-01-22:
also, another thought I just had...
should we stick with the numbered hardware type definitions, or go with something like betaflight, with named 'targets' ???

mikeller, 2018-01-22:
> method for managing HW type definitions:

Can I invite you to have a look into how target definitions are done in Betaflight: https://github.com/betaflight/betaflight/tree/master/src/main/target/

The target specific files are essentially just defines that configure pin mappings and enable features. If done this way, we could largely keep the `TX.h` and `RX.h` plus auxiliary function `.h` files, and move target specific features into them, generalised and guarded by a conditional. Then we split up `hardware.h` into one file per TX / RX, with these files containing only defines.

Thoughts?

fiveangle, 2018-01-23:
Breaking out is really a much better way forward as it also allows potential contributors not necessarily familiar with the architecture to maintain focus on the HW they have, or are looking to implement.  This is probably less of a benefit for BF since there's so many avenues to get community help.  For Marlin 3D printer control software for example, it would be a disaster w/o the targets broken out for individuals to feel more confident in their own sandbox:

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/pins

mikeller, 2018-01-23:
@fiveangle: I disagree there: Code should never be broken out, this will only lead to fragmentation and contributors re-implementing functionality / features (often in a worse way than existing functionality), just because it has not yet been enabled for the target they are working on.
This will also allow consolidation of common patterns, making sure developers' time is used as efficiently as possible when it comes to maintaining existing code.
**Control** of code (through defines) is what should be broken out.

fiveangle, 2018-01-23:
I see I was ambiguous.  I didn't state it explicitly, but I meant the "Breaking out [of target-specific abstractions] (like pins, features etc) into target-specific files] is really a much better way forward…"

I suppose a 👍 on your post would have been sufficent :)