openLRSng/openLRSng #141: configurator baudrate configuration

geofrancis, 2018-01-01:
would it be possible to add an option to set the configurator baud rate? i usually have to go into the code and manually change it so i can access the cli from my bluetooth connection, it would be nice to have this as part of the configurator.

mikeller, 2018-01-01:
Not sure which part you mean. There is a dropdown at the top of the configurator window that lets you set the configurator baud rate. And there is a dropdown in the 'TX Module' tab that lets you set the serial baud rate that the module talks to the configurator.

If you can supply a diff for what you are changing in the code we can look into adding this.

geofrancis, 2018-01-01:
yes but the tx module is only listening at 115200, so if your using bluetooth module with a fixed 57600 you cant connect.

mikeller, 2018-01-01:
So setting the serial baud rate on the TX tab is broken then?

geofrancis, 2018-01-01:
No they are separate. The baud for the serial is normal operation. But in bind mode it uses a hard coded value of 115200 for cli and configurator. It's this hard coded value I would like an option somewhere for.

mikeller, 2018-01-02:
I see. That can certainly be done then. Would it make sense to combine the two? After all it's the same physical port.

geofrancis, 2018-01-02:
That would certainly work.

csurf, 2018-01-02:
I'd vote for a separate configuration baud rate setting for cases where you might need to run different bauds for 'normal' mode and to configure the module.

One weird, but possible, example would be if you needed to run SBUS or some other serial input protocol (the multiproto?) into the module, or if you had some kind of telemetry connection, yet still wanted to be able to connect an external bluetooth module and configure the module wirelessly (as in @geofrancis 's example). In this case, you'd need to configure two separate baud rates: one for configuration and one for normal operation. thoughts?

csurf, 2018-01-03:
we could add a variable to tx_config for the config baudrate. This could either be a stand-alone variable (uint8_t), or we could possibly try to integrate it into the 'tx_flags' variable, and use a bit mask to identify the baudrate. I haven't dug into the code, so I'm not sure if there's space available to do it in tx_flags, but that can probably be easily fixed.

csurf, 2018-01-04:
@geofrancis, @mikeller 
I've got a working version with a separate 'tx config baudrate' setting.
see picture:
[image]

- notice the new 'TX Configuration Baudrate' setting, also that I've connected using 57600.

So, far, it seems to work fine from both the CLI and from within Configurator. Note that this would break compatibility with older versions of configurator, so an upgrade would be necessary for both firmware & configurator (need to look into forcing that requirement. I believe that it's built into configurator; it doesn't allow connecting with older firmware versions).

I just need to do a bit more testing and some code clean-up before setting up the pull request. 

feedback is welcome...

geofrancis, 2018-01-08:
the configuration baud rate should only be active in bind mode.

mikeller, 2018-01-08:
@geofrancis: So your configuration application needs to switch to the non-configuration baud rate to initialise bind mode, and then switch to the configuration baud rate to do the configuration, and then switch back?
This sounds like a recipe for things to go wrong, compared to just using the baud rate that is already configured, and has the potential to break MULTIPROTOCOL.

csurf, 2018-01-08:
@geofrancis,
correct, this baudrate only applies during startup &|| in bind mode. It is the equivalent of changing the hard-coded default '115200' value within the 'setup()' code. Bind mode will simply run at a different, lower, user-configured baud-rate.

@mikeller,
not sure what you mean by switching back & forth. 
can you please explain?

The config baudrate only applies for a short period of time, from boot-up until when the serial port is re-initialized for either one of the serial RC protocols or for the transparent serial/telemetry link. The serial setup happens after the bind mode check completes, so it shouldn't have any effect at all on how the serial port is reconfigured later on.

If you're already in BIND / configuration mode, then it's understood that the serial port is not being used for either telemetry or for processing serial RC input.

Again, unless I'm missing something, this simply achieves what @geofrancis was already doing manually, which is changing the hard-coded baudrate value in TX.h. 

This is the way I tested it:
- connect to TX module via Configurator (or CLI) using the default config baudrate (115200)
- change the config baudrate to something else (ex., 57600)
- save settings to EEPROM & disconnect from Configurator (or CLI)
- changes will automatically take effect as soon as the module reboots / power-cycles
- reconnect via Configurator (or CLI) at the new baudrate (57600)

If bind mode is never entered, then the serial port on the TX module will be configured normally based on the configured settings, and the "config baudrate" will be of no consequence.

csurf, 2018-01-08:
@mikeller 
The only other detail I should mention is that the tx profile configuration that's saved in EEPROM needs to be loaded * before * the the serial port in initialized for the first time (i.e., prior to `checkBND()` call). This is necessary so that the saved 'TX config baudrate' setting can be loaded up and used to setup the serial port.

mikeller, 2018-01-09:
@csurf:

> not sure what you mean by switching back & forth.
> can you please explain?

Switching back and forth between bind mode and 'normal' mode, like when you are binding an RX.

> If bind mode is never entered, then the serial port on the TX module will be configured normally based on the configured settings, and the "config baudrate" will be of no consequence.

So how are you going to bind and then use your module if you never bind it?

geofrancis, 2018-01-09:
it only goes into bind mode if you power on holding the bind button and the only way to get it back to normal mode is to reboot the module so there is no switching back and forward while its on, its started in one mode or the other.

mikeller, 2018-01-09:
> it only goes into bind mode if you power on holding the bind button and the only way to get it back to normal mode is to reboot the module so there is no switching back and forward while its on, its started in one mode or the other.

That only applies to legacy modes.

If you use MULTIPROTOCOL you can start the bind process from the menu on your TX, like for all the other (non-legacy) TX modules.

csurf, 2018-01-09:
@mikeller 

> Switching back and forth between bind mode and 'normal' mode, like when you are binding an RX.

this doesn't affect how bind mode works. It simply changes the baudrate that bind mode uses. Everything still works the same when you bind an RX (the serial port isn't even used for this)

Everything else works exactly the same. The serial port is still configured according to the saved settings (i.e., Multiprotocol should work the same...).

> So how are you going to bind and then use your module if you never enter bind mode?

The way bind mode is handled is exactly the same. nothing changes...
you either enter bind mode by 
1) holding down the 'BIND' button on the transmitter or 
2) sending 'BND!' character sequence to serial port (using the TX configuration baudrate ofc)

mikeller, 2018-01-09:
@csurf:

> this doesn't affect how bind mode works. It simply changes the baudrate that bind mode uses. Everything still works the same when you bind an RX (the serial port isn't even used for this)

So how is the TX going to get data from the module while it is in bind mode? Are you going to submit a pull request to OpenTX / er9x to make the 'bind mode serial port speed' configurable independent of the Multiprotocol speed for these projects?

> The way bind mode is handled is exactly the same. nothing changes...
> you either enter bind mode by
> 
> 1. holding down the 'BIND' button on the transmitter or
> 2. sending 'BND!' character sequence to serial port (using the TX configuration baudrate ofc)
￼
￼So in other words this breaks entering bind mode through Multiprotocol setting the bind flag?
I am really against breaking Multiprotocol just to enable some niche application for legacy modes.
￼￼  ￼￼￼  ￼￼￼  ￼ 
￼￼
￼Write

csurf, 2018-01-09:
@mikeller 
I think there might be a bit of misunderstanding here.
I don't see how this affects the multiproto, but I might be wrong. 
If a user needs to run multiprotocol and wants to be able to enter bind mode through it, then the bind mode / 'tx config' baudrate is simply left at the default value of '115200'. Nothing else changes and everything else works as expected.

For others that need to run a different config baudrate for whatever reason (bluetooth), then they can simply change the setting without having to reflash the firmware... Doesn't seem like a big deal to have it has a user-configurable setting instead of a hard-coded value; it just makes things a bit more flexible.

Unfortunately, I don't have a multiprotocol configuration to test with, and it might take some time for me to figure out how to set it up. Would you be willing to help test this out? We just need to confirm that this wouldn't break anything with multiprotocol as long as the 'bind mode baudrate' is left at default (115200).

@geofrancis, feedback?

mikeller, 2018-01-09:
> If a user needs to run multiprotocol and wants to be able to enter bind mode through it, then the bind mode / 'tx config' baudrate is simply left at the default value of '115200'. Nothing else changes and everything else works as expected.

This way this introduces the potential for users to break their Multiprotocol configuration, with no benefit when using Multiprotocol.

My suggestion would be to make this new 'configuration only serial port speed' setting only apply when a legacy mode is used, but not in Multiprotocol.

I am happy to test this - I am working on a pull request that introduces Multiprotocol feedback from the module to the TX at the moment anyway.

csurf, 2018-01-09:
@mikeller 

> My suggestion would be to make this new 'configuration only serial port speed' setting only apply when a legacy mode is used, but not in Multiprotocol.

this is what I've done, although I still don't understand what you mean by 'legacy mode'. The 'tx config' baudrate setting only applies to bind mode and nothing else... It doesn't affect any of the other serial configuration settings.

please check out my commits and let me know what I'm missing...
https://github.com/csurf/openLRSng/tree/tx_config_baud

mikeller, 2018-01-09:
Ah, I misunderstood you then, I thought you meant you were going to change the baud rate every time the module goes into bind mode. This applies the custom baud rate only on startup, before the module has started processing Multiprotocol frames, so it won't be an issue.

Maybe it should be called `startup_baud_rate` or something?

Also, see my comment in your commit re moving reading of the config to after setup of the RF module interrupt handler.

csurf, 2018-01-09:
@mikeller 
ok, cool.
yeah, I'm flexible on variable names.
maybe something like 'boot_baud_rate' could work.

geofrancis, 2018-01-09:
config_baud_rate?

csurf, 2018-01-10:
how about 'console_baud_rate'? That should imply both CLI and Configurator, or a 32u4 board that has technically has a "console port".

Is that cool? 

In any case, this will just be the internal variable name within code, so users won't see or have to deal with it...

mikeller, 2018-01-10:
`console_baud_rate` sounds good to me. (I thought it would also be exposed in CLI?)

csurf, 2018-01-10:
We can label it whatever we want within the CLI and/or in configurator ;)

csurf, 2018-01-12:
@geofrancis 
you wanna give this a try? 
If we get this tested & confirmed, we can finalize things and close this out

geofrancis, 2018-01-12:
I am away working just now i wont be able to test it until late next week.

csurf, 2018-01-23:
@geofrancis this has been merged into both firmware and configurator, so I'm going to close this. Please re-open if you have any issues with it.

closed 2018-01-23 by csurf