openLRSng/openLRSng #143: Convert serial baudrates to fixed, indexed values?

csurf, 2018-01-04:
Currently, the code is setup to use 'uint32' for storing the serial baudrate for the RC serial link. This allows for setting arbitrary baudrates, but takes up more flash space (32bits x 4 profiles) and also more memory. Probably not a big deal, but every bit counts on the AVR.

Wondering if we should instead convert this to a key-valued setup, similar to how the 'data rate' setting is setup, where the most common baudrates are stored & configured by ID:
0: 1200
1: 2400
2: 4800
3: 9600
4: 14400
5: 19200
6: 28800
7: 38400
8: 57600
9: 115200

We could then have a couple of helper functions to convert between ID and actual baudrate, and for setting the baudrate, without having to store the actual value. 

FYI, the 'protocol' serial rates are already basically configured like this, but with different indexing:
1: SPEKTRUM1024
2: SPEKTRUM2048
3: SBUS
4: SUMD
5: MULTI

This would take a bit of reworking to the existing code, but seems to make the most sense. "Configurator" already has hard-coded baudrates (in a drop-down list), so most users are probably already used to working the existing set of "standard" baudrates (the CLI allows for entering arbitrary baudrate values, but I doubt anyone is using it for the purpose of configuring random, non-standard baud rates).

How does this sound?

mikeller, 2018-01-10:
Not sure about this. The avr libraries accept an uint32_t value for baud rate, so to me it makes sense to store the value in this format, and be ready to support non-standard baud rates (SBUS / Multiprotocol already require 100000 for example). The saving of 12 bytes EEPROM seems to be not worth the hassle of having to deal with things like potential versioning issues if we ever have to change the enum.

csurf, 2018-01-10:
fair enough, makes sense... closing this out...

closed 2018-01-10 by csurf