openLRSng/openLRSng #144: signal monitor broken in Configurator for PPM input

csurf, 2018-01-05:
@mikeller 
there's an issue with how you recently setup the initialization for RC Serial & PPM that causes the 'signal monitor' within Configurator to break.

Basically, what happens is that The calls for setting up both the serial link and/or the PPM input interrupt routines happen **after** the check for bind mode is performed...
see TX.h:
`
  checkBND();

  checkButton();
  Red_LED_OFF;
  buzzerOff();

  setupRcSerial();
  checkSetupPpm();
`

This order is fine & great for avoiding the unnecessary initialization of PPM input for the serial RC protocols. However, it breaks the signal monitor for PPM input within Configurator due to the fact that PPM input is never initialized due to the short-circuit into bind mode.

A work-around to get the signal monitor working is to hit the 'Save to EEPROM' button within Configurator from either the 'TX Module' or 'Signal Monitor' tabs. What this does is forcibly reset/re-init the PPM interrupt due to the fact that the routine for saving TX settings via the binary protocol re-configures the PPM input (see the code for 'PSP_SET_TX_CONFIG' within 'binary_com.h'). 

FYI, this 'PSP_SET_TX_CONFIG' routine also causes it to bypass your 'checkSetupPpm()' function, which means that the code will attempt to initialize PPM even for serial RC configurations. Although this isn't really a big deal since it's understood that exiting configurator should cause/force a module reboot (you're tossed back into bind mode after disconnecting).  Still, I figured I'd mention it as it's an easy fix...

It seems like the logic needs to be re-ordered in some way so that it can correctly handle both PPM and serial RC in all cases (i.e., initialize PPM only when necessary for both 'normal mode' and for 'bind/configuration mode'). 

@mikeller 
I could hack something together, but since you're the one that came up with the refactored serial/PPM detection code, I figured I'd hand it over to you first to see what your thoughts were...

mikeller, 2018-01-09:
Thanks for pointing this out, @csurf. I think once your change in #148 has been merged, moving `checkSetupPpm()` to before `checkBND()` should fix this? (Plus amending `PSP_SET_TX_CONFIG` do use `checkSetupPpm()`.) Thoughts?

csurf, 2018-01-09:
@mikeller 
ok, seems like that should work fine...

PR #155 (2018-01-11): Fixed PPM signal monitor.
Fixes #144.

@csurf: Can you please check if this fixes the problems with signal monitor when using PPM?
closed 2018-01-11 by csurf