openLRSng/openLRSng #149: Problem Telemetry  Multiprotocol  OLRS  with Taranis

mikerc2017, 2018-01-09:
Hello Developer

i have problem with Telemetry  in Mode multiprotocol OLRS.    RSSI  worked OK,  when connect to FC F3 , use same a uart . Taranis usually warring ""Lost sensor """, and i see sensor not stable ,wrong value . I used telemetry Frsky 19200 band 

i wonder why when i use two Uart , it not work , Same a uart it work ?  

thanks

mikeller, 2018-01-25:
Hi @mikerc2017. The easiest way to make this work is indeed to use the same port on the flight controller for serial RX and telemetry. The reason for this is that the openLRSng RX module has only one serial port, and has to use this port for serial RX and telemetry. This means that both of these protocols have to run at the same speed, and with the same protocol settings.
When sharing the same port as well on the flight controller, this works fine. When using different ports, independent settings for speed and parameters will be used for serial RX and telemetry. In most cases they are not the same, making this set up unusable on the openLRSng side. If you want to use two separate ports on the flight controller side, make sure that both of them use the speed / port settings that is the default for the serial RX protocol that is used.

mikerc2017, 2018-02-12:
mikeller 
can fix problem atm ?

mikerc2017, 2018-02-13:
i used invert by transitor though pin Smartport
setup bandrate auto 

i also tested on Er9x (no invert ) .Same problem .

mikeller, 2018-02-13:
This is odd, I have tested this and found it working for both  er9x on 9XR Pro without inverter, and OpenTX on Taranis with inverter. Do you know that you have to disable the built in inverter in er9x in the 'Telemetry' menu?

mikerc2017, 2018-02-13:
Yes.i knew way telemtry work on tararanis. and er9x ,er9x pro . 
Can you try test with a fc F3 ?  i very need someone test it to see problem .

mikerc2017, 2018-02-19:
hmmmm.

mikeller, 2018-02-19:
Sorry, I haven't exactly had much time to look into this as of late.

> Yes.i knew way telemtry work on tararanis. and er9x ,er9x pro .

So you got it working with openLRSng?

> Can you try test with a fc F3 ? i very need someone test it to see problem .

What exactly does not work on F3? It should work if you're running it in the same way as on F1. If using Betaflight / cleanflight, you'll have to `set tlm_inverted = on` in order to make it work, which is different for most F1 boards that don't have external inverters.

mikerc2017, 2018-02-19:
Oh No 
i tested with F3 spracing , f4 onimbus ....both worked . 

but as whats i said . Still little problem " lost sensor " . i hope you test to see problem same me .

mikeller, 2018-02-19:
Yes, true, we've been there before. I get stable telemetry data in my tests (or no data at all if misconfigured). This suggests that it is the inverter, or something else in your configuration. Do you use the pins in the JR module bay for the telemetry connection, or an external serial connection?

mikerc2017, 2018-02-20:
yes . i used 5 pin on Jr modul. 
i do not think i wrong when i have same a problem on 3 tranmitter . X7, x9 and er9x.

i read on betaflight , seem use same a uart on Fc have problem . Have to modifi code on betaflight to it work .
Xsr/ x4r / D4r also use two uart for sbus and telemetry. But Openlrs only nees a uart ....

mikeller, 2018-02-20:
Try the external serial port on your TX, directly or through an inverter to the serial port of your module, and see how that goes.

The FrSky RX are different from openLRSng, as they use two different serial ports on the RX module, and run both protocols at different speeds. openLRSng uses only one serial port on the RX module (since there is only one), and uses the speed of the serial RX protocol for both serial RX and telemetry.

mikerc2017, 2018-02-20:
ok . i will try on a external serial port on Tx ( in bõ batterry x9 ) and report result.

mikerc2017, 2018-03-15:
everyting no change .Mikeller 
can you check again one more ? Wait about 5- 10 minutes .
Thanks

mikeller, 2018-03-15:
As I said, it works fine for me with ersky9x. Have you tried er9x? I know it's not ideal, but it could be that OpenTX uses different timeouts to detect sensor loss, and is therefore more sensitive.

mikerc2017, 2018-03-15:
i also used ER9X  firmware Multiprotocol . Same problem , sometime lost value ....
unfortunately i have not a ersky9x to test .

Opentx   sometime  warring  lost sensor  then have  sensor back again ....and ....continues....as whats in video . I  do not know  which version OPENTX fixed problem . Sad

geofrancis, 2018-03-16:
Have you got all the ports running at the same speed? You need to run the frsky Telemetry at the same speed as serial Rx. Since they are using the same serial port on openlrsng. 

So if your using sbus that means 100k Sumd and DSM run at 115200 I think.

mikerc2017, 2018-06-07:
@mikeller 
i used 3V3  for atmega328 16MHZ on RX .so  can this is cause ? did you use 5v or 3v3 on RX ?

mikeller, 2018-06-07:
I used an orangeRX module. Presumably Vcc for the MCU is 3.3V, but I never checked this.

mikerc2017, 2018-06-07:
@mikeller 
maybe that is not cause .I almost have not idea to fix problem now

mikeller, 2018-06-07:
I wish I could help you, but diagnosing something that I cannot reproduce is very hard.

mikeller, 2018-06-08:
@mikerc2017: What do you want to see in the video?

mikerc2017, 2018-06-08:
@mikeller 
i want to see RX connect Uart to FC and value Sensor on taranis screen ( VFASS/ GPS ..) . In 2 minute no warring lost sensor

mikeller, 2018-06-08:
I don't know what you want to learn from this. I can tell you it's working.

mikerc2017, 2018-06-08:
@mikeller 
I want to see it in real.please  ,a people like Openlrs

mikeller, 2018-06-08:
It works. You can either choose to believe me, or not. In both of these cases, having a video of it will not help you solve it.

mikerc2017, 2018-07-06:
@mikeller 

i saw cause of problem . When i OFF accelerometer on Betaflight , It work normal . So much  sensors on Telemetry will be problem ????

So this still is problem data from RX  ?

mikeller, 2018-07-07:
@mikerc2017: That is an interesting bit of information. There's a limited amount of bandwidth available in openLRSng to send back telemetry information. If too many sensors are active in the firmware, then the telemetry channel will be overrun,from the looks of it leading to no telemetry at all being received.

mikerc2017, 2018-07-19:
@mikeller if Off ACC . can not use mode horizon or angle . 
When i connect GPS  , it sometime lost sensor I delete many sensor not use ...... Hmmmmm  So much problem about telemetry

Daniel-1276, 2018-08-27:
Hi, did you found a method of disabling single telemetry data types? I am using ORangeRX 1W as TX and OrangeRX 100mw as RX. Same issues  here, using SUMD and FRSKY telemetry protocol on both sides (using INav on F4 board, same serial Port, telemetry set to 115k manually). Did tried SBUS before, but same issue. Seems to be a bandwidth problem on openlrsng side.

mikerc2017, 2018-09-02:
@Daniel-1276  not yet. @mikeller do not support this on Taranis . You can do it by Config  although i see it  inconvenient

mikeller, 2018-09-02:
@mikerc2017: Please stop making rubbish claims about what I do or do not do. You were told that this is not possible, you still seem to refuse to believe what you are told.