openLRSng/openLRSng #150: FR-SKY X series sensor polling.

geofrancis, 2018-01-10:
to get frsky X series telemetry from a flight controller to a handset I use a mavlink-frsky converter normally used on a pixhawk>X8R setup to get telemetry down via smartport but I use it connected directly to the taranis so i can get mavlink down through openlrsng-gitsly then convert it to smartport for the handset.

normally this isnt possible but I discovered that you could wire a flight controllers smartport directly to a taranis and pick up the sensors if you had a X8r receiver connected to the internal module. as it turned out the flight controller would still receive the polling requests from the receiver and the other end of the RC link.

so it got me thinking then I noticed that a new feature had been added to the S.port telemetry library that could poll sensors so when i put it inline with the telemetry converter I had data on the screen.

this got me thinking about S.port on openlr-sng, I think to get frsky X series sensors to work you would just have to add a small amount of code to poll the sensors then send whatever it gets to the taranis and it should display them.

look at the part called DATA POLLING 
https://www.rcgroups.com/forums/showthread.php?2245978-FrSky-S-Port-telemetry-library-easy-to-use-and-configurable

mikeller, 2018-01-10:
This should theoretically be possible - at the cost of giving up serial RX, since SmartPort requires both TX and RX lines on the receiver to talk to sensors.

geofrancis, 2018-01-10:
true but not everyone needs serial RX.

mikeller, 2018-01-10:
I didn't mean to imply that.

geofrancis, 2018-01-10:
anyway, like I said I think it would take minimal amount of code to make this work, it wont do betaflight lua and all that jazz but should work with anything that puts out normal smartport sensor data.

this is when i was discussing it with mike of er9x 
http://openrcforums.com/forum/viewtopic.php?f=7&t=8396&hilit=polling&sid=2236672ba07f4249a32974fec0965c69&start=450#p119439

geofrancis, 2018-01-10:
from the smartport library

`ifndef _FRSKY_SPORT_POLLING_H_
#define _FRSKY_SPORT_POLLING_H_

#include "Arduino.h"
#include "FrSkySportSensor.h"
#include "FrSkySportSingleWireSerial.h"

#define FRSKY_POLLIED_ID_COUNT 28

#define POLLING_RSSI_DATA_ID   0xF101
#define POLLING_ID_POLL_TIME   12
#define POLLING_RSSI_POLL_TIME 444

class FrSkySportPolling
{
  public:
    FrSkySportPolling();
    FrSkySportSensor::SensorId pollData(FrSkySportSingleWireSerial& serial, uint32_t now);

  private:
    FrSkySportSensor::SensorId getNextId();
    const FrSkySportSensor::SensorId POLLED_ID_TABLE[FRSKY_POLLIED_ID_COUNT] = { FrSkySportSensor::ID1,  FrSkySportSensor::ID2,
                                                                                  FrSkySportSensor::ID3,  FrSkySportSensor::ID4,
                                                                                  FrSkySportSensor::ID5,  FrSkySportSensor::ID6,
                                                                                  FrSkySportSensor::ID7,  FrSkySportSensor::ID8,
                                                                                  FrSkySportSensor::ID9,  FrSkySportSensor::ID10,
                                                                                  FrSkySportSensor::ID11, FrSkySportSensor::ID12,
                                                                                  FrSkySportSensor::ID13, FrSkySportSensor::ID14,
                                                                                  FrSkySportSensor::ID15, FrSkySportSensor::ID16,
                                                                                  FrSkySportSensor::ID17, FrSkySportSensor::ID18,
                                                                                  FrSkySportSensor::ID19, FrSkySportSensor::ID20,
                                                                                  FrSkySportSensor::ID21, FrSkySportSensor::ID22,
                                                                                  FrSkySportSensor::ID23, FrSkySportSensor::ID24,
                                                                                  FrSkySportSensor::ID25, FrSkySportSensor::ID26,
                                                                                  FrSkySportSensor::ID27, FrSkySportSensor::ID28 };
    uint8_t  nextPollIdIdx;
    uint32_t nextPollTime;
    uint32_t rssiPollTime;`

geofrancis, 2018-01-10:
a good test would be to put an smartport poller inline with a receiver with rs232/ttl coverter and diodes , then set openlrsng to transparent @ 57600 to see if any sensors are discovered.

https://www.rcgroups.com/forums/showpost.php?p=35812213

mikerc2017, 2018-02-22:
midelicom Rcgroup  is a smart guy . He intergrated some project 2.4ghz  on Fc . Even he intergrated OLRS  on Fc .So if he can intend OLRS , it will great to develop OLRS

jack113113, 2018-05-13:
i really want to use feature change PID  on smartport or Fport . So i hope Smartport or Fport can complete now .