openLRSng/openLRSng #152: MAVLINK support

csurf, 2018-01-10:
Let's post all questions, comments, and discussion relating to MAVLINK support here...

@mikeller ,
how should we go about dealing with the old gitsly/mavlink branch? It looks like the branch has fallen behind but can still be merged. We should at least get it rebased & up to date. Then, maybe we can start thinking about doing some testing and maybe getting the changes merged into trunk?

geofrancis, 2018-01-10:
as far as i know the main reason it was left as a separate branch was that it used a lot of space to do mavlink and the fast serial implementation wouldn't work with the 32u based hardware.

csurf, 2018-01-10:
@geofrancis 
ok, good to know.
I guess that's more reason to get the gitsly stuff updated in order to compile and see how bad the space/memory usage is...

We just saved some space by conditionally removing the old i2c 'slave mode' feature. I also have some code refactoring in the works that might be able to save us some more space as well. 

Perhaps we should revisit the fast serial/32u stuff and see if there are any other options or some other way to get it working.

mikeller, 2018-01-10:
@csurf: Yeah, I'd be all for reviving the Mavlink branch, it seems to be useful functionality. Getting it merged into the main branch would be the optimal outcome, but if size constraints don't allow it, keeping it as a 'flavour' would be an alternative option (the same will potentially have to be done for SmartPort telemetry support, there is also quite a bit of code that is needed there).

Rebasing the Mavlink branch on latest master seems to be a reasonable first step - probably best to be done with somebody with hardware to test it.

geofrancis, 2018-01-10:
#54 could also be closed

geofrancis, 2018-01-10:
I would also check the gitsly fork for other changes, I vaguely remember him saying that he had to remove some features to get it to fit, or at least he did in v3.5 im not sure if that is still the case in later versions.

geofrancis, 2018-02-12:
I found the source code for ULRS V1, it might be worth looking at as a transparent telemetry mode so we can run mavlink without all the libraries.

https://github.com/geofrancis/Projects

csurf, 2018-02-12:
@geofrancis 
I'll take a look at it when I get chance.
Having an air rate of 57600 sounds like it would greatly reduce sensitivity & range, but I guess this might not matter in some situations (mid-to-short range APM setup).

I've actually been testing & working on some of the serial link code in order to make sure it works with all of the recent changes. So, I've gained some more insight on how it works. Down-linked serial packets 
(from RX to TX) are sent almost immediately / on the next out-going telemetry packet, while up-linked serial packets have to wait in a type of queue which places them in a lower-priority than RC packets. 

It's a pretty basic system right now that forces the serial packets to be sent after the RC packets. However, I'm wondering if perhaps we could re-think this design so that we can somehow adjust the 'weight/priority' of the serial packets with respect to the RC packet traffic (this would apply to the TX up-link serial traffic).

For example, you might have an APM semi-autonomous setup where you want to prioritize MAVLINK serial traffic over RC traffic (like maybe an 80-20 split). Or, you might have a setup where you want a higher RC refresh rate & don't care as much about the serial up-link, in which case you'd prioritize RC traffic. Does something like this sound good?

I'm also seeing that there's some ACK / re-transmission logic that wasn't fully implemented, and/or it was only implemented on the TX side / for re-sending lost up-link packets. This might also be something to take a look at.

geofrancis, 2018-02-14:
what crossfire is doing with mavlink is interesting, rather than having separate RC and Telemetry, its just running as a pure telemetry link and injecting mavlink radio packets into the uplink, essentially doubling the upload bandwidth and simplifying the wiring so its just a single serial to the flight controller rather than ppm+telemetry.

madhunm, 2018-02-15:
One observation from using the gitsly branch from address ago: the telemetry logs aren't saved by mission planner.

Before you point fingers there, let me tell you that using the 3DR radios will save the telemetry logs; no problem.

geofrancis, 2018-02-15:
first i have seen that? mission planner should log anything coming in.

madhunm, 2018-02-15:
I've used gitsly on at least 5 setups; in all of them, there were no telemetry logs.

madhunm, 2018-02-15:
Telemetry works fine mind you... Just that the logs aren't saved. 

Does it have to do anything with the CTS pin not being connected?

danielsmd, 2019-04-21:
I'm trying to understand the bottleneck on the mavlink communication (HW or SW).
I'm using Gitsly 3.8.5 , and Mission planner reports a mavlink quality is 60-70% ( I assumed arrived packets- vs lost).
Data rate is 57.6k (air and uart) . Packet size is 29 (default).
I've tried with default configuration and with HW SPI @8Mhz(on Tx & Rx).  No improvement.

whoim2, 2021-07-13:
We need 2side mavlink!)