openLRSng/openLRSng #153: repo clean-up

csurf, 2018-01-10:
@mikeller,
the 'pavn' branch looks like an old, stale branch that's no longer of any use. It looks like kha had used it for testing when he was building the channel scanner, but there doesn't appear to be anything useful in it. It should probably be deleted...

Also, the development branch should probably be cleaned up / rebased so that we can start using it to test stuff before moving into master. thoughts? 

In general, perhaps we should probably come up with a general strategy for dealing with development, collaboration, new feature testing / betas, versioning, etc.  Up until now, kha would basically handle all of the development. There wasn't much collaboration going on, so there really wasn't a need for a system to keep track of repo activities... 

Not sure what your opinion is on this, but it seems like it might be a good idea to come up with some simple kind of organization/system.

mikeller, 2018-01-10:
I am not fazed about having old / stale branches around - they are not in the way and don't create extra work, but if there is nothing of value in it it can be deleted as well.

Re development branch, I don't think we need that. Seeing how we are working in Betaflight (which has a lot more code and change than this repo), we have given up on parallel branches, and use a model similar to 'cactus' (https://barro.github.io/2016/02/a-succesful-git-branching-model-considered-harmful/):
- all pull requests are made against master (makes it easy for outside contributors);
- changes are tested on the pull request branch, then merged;
- SemVer! (https://semver.org/);
- major / minor releases are tagged on master, with an announced 'feature freeze' / release candidate phase beforehand;
- bugfix releases (if required) are tagged on a maintenance branch that is cut off the release tag in master, with bugfixes made by pull request into master, then cherry-picked into the maintenance branch.

This model is working reasonably well in Betaflight, and it does not have the need for a 'development' branch.

Thoughts?

csurf, 2018-01-10:
excellent!
this is exactly the kind of feedback I was looking for.
I'm all for getting some more structure & discipline going here, so this sounds great.
I'll follow your lead, so please let me know if&when I screw up; I'll to my best to get up to speed (I'm not a git pro, but I'm trying to get up to speed).

csurf, 2018-01-11:
@mikeller 
question about small changes (as in #157): For small, simple things such as this, is it ok to just push a commit directly onto master, or should we always do a pull request & merge for even the simplest of changes (typos, comments additions/fixes, formatting, etc.) ?

geofrancis, 2018-01-11:
My only concern with things just been added to master is that unless the GUI is updated at the same time your going to make it incompatible. As far as I know the way kha done it was to put all new changes into a Dev branch then push it into master when a matching configurator was working.

mikeller, 2018-01-11:
In my opinion, everything should be done through pull requests. A big part of this is that GitHub seems is build around the concept of doing it this way:
- pull requests will generate notifications so users wanting to keep up with change can use them to do so. Direct commits into master will often go unnoticed until stuff breaks;
- discussions and comments are a lot easier to have on pull requests, and they are certainly a lot easier to find in the archive.

As a mitigation for the overhead from making pull requests, doing cleanup like #157 can be done as part of a general cleanup of an entire file in one pull request.

csurf, 2018-01-12:
ok, cool, understood.

I think there was a comment by @geofrancis that got deleted regarding keeping the compatibility between the firmware and Configurator synchronized. Perhaps we should agree that any firmware PR that affect or break Configurator should only be accepted if there's a corresponding PR on the Configurator repo? How do you guys manage this in betaflight?

mikeller, 2018-01-12:
Pretty much by expecting matching pull requests, and synchronising their merges. We use labels in both repos (all three now with `betaflight-tx-lua-scripts`) to mark pull requests requiring coordination. It's not ideal, but it works.

geofrancis, 2018-01-12:
yes, sorry I was going to repost it and got distracted, I think the way kha done it was to keep all new changes in the dev branch until there was a matching configurator and release them at the same time to stop it breaking compatibility. 

I suppose if any changes dont effect the configurator then it should be ok.

geofrancis, 2018-01-15:
have either of you looked at the 32bit openlrsng? there is not much information about it.

mikeller, 2018-01-15:
I have heard about it. But there does not seem to be any hardware available for it commercially. :confused:

geofrancis, 2018-01-15:
from the look of it the last release was 3.2. 
https://github.com/openLRSng/openLRSng32/releases

im not even sure what cpu its running on but if its a f1 those are pretty cheap to do diy.

csurf, 2018-01-15:
@geofrancis 
seems like hardware support would be the main problem.
Maybe once a hardware design is finalized and we can get some functional code for it, someone might come along and build some modules for it? DTFUHF/hawkeye appear to have shut down their hardware production...

I talked to englishman ( @DTFUHF ) on the openlrs irc chat channel, and he had recommended the F0 chip for future development. I haven't read up enough on the stm to be able say if that's a good call or not. Other users have pointed to other chips (ESP32). 

There's still also the question of whether or not we should look at switching over to LoRa or some other, better (FSK) RF module.

Timvrakas, 2018-01-15:
Hello LRSrs people,
I am working on a project with fellow University students to build a radio for flying projects. We don't have any released information yet, and our prototypes are far from complete, but early tests showed scuess up more than 100 miles at 500bps. The hardware is based on the SiLabs4463 and a Teensy 3.2 currently. At the moment we are using slightly modified RadioHead code. If any of you are interested, I can work on getting designs released and keep you updated.

mikeller, 2018-01-15:
Hi @Timvrakas. This does sound interesting. Can I suggest that you create an issue for this, and post updates to this issue? This will make it easy to keep track of it.

geofrancis, 2018-01-16:
I think its a bit of a chicken and the egg problem, no one is going to make hardware if there is no software for it, so i think the best option would be to get the code working on a prototype using a devboard like https://www.ebay.co.uk/itm/152246242164 and a rfm22b/23bp, that would let people build there own, then people will eventually make proper pcbs etc.

csurf, 2018-02-14:
@mikeller @fiveangle and other devs
what's the consensus on line endings for repo files? Are we doing CRLF or LF? 

I'm trying to get all of my git setups across windows (posh-git), cygwin, and linux consistent.

csurf, 2018-02-14:
@mikeller 
are you on IRC or some other form of chat? What's the best way to get a hold of you? (github doesn't support PM's, or does it?)

mikeller, 2018-02-15:
Also, I am all for all `LF`.

Also, you might want to look into the Windows subsystem for Linux as an alternative for building in Windows - it seems to be less fiddly than cygwin: https://github.com/betaflight/betaflight/wiki/Building-Betaflight-on-Windows,-the-Linux-subsystem-way

csurf, 2018-02-15:
@mikeller cool, i will try to get on slack. I don't personally like or use IRC much either, but it was the only thing available for openLRS at the time. I'm open to using whatever works, but don't currently know of any good options.

line endings: ok, I'll go ahead and convert everything to LF.

WSL: I actually started playing with it this past weekend, but I'm still trying to get the hang of it. It also actually seems a bit buggy in some ways compared to cygwin (which I also hardly use). I'm trying to figure out which will work best for use with git on windows (compared to posh-git which is slow and a bit strange in some aspects).

atomiclama, 2018-02-15:
@csurf @mikeller 
As mikeller is already active on slack why not set up a channel for openlrsng over on betaflight slack? Convenient place for discussions.

mikeller, 2018-02-15:
@atomiclama: We could do that, or if we get more traffic we could register our own free Slack organisation. The Betaflight Slack is constantly pushing the boundaries of being a free organisation in Slack, with only about 2 week's worth of message retention, and constantly overflowing file storage.

atomiclama, 2018-02-15:
Hmmm ok forgot about the limits of slack, thought it might make life easier for you.

Discord?

fiveangle, 2018-02-15:
If not already evident, I’m a big fan of platformio build chain due to its cross platform nature (since I use all three main OSes) and the widest support for MCUs, but everyone here using different build chains is probably a good thing actually.

platformio is tested via CI so that’s covered but if others want to try it I’ll prioritize putting together a quick start guide for what I feel is the easiest setup (it’s toward the end of my todo list right now).

mikeller, 2018-02-16:
@fiveangle: Yes please for platformio. One of the problems we currently have is the lack of a consistent way to do builds.