openLRSng/openLRSng #159: Spektrum CrossFire "Hack" support? CRFS as Tx input?

JulianGoesPro, 2018-01-14:
With Spektrum releasing the CorssFire hack to output CRSF from a DX9 (what the iX12 does natively) I am looking into doing this and digging out my LRS (which I put to sleep since PPM out on the Tx side was just not cutting it for my racing quads).

Is there the goal to implement CRSF support for the Tx module so we Spektrum DX9 and iX12 users can use the CRSF out of your transmitters to get low latency LRS :)

Thanks a lot!

mikeller, 2018-01-14:
Changing the serial input / output protocol of the openLRSng modules will not change the fact that the maximum speed that the RF modules used in openLRSng can support is slow compared to the LoRa technology used in Crossfire hardware. Since this is the limiting factor for the update rate in openLRSng, even when using CRSF the update rate will not be higher than what it is now when using the serial protocols that are already supported (Multiprotocol, SUMD, SBus).

JulianGoesPro, 2018-01-14:
Totally understand that but are you saying the latency is so high going from PPM to CRSF wont be noticeable? I am less interested in super high update rates, just want to avoid that painfully slow PPM output from the trainerport of Spektrum raidios and since the only other protocol they support is/will be CRSF it seams like a nice way to reduce latency, I mean it’s not like the thing that’s causing the biggest latency decides it 100%, latency is something that adds up after all :)

JulianGoesPro, 2018-01-15:
All they do is PPM out, despite the DX9 and the iX12, with are adding CRSF support and if I am not mistaking right the QX7 was kind of too slow as well (bi-directional that was) so CRSF can run slower, but you are the expert :) If you say it won't work I will believe you... I will cry a little... but I will believe you ;)

mikeller, 2018-01-15:
> All they do is PPM out, despite the DX9 and the iX12,

That is a bit sad indeed. It might be a squeeze to add support for CRSF onto the MCU used in openLRSng, and it will probably require a custom firmware with the other protocols removed. But if CRSF is the only digital protocol that Spektrum hardware supports then this will probably be the only way to make it work with less delay than PPM.

JulianGoesPro, 2018-01-15:
Yeah, my thought... so you are saying it's not worth the trouble? :)

mikeller, 2018-01-15:
No. I am saying that it will probably be **a lot** of trouble. But if somebody wants to put in the effort it will be a useful addition - not having any Spektrum gear, I am simply not in a position to test this with what is effectively the target application.

csurf, 2018-01-15:
I think it might be worth a try...

csurf, 2018-01-15:
I also don't have any spectrum gear to test with, but I'd be willing to look into getting this setup.

Does openTX support CSRF output?

mikeller, 2018-01-15:
> Does openTX support CSRF output?

It does. But there are already differences in what the different FrSky TX models support, so I am not sure if making it work on OpenTX (where it is not needed) but ultimately targeting Spektrum hardware is a good approach.

csurf, 2018-01-15:
I'm not much of an AVR expert, but from the looks of it, this might be tougher than I thought. 400kbs baud appears to be way too fast for some kind of software serial implementation (might be wrong, but this is what I gathered). Hardware UART would be the only practical way to do it. 

I'm seeing a few things online saying that it is technically possible to setup a one-wire serial configuration by tieing the the TX and RX lines together using some resistors and also code in order to avoid clashes on the line. However, that would leave the issue of signal inversion (CSRF uses inverted serial). I'm thinking one way w would be to use some kind of small, external adapter/inverter board (with transistors?) that users could DIY and place between the radio and olrs tx module.

@JulianGoesPro, would this be an option for you?

JulianGoesPro, 2018-01-15:
I am surprised to hear CSRF is inverted, are you sure? The "hack" on the DX9 to get it working is just to solder a wire to an empty Tx pad (it's not using the bi-directional Rx combo, it will only "send" anyway) , while the iX12 has the "plug" for digital out (CRSF), the DX9 needs to be "modded" and that's why I am quite sure there is no inverter since its only a wire that we need to "add" :)

mikeller, 2018-01-15:
> I'm not much of an AVR expert, but from the looks of it, this might be tougher than I thought. 400kbs baud appears to be way too fast for some kind of software serial implementation (might be wrong, but this is what I gathered). Hardware UART would be the only practical way to do it.

I've experimented with software serial on openLRSng hardware, and even at 100 kbps (SBus, Multiprotocol) it does not work reliably. Also, sending data takes up a lot of CPU time, so supporting a 'chatty' protocol like CRSF is probably out of the question.

400k is even faster than what the MCUs used in openLRSng hardware can support with hardware serial. 115200 seems to be the maximum there.

As for 'making' an inverted bidirectional serial out of RX and TX of a hardware serial port, that's the easy part, all it requires is two inverters (can be had in one SMD component) and a diode.

mikeller, 2018-01-15:
@JulianGoesPro: The FrSky modules send telemetry on the JR module bay pins inverted, so it's likely that the Spektrum TX default to inverted for this pin.

csurf, 2018-01-15:
@mikeller 
Just as a quick fyi... page 243 of the atmega328 datasheet appears to state that it should be able to run bauds as high as 1Mbps with 0% error @ 16MHz. However, I have no idea if that's feasible in a real-world application, or what kind of load that would put the MCU under. I've never tried running a baud this high on the hardware serial port with an AVR. My 3D printer w/ atmega2560 seems to run fine at 250k baud, so maybe it's possible??? dunno...

@JulianGoesPro, so you're saying that you'd only need to receive and not necessarily TX telemetry back to the radio? That might make things a bit simpler (assuming it's even possible to run the 400kbps baud & invert the signal).

csurf, 2018-01-15:
@mikeller 
FYI, I found this a bit interesting...
https://arduino.stackexchange.com/questions/296/how-high-of-a-baud-rate-can-i-go-without-errors

mikeller, 2018-01-15:
@csurf: That does look interesting. 250kbps and 500kbps are supported, 400kbps isn't. :-(

csurf, 2018-01-16:
this bit rate calculator makes it seems that it should be possible...
http://wormfood.net/avrbaudcalc.php?bitrate=400k&clock=16&databits=8

would need testing though...

JulianGoesPro, 2018-01-16:
@mikeller Isn't the hardware hack for the QX7 an inverter? I have no QX7 but know it's hardware is limiting CRSF and that's why Trappy Made those inverter boards for QX7 users, not 100% sure though.

@csurf yeah, the Firmware Spektrum has for both the DX9 and iX12 does not support any telemetry from extern modules, so no need for that with Spektrum.

Thanks a lot guys! :)

mikeller, 2018-01-16:
@JulianGoesPro: The TBS 'hack' for the QX7 is a replacement for the built-in inverter, because it is not able to support 400kbps.

geofrancis, 2018-01-16:
yes but doesent crossfire it also work at 100kbps? im sure the module still worked in the qx7 but at a lower speed.

mikeller, 2018-01-16:
@geofrancis: Yes, that's true. That would be an option for openLRSng too (if there is a way to set it with Spektrum gear).

JulianGoesPro, 2018-01-17:
I will try to find out if Spektrum supports both 400kbps and 100kbps though I would think if they only do "control" without TLM the will at least try to shoot for 400kbps and doing that, why implement 100kbps as well then :)

Thanks again guys!

mikeller, 2018-01-17:
@JulianGoesPro: Not sure if 'without TLM' actually means that CRSF can work with a unidirectional serial port - as far as I know the protocol requires bidirectional communication in any case.

JulianGoesPro, 2018-01-18:
@mikeller I am quite sure the DX9 does not have the hardware to offer bidirectional communication, it's only a Tx pin that you solder the "wire" to, so I am quite confident it's working without, but will check and report back about that as well.

Thanks a lot :)

PS: I think last time I asked about TLM support for crossfire, Andy from Spektrum told me the iX12 would have the hardware but the DX9 doesn't... and both won't offer TLM because of... I guess they are too lazy :-P (or think it's not important right now since the DX9 won't support it anyways...)

JulianGoesPro, 2018-01-18:
"Spektrum" answered:

Test versions support 1 & 2, which are 200k and 400k. There is no support for 100k. What will be supported when finally released will depend on how testing goes.
It uses only a tx pin, not an rx pin.
Andy

https://www.rcgroups.com/forums/showpost.php?p=39006344&postcount=53762

JulianGoesPro, 2018-01-20:
Forgot to link that back then:

https://www.rcgroups.com/forums/showthread.php?2727822-New-Spektrum-DSMX-Quad-Race-Serial-Receiver-with-Telemetry-(SPM4649T)/page142

JulianGoesPro, 2018-02-04:
Quick update :)

The Beta FW is live, the DX9 now outputs CRSF <3

[image]
[image]

csurf, 2018-02-12:
@JulianGoesPro 
it might be a while until anyone can get to this. I'm trying to get the TX code cleaned up first, and then I or someone else would have to get started trying to port the CRSF protocol over, and then we'd need to see about trying to get the serial port to run @200k. You'd have to check with Spektrum to see if the signal is inverted (guessing it is), and, if so, buy or DIY your own inverter...

Are you sure you still wanna try this??? hehe :P

JulianGoesPro, 2018-02-14:
@csurf I am so gonna hack my DX9 if there is a reasonable chance of the CRSF implementation working... I mean... I can confirm with my Taranis if CRSF is "understood" by my Wolfblox/Orange Tx Module and since I have seen plenty of Spektrum pilots flying CrossFire with the "hack/mod" I have no big doubts why it shouldn't work with openLRSng in the end :)

...yes I would be willing to buy a inverter (are you talking about that thing the QX7 guys are buying for theirs? I thought that only applies to the TLM part?)

mikeller, 2018-02-15:
@JulianGoesPro: You can easily make up an inverter for this from a spare transistor.

csurf, 2018-02-15:
@JulianGoesPro 
banggood used to sell little DIY inverter modules for cheap. I can't seem to find them now, but maybe they're buried in their site somewhere. Here's one from dx.com:
http://www.dx.com/p/signal-inverter-for-naze32-flight-control-sbus-receiver-black-429018?tc=USD&ta=US

BG also sells a prebuilt 'deluxe' version, but it costs a tiny bit more:
https://www.banggood.com/NAZE32-Flight-Controller-SBUS-Signal-Inverter-for-RC-Multicopters-p-1034503.html?rmmds=search&cur_warehouse=CN

You can search around on the Chinese vendor sites for something similar, or just DIY with transistor & resistors.

JulianGoesPro, 2018-02-17:
@mikeller I feel confident swapping out chips with a hot air gun but I am super bad with building circuits from the ground up :)

@csurf I ordered 3 of the prebuild ones so I can strip one down and solder it in directly and if I kill it I still have spares :-P

[image]

mikerc2017, 2018-05-17:
i can use a Satelite for input signal modul TX ? which is it ? it fast ?

TheRealOrange, 2019-04-27:
I know spektrum has now released support for CRSF on the DX9, but it’s send only. Is it possible for the DX9 to receive CRSF telemetry as well, perhaps by soldering a wire to a rx pad on the radio?

JulianGoesPro, 2019-05-20:
I was told that the DX9 has only one UART input and that is used for the wireless trainer and they are not willing to allow us to "rip that out" just to get a receive pad for CRSF TLM... :(

fiveangle, 2019-05-20:
You were told by whom ?

https://www.horizonhobby.com/pdf/SPMR9910_Crossfire_Instruction.pdf

TheRealOrange, 2019-05-20:
yeah but that is only Tx and it cannot receive crossfire telemetry which requires an Rx port