openLRSng/openLRSng #160: A use case for encryption of the radio link

podhrmic, 2018-01-17:
Hello,

first of all, I would like to say that I am very excited to see all the recent activity on openLRSng!

The reason I am opening this issue is that a couple of months ago I attempted to add optional encryption to the openLRSng radios. I developed the algorithm for a key-exchange and ongoing communication and made an Arduino prototype. I never got to actually make a PR to the openLRSng because:
1. the openLRSng code needed significant refactoring both in terms of functionality (like a more structured handling of incoming/outgoing messages) and saving space (moving some stuff to EEPROM and some on flash)
2. the lack of feedback from the openLRSng community (like help with debugging etc)

So given the new development, I am curious if there is interest in encrypted radio link and perhaps a continuation of this work? I know encryption was discussed before (like in #85 ), but I am not sure how many users still desire such a feature.

Regards

csurf, 2018-01-17:
you could setup a pull request just so that others can take a look at it. I'd be interested in check out the refactoring and code savings. Could the encryption be optional/disabled? How much overhead is added due to it?

mikeller, 2018-01-17:
@podhrmic:

A couple of points about adding encryption to the radio link:
- what is your use case for this;
- taking into consideration the limited processing power of the atmega MCUs used in openLRSng hardware, the options for encryption algorithms will probably be limited to ineffective ones;
- in most jurisdictions, using the ham bands for encrypted transmissions is against the law.

podhrmic, 2018-01-18:
@mikeller the main reason is to avoid this: http://www.businessinsider.com/icarus-gadget-hijack-almost-any-drone-mid-flight-2016-10 

It turned out that the processing time wasn't the bottleneck (end-to-end the processing time was < 4ms), but the memory requirements (around 400 extra bytes, which is challenging on the 1kb RAM that arduino has.

I am not familiar with any legal restrictions or requirements so I can't comment on that.

@csurf The ecnryption algorithm (including how to make-do without having a real random number generator) is summarized in this article:
http://ieeexplore.ieee.org/document/7991460/

Let me know if you don't have access, I can send you a private copy.

mikeller, 2018-01-18:
@podhrmic:

> @mikeller the main reason is to avoid this: http://www.businessinsider.com/icarus-gadget-hijack-almost-any-drone-mid-flight-2016-10

 Did this article mention anything of openLRSng or 433 MHz technology? All I could see mentioned was 2.4 GHz technology, and more specifically the DSM protocol.

> I am not familiar with any legal restrictions or requirements so I can't comment on that.

Can I recommend you familiarise yourself with the legal restrictions on what users are allowed to do in the band used by openLRSng? I am not sure if adding something that will be illegal to use outside of lab conditions makes much sense to be added to released firmware.

geofrancis, 2018-01-18:
Your assuming ham bands are used everywhere, in the UK we have a dedicated 458.5-459.5 for RC outside the ham bands that can be used with encryption.

mikeller, 2018-01-18:
@geofrancis: That's good to know. That's why I asked @podhrmic to make a case for how his proposed change can be used without breaking the law.

podhrmic, 2018-01-18:
@mikeller the article shows how easy it is to hack a non-trivial protocol such as DSM. OpenLRSng has the same vulnerability (and so do all the other radios out there). All you really need is to listen to the spectrum to get the right frequency, and then get the receiver ID. Not really a rocket science, but maybe that is not an issue for most users? That is what I am trying to gauge. 

I went through my previous notes - there are other 433MHz modules that offer encryption (like this one http://www.hoperf.com/upload/rf/RFM69CW-V1.1.pdf ). RFM69CW uses AES-128 for example. Judging from the fact that it is a readily available product, I think it is safe to say that at least in the US there is no restriction on encrypted communication at this band. Does it answer your question?

mikeller, 2018-01-20:
@podhrmic:

> @mikeller the article shows how easy it is to hack a non-trivial protocol such as DSM. OpenLRSng has the same vulnerability (and so do all the other radios out there).

Can you outline what risk you actually want to mitigate against by adding link encryption to openLRSng? Are you afraid of somebody taking over control over your craft in order to steal it?

> I went through my previous notes - there are other 433MHz modules that offer encryption (like this one http://www.hoperf.com/upload/rf/RFM69CW-V1.1.pdf ). RFM69CW uses AES-128 for example. Judging [...]

No, it does not. Can you explain how the existence of an encryption capable RF module (that is not useable as is without knowledge of electronical engineering to turn it into a product) that seems to be capable of emitting in a whole range of commercial frequency bands that encryption on ham bands should not be illegal in the USA?

I don't know what country you got your ham radio license in, but in most countries, knowledge of the legal restrictions on the use of ham bands is a mandatory part of license testing. Can I suggest you re-read the respective chapters your your ham operator training manual?

podhrmic, 2018-01-20:
> Can you outline what risk you actually want to mitigate against by adding link encryption to openLRSng? Are you afraid of somebody taking over control over your craft in order to steal it?

Yes, that is exactly the use case. Somebody taking over your craft and either flying away with it or worse deliberately crashing it. 

I think we are talking about two different things here. One thing is obtaining a license for 433MHz, and that should be the responsibility of the end user (and I am not familiar with the process as I am not a pilot). The other thing is an engineering solution for an existing problem (a vulnerability of the openLRSng radio communication) and I can help here.

The radio module I mentioned can be soldered on a regular receiver (like this http://www.hobbiesfly.com/transmitter-receiver/brotronics-4ch-remix-long-range-receiver.html), so I doubt such product would be offered if it was not legal to use. I also never heard about any restrictions of the radio communication encryption.

But I said, I am not a pilot and don't have any radio licence as I always operated with the unlicenced bands, so I can't tell you about the regulations. If you think that the encryption of the radio link is not an issue or it is not interestig at the moment for OpenLRSng, that is OK, I was just pitching in with what I knew.

mikeller, 2018-01-20:
> Yes, that is exactly the use case. Somebody taking over your craft and either flying away with it or worse deliberately crashing it.

For what you outline as 'even worse', your proposed solution is ineffective: openLRSng uses a short list of fixed frequencies. If somebody wanted to crash your craft, it is unlikely that they would go through the effort to actually take control over it, they would just use their transmitter to block all of the used frequencies, thus effectively preventing anybody from controlling your craft.

> But I said, I am not a pilot and don't have any radio licence as I always operated with the unlicenced bands, 

> The radio module I mentioned can be soldered on a regular receiver (like this http://www.hobbiesfly.com/transmitter-receiver/brotronics-4ch-remix-long-range-receiver.html), so I doubt such product would be offered if it was not legal to use.

Your logic is deeply flawed. According to it, it would not be legal to offer blocks of steel for sale, because they can be turned into guns by a gunsmith.
And for the second time: This module can operate on a whole range of commercial frequencies, in addition to the ham bands. There are no restrictions on the use of encryption in commercially licensed bands, just for ham bands.

> I also never heard about any restrictions of the radio communication encryption.

I said it before: How about you go and inform yourself first, before making assumptions.

> But I said, I am not a pilot and don't have any radio licence as I always operated with the unlicenced bands,

Can I then propose that you go and propose your solution for systems that operate in unlicensed bands, and not here? To me it is a bit odd that you are trying to 'improve' something that you don't have a license to use, and when it is suggested that you at least inform yourself about the legal environment that this is operating in, you respond that you don't want to do that.

closed 2018-01-20 by mikeller