openLRSng/openLRSng #161: Different frequencies and power levels for RC and Telemetry

geofrancis, 2018-01-18:
in most places there are different laws for radios,

power levels and frequency you can use on the ground ie in the uk its 100mw on 458.5-459.5 but your not allowed telemetry on this frequency.
power levels on the aircraft for telemetry are generally much lower and on a slightly different frequency usually 10-25mw @433 and at a low duty cycle.

so as a solution to all the different laws i would suggest (if possible)

splitting the RF power settings between the TX and RX, currently it just looks to see that the TX is set to when binding so if you have level 7 on the tx your getting level 7 on the rx.

separate uplink and downlink frequencies, so for me it could be 459 up and 433 down.

csurf, 2018-01-19:
RF power shouldn't be a problem I don't *think.

separate frequencies might be pretty messy and it might require a big overhaul.  I'd have to think it through. I'm not sure if the hop channel sync'ing would work as well (if at all).

mikeller, 2018-01-19:
Re RF power, from what I've read, not all modules actually support reducing the output power, and for others the range of power levels is pretty restricted.

Re separate frequencies, the 459 up and 433 down that you are suggesting will require two separate RF module on TX and RX, since there is no single module that supports both of these frequencies. Is this the only use case that you have got for this, or are there others where disjunct frequency ranges could be an advantage?

csurf, 2018-01-19:
I've always just flown and full TX power w/ no telemetry, so I've never had to look into changing power levels. But, from the little I've gathered, the output power settings don't offer too much in terms of scaling.

yeah, come to think of it, 459 is over what the 433 RFM is rated at (413-453). Not sure if that's a hard limit as to how high the chip can be tuned, or if it's a recommended performance limit (perhaps the chip can be tuned higher? haven't looked into it). S

geofrancis, 2018-01-19:
I have flown for years on 459 on openlrsng, the power output is a little lower but it works fine.

mikeller, 2018-01-20:
> In the uk the legal uhf frequencies are 100mw 458.5-459.5mhz for rc but no telemetry

I always assumed that the restriction on 'telemetry' was applying to all emissions from the airborne model (due to potentially far wider spread as a result of the elevated position of the antenna)...

geofrancis, 2018-01-20:
I think the problem is that many years ago when the regulations were made I dont think they had ever thought that somone would have systems that would do multiple functions.

mikeller, 2018-01-20:
So I guess then that rules out legally using systems like openLRSng that require emission from the model to synch up with the TX on this band...

geofrancis, 2018-01-20:
yes currently my setup is not legal unless I disable telemetry.

mikeller, 2018-01-20:
@geofrancis: That was my question really: Is it legal to use these frequencies if you disable sending of telemetry values and instead send a stream of zeros, or do you have to disable emission of any RF from your craft (which is to my understanding what the intent of the law is).

geofrancis, 2018-01-20:
the rules are for transmitters, so if your transmitting zeros or otherwise i think it will apply.

mikeller, 2018-01-20:
:-(

atomiclama, 2018-01-22:
For the UK
From my point of view, one of the main reasons to go for a UHF system is for the range.  But if you are going for range then you are going to be break the rules.  So you are not going to be that fussed about the frequencies and power. So I would not worry about it too much.

> In the uk the legal uhf frequencies are 100mw 458.5-459.5mhz for rc but no telemetry, for that your allowed on 433 at 10mw, 

As a side note how does TBS crossfire get around the UK allocated frequencies for RC control? @geofrancis you have any idea?

mikeller, 2018-01-22:
@atomiclama:

> But if you are going for range then you are going to be break the rules. So you are not going to be that fussed about the frequencies and power. So I would not worry about it too much.

That sounds like a very lame excuse for not getting a ham radio license that allows you to use more powerful devices without breaking the law.

atomiclama, 2018-01-22:
UK based not sure about the rest of the world.

Just trying to point out that users going long range would be breaking other rules so I doubt they would be bothered about power and frequency limits. 

Also from my understanding a ham radio license would not allow the use of higher power devices. The power levels are set for RC control and telemetry.

mikeller, 2018-01-22:
@atomiclama:

> Just trying to point out that users going long range would be breaking other rules so I doubt they would be bothered about power and frequency limits.

I don't know. What other laws are putting restrictions on the type of long range flying that you want to do?

To me 'I was not able to comply with law A, so I did not bother to put in the effort required to comply with law B' still sounds like a lame excuse - and I would assume that if anybody ever goes to trial for anything related to RC flying, such an attitude will be used against them as evidence of bad character.

> Also from my understanding a ham radio license would not allow the use of higher power devices. The power levels are set for RC control and telemetry.

The 433 - 438 MHz band in the UK seems to be having a power limit of 400W EIRP - if that is not enough for your long range flying, then I suspect there is another problem.
(Also, getting a ham radio license includes getting some knowledge about these regulations, and how to find them, which to me seems to be useful when using experimental / home brew radio equipment anyway.)

geofrancis, 2018-01-22:
Yes my point is there is a dedicated rc band at 459 for RC in the UK, since I don't have fight hams with their huge power output I can get better range compared to 433 due to the lower noise floor.

mikeller, 2018-01-23:
@geofrancis: That's a reasonable choice. My point is that, if you decide to instead use the 433 MHz ham band, I think the appropriate way to go on about this is to get properly licensed as a ham operator:
- it is not hard, most of the knowledge you need you will have gained already from being active in RC flying anyway;
- some other knowledge, like the legal limits on the use of bands, should really be obtained before operating equipment on these bands;
- having a 'I know I am acting outside of the law, but I can't be bothered to take the few steps required to make what I am doing legal' attitude will not make you many friends.

geofrancis, 2018-01-23:
your not allowed to control RC aircraft with ham equipment here. thats why we have dedicated bandwidth. 

the different frequencies is not essential but being able to set power levels independantly is as it effects anyone with a 1w tx and 100mw rx. since lowering the power on a 1w tx will lower the receiver output.

mikeller, 2018-01-23:
@geofrancis:

> your not allowed to control RC aircraft with ham equipment here. 

Do you have a source for this restriction? In most other countries this does not seem to be the case, that's what the whole openLRSng project is based upon.

> being able to set power levels independantly is as it effects anyone with a 1w tx and 100mw rx

Can you explain what exactly you mean by this?

csurf, 2018-01-23:
aside from the conversation about legalities & ethics of this type of setup (I'm neutral in this discussion), here's a possible way for it to be achieved...

looking at the `bind_data` structure in `binding.h`...
```
struct bind_data {
  uint8_t version;
  uint32_t serial_baudrate;
  uint32_t rf_frequency;
  uint32_t rf_magic;
  uint8_t rf_power;
  uint8_t rf_channel_spacing;
  uint8_t hopchannel[MAXHOPS];
  uint8_t modem_params;
  uint8_t flags;
} bind_data;
```
we could add some new variables, perhaps like this...
```
struct bind_data {
  uint8_t version;
  uint32_t serial_baudrate;

// separate this into two frequencies for tx and rx
// these can be set to the same value for standard, non-'frequency jumping' operation
  uint32_t rf_frequency_tx;
  uint32_t rf_frequency_rx;

  uint32_t rf_magic;
  uint8_t rf_power_tx;
  uint8_t rf_power_rx;

 // here, we could add an extra channel spacing variable to vary the rx hop sequence
// or just reuse the same value for both the TX and RX hop sequences
uint8_t rf_channel_spacing; 
// uint8_t rf_channel_spacing_tx;
// uint8_t rf_channel_spacing_rx;

// we could either reuse this array for both tx & rx hop sequences
// or split this into two distinct arrays for tx and rx
  uint8_t hopchannel[MAXHOPS]; 
// uint8_t hopChannel_tx[MAXHOPS];
// uint8_t hopChannel_rx[MAXHOPS];

  uint8_t modem_params;
  uint8_t flags;
} bind_data;
```
The comments I made above are referring to the fact that we'd either need to add separate variables to describe the hop channel sequence for the RX, or reuse those that are already set for the TX yet using a different base frequency. 

The `hopChannel[]` array simply holds indexes pointing to hop channels. You can calculate the individual hop channel frequencies like this:
```
hop frequency = rf_frequency + (hop_channel_index * rf_channel_spacing)
```
where `hop_channel_index` is the value stored in the `hopChannel[]` array. 

So, in theory, we could have the TX and RX both use the same 'hop sequences' yet with different base RF frequencies (via the `rf_frequency_tx/rx` variables).

As far as sync'ing the hops goes, we'd need to retune the RF module on (both the TX and RX) to the RX's hop frequency (rf_frequency_rx) in the time period between when the TX sends/RX receives an uplink packet, and RX sends/TX receives a downlink telemetry packet... 

I *think that this should work ok, but we'd need to test and make sure the timing doesn't get all screwed up.

geofrancis, 2018-01-23:
@mikeller currently when you bind a transmitter to a recevier, the recevier just gets a copies the power output of the TX. so if you have a 1W transmitter and you reduce the power level down from 7 to 5 it will also drop the 100mw receiver down too.

The difference is in most countries you are required to use a ham licence that few people bother with so fly illegally but in the uk we have unlicensed spectrum at 459 we can use without a ham licence so it can be legal without a licence.

mikeller, 2018-01-23:
@csurf: Good suggestion. I think we do need separate hop lists, so that TX and RX frequencies can be spread out into bands of different width. (This might be useful even if they are used inside the same band, to be able to adapt the TX and RX bands to different power levels of the emitter.

@geofrancis 
> currently when you bind a transmitter to a recevier, the recevier just gets a copies the power output of the TX. so if you have a 1W transmitter and you reduce the power level down from 7 to 5 it will also drop the 100mw receiver down too.

Ah, now I understand. Making these settings independent sounds reasonable.

> The difference is in most countries you are required to use a ham licence that few people bother with so fly illegally but in the uk we have unlicensed spectrum at 459 we can use without a ham licence so it can be legal without a licence.

This does not answer my question as to why it would be illegal for licensed ham operators in the UK to use ham frequencies to control RC craft. Breaking the law just out of convenience and laziness is, and 'I do it because others do it' is just as lame.

geofrancis, 2018-01-23:
I cant remember off the top of my head specifically why you cant use 433, its possible that you can with correct licences, but I know the only other system I can think thats legal in the uk is ezuhf, it has a UK firmware that sets it too 459@100mw.

mikeller, 2018-01-23:
@geofrancis:

 > I cant remember off the top of my head specifically why you cant use 433, its possible that you can with correct licences, 

Sounds like a good reason to get a license then. :-)

geofrancis, 2018-01-23:
but 433 is very noisy, 459 is dedicated to RC

fiveangle, 2018-01-23:
> I cant remember off the top of my head specifically why you cant use 433

Looks like a license in the UK doesn't gain you additional digital transmission frequencies closer to the 433 range at all due to channel ident requirements.
> 459 is dedicated to RC and unlicenced.

…and surely just as cluttered, as it's apparently not dedicated to RC at all, but all 458.5-459.5 SRDs (garage door openers, rc control, projector screen remotes, X10, IoT, etc, etc, etc): https://www.ofcom.org.uk/spectrum/radio-spectrum-and-the-law/licence-exempt-radio-use/licence-exempt-devices/short-range-devices-information

@geofrancis - while being legal should be a primary goal, I think anyone attempting to perform long range control of aircraft in the UK should review this short thread: http://fpvlab.com/forums/showthread.php?8714-433-or-459Mhz-EzUHF-and-a-2-4g-VTx-question

The barriers to overcome are quite conflicting, but in the end, I think this sage advice in that thread is probably the most apropos:
> _ukneek_:  Dont forget that a UK ham licence does not give you any additional benifits when it comes to aerial video transmissions in terms of frequency or power. UK legal: 2.4gHz @ 10mW and 5.8gHz @ 25mW. Still, if your going to go illegal on one thing you may as not worry. Common sense and a low profile go a long way.

fiveangle, 2018-01-23:
> Sounds like a good reason to get a license then. :-)

Even though a license won't gain you anything in the UK, I highly suggest campaigning for your license as you gain a lot more knowledge than you would expect; at least I did when doing the Amateur Extra here in the US.

mikeller, 2018-01-23:
@fiveangle: Couldn't agree more. Being a (these days very sporadic) ham radio operator myself, I am pissed off with ignorant and untrained users of the ham bands who take the fun out of using VHF/UHF repeaters by programming their illegal devices to have an overlap with the repeaters' input or output frequencies (be it with openLRSng RX/TX or other devices they are not licensed to use), ruining it for everybody. Getting a ham license will at least make sure the user knows where to find information on permanently allocated frequencies for the area they are operating in, and how to not interfere with them.

> Common sense and a low profile go a long way.

That's exactly that.

atomiclama, 2018-01-23:
Oh that's kicked off a bit ;-)
Got me brain cells interested so I had a rummage around. 
The rules for rc control and telemetry come from here: -
https://www.ofcom.org.uk/spectrum/radio-spectrum-and-the-law/licence-exempt-radio-use/licence-exempt-devices/ofw311?SQ_VARIATION_73584=0

> This does not answer my question as to why it would be illegal for licensed ham operators in the UK to use ham frequencies to control RC craft. Breaking the law just out of convenience and laziness is, and 'I do it because others do it' is just as lame.

with a section about Amateur radio license.

> 3. Can I use an Amateur Radio licence for model control including airborne use?

with details on power and frequency here: -
https://www.ofcom.org.uk/__data/assets/pdf_file/0028/84970/ir-2030-july-2017.pdf

> Still, if your going to go illegal on one thing you may as not worry. Common sense and a low profile go a long way.

This is what I'm trying to get across and that is the sentiment that I get from a lot of FPV flyers.

Personally I like to know what rules I'm breaking and by how much when I go flying so I can do my own risk assessment.

mikeller, 2018-01-23:
> Amateur Radio licence expressly prohibits use in any aircraft or airborne vehicle. 

The UK seems to be a lot more restrictive in most other countries - airborne comunications are commonly used in ham radio, and there is even the standardised callsign suffix '-airborne' to indicate that condition.

geofrancis, 2018-01-23:
I dont use analog video in any of my aircraft, i use a 2.4ghz 100 mw digital system that falls under a different classification specifically "high bandwidth data link" this is the same rules lightbridge works under so you can use far more than the 10mw you get under analog video systems.

mikeller, 2018-01-23:
@geofrancis: 2.4 / 5.8 GHz is a different kind of beast altogether. Unlike the ham bands, these bands are allocated for a host of wide spectrum / high(-ish) power devices like WiFi, so an RC protocol will have to contend for spectrum with these, and there is not the problem of interfering with small bandwidth / fixed allocation services like there is in the 433 MHz band.

csurf, 2018-01-23:
so, should we go ahead with trying to setup a prototype of this feature? or is it still in debate?

mikeller, 2018-01-23:
I think this does make sense as functionality in general, even outside of the scope of enabling configurations that spread across bands with different licensing terms, so implementing it will be worthwhile.

mikeller, 2018-01-23:
@geofrancis: Yeah, the LoRa technology that these are using looks very promising, both for long range but also for general purpose RC. It combines a long range with a high data rate, and does so in a performant way, thanks to a lot of research money having been sunk into it.
Unfortunately, acceptance and band allocation is still spotty internationally, and there are different bands allocated for it in different countries: https://www.thethingsnetwork.org/wiki/LoRaWAN/Frequencies/By-Country

csurf, 2018-01-24:
I still don't know much about LoRa, but I was under the impression that the data rate wasn't high enough for RC (this is based only on some stuff I glanced at on RCG from user srnet). From what I gathered, increasing the data rate significantly lowered the range, so it was best for low transmission rates @ long distances.

Apparently, LoRa modules (such as the hopeRF modules) also do FSK. I'd be curious to see if their sensitivity for FSK is any better than the older RFM modules we're currently using. Once cool idea I thought about was being able to switch between LoRa and FSK based on conditions. I.e., FSK for 'normal' RC operation, and then switch to LoRa for something of a reduced rate 'emergency' or 'long-range' mode that can be switched to on failsafe, for attempting to regain control or for finder beacon.

atomiclama, 2018-01-24:
LoRa
Everything I have read on this points to it not being usable for RC control, low data rate, low update rate.
But very good for telemetry. 

868 band. 
Well for the UK that document I linked to above, there is a section for non specific short range radio which I think the FrSky R9 and crossfire fall under is limited to 25mW, this is  a European harmonised band. I have seen posts from trappy to this effect as well. Might be able to find them again.

mikeller, 2018-01-24:
@atomiclama:

> Everything I have read on this points to it not being usable for RC control, low data rate, low update rate.

I think both TBS and FrSky beg to differ on this point. ;-)

csurf, 2018-01-24:
> I think both TBS and FrSky beg to differ on this point. ;-)
yeah, that's what I don't really understand about LoRa... Many claims about it "not being practical for RC", but yet, apparently, crossfire (and FrSky?) use it successfully... Would be great to know how they manage to achieve this...

atomiclama, 2018-01-24:
> Would be great to know how they manage to achieve this...

Yep agree, but what are the chances :-)
> I think both TBS and FrSky beg to differ 

I don't doubt their performance, didn't realise they were using LoRa though.

mikeller, 2018-01-24:
> I don't doubt their performance, didn't realise they were using LoRa though.

At least for TBS, there are claims to this end on the internet:

http://fpvlab.com/forums/showthread.php?42097-TBS-CROSSFIRE-Next-Generation-Long-Range-R-C-System&s=76427c9f2a9107a65c8a1296fce9b64c&p=762346&viewfull=1#post762346

For FrSky, I am making an educated guess, based on the fact that their R9 series operates in the LoRa frequency range.

geofrancis, 2018-01-24:
there is another system called QLRS its using the same modules as tbs and frsky, what what i have read about it the modules are limited to 19200 baud. so to get mavlink telemetry and rc they use it as a pure telemetry link and inject mavlink RC override packets over it.

fiveangle, 2018-01-24:
@geofrancis - I think you are confusing LoRa RF _module_ with LoRa _modulation_.  These modules are not locked to LoRa modulation.  In FSK modulation the Microchip 900Mhz parts claim 300kbps capability.  LoRa modulation is of little use for the purposes we are interested in, I would imagine.

csurf, 2018-01-24:
@fiveangle
that is what I was referring to in my earlier post; the fact that the Lora modules can also do FSK modulation. Seeing as how they obviously use a more modern RF IC, I figured that perhaps they could outperform the RFM22 even when used as regular FSK module. I think this would be worth testing...

fiveangle, 2018-01-24:
@csurf - I guess assumed everyone was thinking about this, but agree 100%

Just comparing specs and design improvements I  would suspect it a given that they will outperform RFM22 in probably all respects.  Max data rate on RFM22 is quoted as 128kbps vs 300kbps for a LoRa-targeted module like RN2483:
http://ww1.microchip.com/downloads/en/DeviceDoc/40001784F.pdf

And you may be aware, but others have already repurposed these types of modules for LRS use:
https://www.rcgroups.com/forums/showthread.php?2837542-QCZEK-LRS-DIY-433MHz-1W-(30dBm)-LORA-RC-LINK-with-telemetry
and the Qczek originator claims with the SX1278 module, 250k data rate is achieved (and "500k in race mode" whatever that is): http://qczek.beyondrc.com/qczek-lrs-433mhz-1w-lora-rc-link/qczek-lrs-technical-specyfication/

Datasheet for SX1278:
http://www.cdebyte.com/en/downpdf.aspx?id=108

Datasheet for Si443x (250kbps claimed in ds):
https://www.silabs.com/documents/public/data-sheets/Si4430-31-32.pdf

Some claim LoRa modules cannot match performance of the higher data rate RFM (but clearly states only in theory): https://www.rcgroups.com/forums/showpost.php?p=38780621&postcount=12566
but I suspect they mean "when running in LoRa modulation" and aren't considering these newer core modules running FSK vs RFM FSK.  Or if they are, perhaps they suspect the RX sensitivity is off due to being optimized for LoRa modulation, giving the RFM modules an edge ?  Regardless, I don't think there is a lot of empirical comparisons so certainly seems worth investigating, especially for the easy access to 900MHz band solutions.

geofrancis, 2018-01-24:
it was my understanding that to do the higher speeds it switched to fsk mode.

fiveangle, 2018-01-24:
LoRa and FSK can both be modulated at 19.2 and below.  For the datasheets I've reviewed, every one of them states LoRa modulation is not possible above 19.2.  Given that LoRa is proprietary, I suspect every implementation must disallow LoRa modulation above 19.2 to retain the ability to sell the modules as LoRa compatible.

csurf, 2018-01-24:
@fiveangle or anyone else
would you happen to have any links/resources on how to do FSK with the semtech SX1231 (hopeRF RFM69W module RF IC)? I've looked around but haven't been able to find anything that's very useful (aside from the datasheet).

csurf, 2018-01-24:
let's move discussion on RF modules to #168, so we can get back to discussion on 'frequency splitting' feature (or whatever we decide to call it).