openLRSng/openLRSng #162: USB issue with 32u4 builds with latest Arduino versions

csurf, 2018-01-20:
There's a problem with how Arduino versions post 1.06 are compiling code for the 'leonardo' 32u4 target that causes the chrome API to fail to detect the 32u4 as a valid USB device. The VID and PID look the same as before, but, for some reason, chrome can't pick it up. Currently, the only work-around is to use Arduino 1.06, but we should probably try to figure out exactly why this is happening.

FYI, this is for hardware type 6, the @DTFUHF deluxe 1W TX...

@mikeller, do you have any experience with the 32u4?

mikeller, 2018-01-20:
No, I don't. Have you tried on linux? `dmesg` output is usually quite verbose for USB device detection.

csurf, 2018-01-20:
I did some simple testing both on linux and windows.
haven't looked at dmesg output, just lsusb output, and things looked pretty much the same between the older Arduino 1.06 builds and the newer 1.6/1.8 builds. I'll try to take a closer look at it on linux though.

Timvrakas, 2018-01-20:
I had a similar USB radio prototype last year that used the 32u4 and I can confirm the USB can behave oddly. I ended up widening the USB traces for better impedance matching, but even then there were some issues.

csurf, 2018-01-20:
@Timvrakas 
interesting find...

In this case, it definitely appears to be a software issue. Something within the Arduino USB code for the Leonardo changed between the legacy 1.06 version and the newer versions that's causing issues with chrome.

yes, the 32u4 is very finicky when it comes to USB compatibility. Unfortunately, there's still a lot of DTF UHF units that are running it, so we must continue to support it.

csurf, 2018-01-21:
@mikeller 
ok, it seems that the newer 1.6/1.8 Arduino code *does work on linux. The chrome API doesn't have any issue with detecting the 32u4 as a USB device with builds based on either the newer or older libraries.

Firmware builds based on the 1.6/1.8 Arduino libraries don't work with the chrome USB API on the Windows 10 machines I've tested. The devices show up as valid COM ports within Windows / Device Manager, and even within chrome; chrome is able to open the port and communicate with it. 

However, for some reason, chrome can't detect the device as being a 32u4 based on the PID and VID. The Configurator code relies on being able to detect the 32u4 using `chrome.usb.getdevices()` call in order to initiate the reboot sequence that's specific to the 32u4.

The are some noticeable differences in the Arduino USB core code between the 1.0.6 and 1.6+ versions. The main thing I can see is that the older lib's include HID support, which sets up the Leonardo as a USB HID device. The built-in HID functionality was removed in the newer libs. I'm guessing this might be clue as to why chrome can't pick it up...

mikeller, 2018-01-21:
Setting the device up as a USB HID device if it's nothing like an input device sounds like a very bad idea in any case, and should probably be removed.

Just one thing I can think of that is completely broken for fake USB HID devices is that they are unusable in virtualised operating systems (because forwarding them to the guest OS as raw device isn't supported).

csurf, 2018-01-21:
I agree that the HID functionality is not necessary, but perhaps just the fact that it's presented to the Windows OS as a HID makes chrome able to use it? not sure... 

In any case, the current working builds already have the HID support built in, as the 1.0.6 libs already have it integrated into the USB code for the Leonardo. 

Obviously, if there's a way we can remove HID and still get the newer libraries to work, then that would be preferable...

I'm also looking at the Arduino USB code and seeing if maybe there might be an issue with how the USB descriptor data is setup and sent to the OS. Maybe there's an issue with how the new libs present the device to Windows that causes issues with chrome.

FYI, the Windows & linux testing I've done has been with actual hardware installs and not VM's; I've also found that USB hubs can sometimes cause issues, so I've had to test with USB ports that are directly on the machine. It seems the 32u4 chips are quite a headache in terms of USB support...

mikeller, 2018-01-21:
> FYI, the Windows & linux testing I've done has been with actual hardware installs and not VM's; I've also found that USB hubs can sometimes cause issues, so I've had to test with USB ports that are directly on the machine. It seems the 32u4 chips are quite a headache in terms of USB support...

I've seen some USB hubs that are emulating their own set of (virtual) HID devices, and then forwarding the input from any actual HID devices plugged into them through those HID devices. This will be another use case where application support for devices registering themselves as HID devices will break...

csurf, 2018-01-22:
It appears that chrome does, in fact, rely on the HID interface for the `chrome.usb.getdevices()` system call to work. 

The good news is that I've found a work-around, which is to instead enumerate the serial devices using `chrome.serial.getdevices()` in order to check & see if the current serial port is an ATMega32u4 device (based on PID and VID). After testing, this has proven to work reliably on all platforms and for both the older and newer Arduino libs. See the above PR for more details...

Will close after allowing some time for final testing and merging the PR.

mikeller, 2018-01-22:
Good work, @csurf!

closed 2018-01-23 by csurf