openLRSng/openLRSng #163: Start-up timing issue for 32u4 TX

csurf, 2018-01-20:
it seems there's a timing issue going on with the 32u4 code that causes configurator to fail to pickup the 'OpenLRS TX starting' message on boot. This causes Configurator to wait for the message & ultimately time out / disconnect.

Currently, I'm playing with the TX code and how&when it initializes serial & prints the startup message (basically, just trying to add a delay before the message is printed). I'm also messing around with the code on the Configurator side in order to get a safe timing setup that won't fail and allow them to sync up...

@mikeller I know you made a lot of changes to TX.h within the past year when you added MULTI support. I'm not sure if somehow something shifted the timing of the TX module start-up that possibly threw off Configurator? 

FYI, overall, perhaps we should rethink the way it currently works. Perhaps it's not such a great idea to have Configurator rely on catching & parsing the start-up message from the serial buffer. Maybe there's a better, more reliable way to do this?

mikeller, 2018-01-20:
@csurf: Yes, I did quite a bit of reordering of the TX startup code. This was required in order for MULTI to work in such a way that the first MULTI frame can be received before the module starts sending. Because with MULTI, the frame contains configuration data such as the profile to be used, and the RF power level, this makes sure that no emission happens on unwanted frequencies or at wrong power levels.

I tested with the configurator after all of my changes, and confirmed that it's still working. But that is obviously just one case, there might be problems with different hardware configurations.

I think the way it is currently done with the configurator waiting for the start-up message is not ideal, but I don't know that there is much improvement possible over it with the currently supported hardware. Most of the modules seem to be having only one serial port, so this port has to be used for configuration and for serial RC data transfer (which in my opinion is preferrable over PPM). Once the RC data transfer has started I think it is the right thing to do to not allow the module to enter the configuration mode - having this happen accidentally during flight (think through a serial transmission error) would most likely have disastrous consequences.

csurf, 2018-01-20:
I might have a simple hack/fix...

Adding an 850ms delay at the very beginning of the `setup()` in TX.h seems to stabilize things. This more/less mimics the delay caused by the bootloader on a regular ATMega (optiboot, ~1sec), and gives Configurator enough time to catch up and grab the OpenLRS start-up start-up message after opening the port. Otherwise, the start-up message will be spit out much too fast on the serial line, almost instantly after the port is opened, and Configurator will miss it.

csurf, 2018-01-22:
fixed by #164

closed 2018-01-22 by csurf