openLRSng/openLRSng #166: Diversity receiver

madhunm, 2018-01-24:
Is the diversity receiver being revived as well?

csurf, 2018-01-24:
The diversity code is still functional.

However, this is just the code repo, for the development of the firmware code. Hardware is completely separate. You'll have to look around online to see if anyone still sells the diversity RX's, or make one yourself...

madhunm, 2018-01-24:
@csurf i made the broversity receiver hardware. However, the antenna switch used in that design is obsolete. If there is interest in the hardware, I can make a new version. Let me know.

fiveangle, 2018-01-24:
What would be really exciting is full master/slave RX diversity that would not be tied to only very specific HW.   And since we're tossing coins in the wishing well, to do it with a serial RX as small as these again 😄: https://github.com/Brotronics/subMicroLRS

csurf, 2018-01-24:
@fiveangle 
we actually just recently commented out the 'slave mode' code since we figured that no one was actually using it. You can still enable it by modifying the openLRSng.ino file and uncommenting the `define ENABLE_SLAVE_MODE` directive.

@madhunm 
what do you mean when you say that the antenna switch is obsolete? So, are you saying that you actually designed the diversity receiver? I had assumed that user 'ReadError'had designed it...

madhunm, 2018-01-24:
@csurf  I am not the original designer. It was designed by brotronics and licenced to someone else for manufacture.

What I'm saying is that there's a hardware antenna switch that's obsolete and beds to be redesigned.
@fiveangle  like this ?
[image]

csurf, 2018-01-24:
@madhunm 
ok, I see.
You're saying that the RF switch is EOL and no longer available (or difficult to acquire). That means that a new, different RF switch would have to be chosen, and the RX design would need to be updated. There is probably a newer model of the IC that can be used as a drop-in replacement. If not, the design will have to be modified to use a different RF switch.

In any case, this is beyond the scope of this repo. You're probably better off contacting @readerror67, and/or leaving an issue on the Broversity RX repo here on github. He can help you figure out what newer RF switch to use, and/or can probably help with upgrading the design.

madhunm, 2018-01-24:
@csurf I raised this issue to understand if there was an interest in a dedicated diversity receiver. If there is none, we can close this issue and then carry on. 

However, if there is interest, then I'll wear my thinking cap.

Let me know.

csurf, 2018-01-24:
@madhunm 
I'm a bit confused.
The diversity functionality is something that's already built into the RFM22 RF module. It just needs to be enabled in code. and it requires the external RF switch to actually do the phsyical switching of antennas.  Any new diversity receiver designs that are based on the RFM module will function in basically the same way. The only differences would be in the RF switch IC that's used and on the basic circuit layout & pin mapping. 

However, on the software side, there's not much left to do, as the diversity support has already  been coded.

I see that you opened an issue on the Brotronics repo. That's a good first step to getting the design updated.

FYI, the general goal of this project is to start looking at upgrading the MCU from the ATMega328 to something more modern and powerful (STM32 or something similar). So, in general, I wouldn't recommend investing very much time, energy, or money in trying to focus on the older ATMega328 designs, unless you're really happy with those designs.

atomiclama, 2018-01-24:
Can someone clarify is this please
Reciever diversity as I think used when #define ENABLE_SLAVE_MODE is set.

or 

Antenna diversity where the external RF switch is required.

csurf, 2018-01-24:
@atomiclama 
the `ENABLE_SLAVE_MODE` configuration is used for enabling the DIY diversity using two separate RX modules connected to each other over i2c. You need to enable this in the code and recompile it yourself.

antenna diversity support is built into the standard code, but it requires a diversity receiver like the brotronics broversity receiver (i.e., a receiver with a single RFM22B module, yet with an RF switch IC that switches between two antennas/two SMA connectors).

mikerc2017, 2018-01-24:
yeah . move to stm32 and diversity . Great 

i wonder , version 868mhz also work OK if use diversity ?

rodizio1, 2018-01-27:
Solutions with switches always suffer from several problems:

Isolation, explained here by Thomas Scherrer:
https://youtu.be/C6ZygOeCwLU?t=10m10s

And the general problems that occur with packet/preamble detection, preamble, switching over quickly enough, that's not easy to get right (The Atheros 722n cards used for wifibroadcast for example also have this problem, under certain conditions, the switch doesn't work right and packetloss occurs although the signal is good with only one antenna).

Easiest and best is to use two complete seperate RF paths and simply collect all packets on both RF modules or receivers and throw away the dupes (and make sure out-of-order delivery doesn't occur or is re-ordered ...).

geofrancis, 2018-01-31:
if we went down the route of dual modules for diversity that would also open up the idea of dual transmitter modules, you could run on 2 frequencies simultaneously either dual 433 or 868 or perhaps a mix? for a fully redundant RC system.

mlyle, 2018-01-31:
If you're doing dual receivers, and you have any kind of FEC, you can receive data on both sides-- if a packet decodes on neither, you can combine them and treat any differing bits as known-erasures.  Of course, this requires the same data be transmitted to both or they be on the same frequency.

linuxslate, 2018-02-01:
I know that we are trying to get away from the RFM2x modules (and I agree that we should), but I would like to point out that two Anarduino-22-433's could make for a small, light, and cost effective "true" diversity setup with (mostly) existing code. (Ref #140, my efforts to get Openlrsng working on the Anarduino-22-433).

I'd be willing to help, but due to #124 and #87/#70, it has already taken me 6 RFM2x modules, and I do not have a flight worthy TX/RX set yet.

If I end up with extra working modules (Complete Anadruinos in this case - not just RFM22B's), (or if someone would like to do a group buy), I'd be willing to try to assemble 2 Anarduino's into a R/C receiver like unit.

I just can't keep doing weekly Hope US/Anarduino orders.  (Note that it is not so much a financial thing;  It's more just embarrassment.)

mikeller, 2018-02-01:
@linuxslate: I wonder if it would work to run the modules with a current limiting device (aka smoke stopper), or to add a heat spreader to the power amp IC?

linuxslate, 2018-02-01:
@mikeller :  Short answer: No.  #124 Discussions should probably happen in #124, but I will make a short response here - Q5 (the RF driver) is what gets hot when the module is used incorrectly (bad Controller code) Other low power components (U4?) are also reportedly damaged. The current involved in damaging these smaller components is overshadowed by nominal operation of the power amp (Q3?).  I have some tiny heat sinks that may help with nominal, long-duration operation of the power amp, but I have to get my modules to last that long, first :-/.

Back to on-issue discussion:  I received two more Anaduino-22-433's today.  If my RX module is not damaged, I'll have them to try a diversity setup with. I'll try to make at least a rough drawing of how two Anarduino's can be wired to make a true diversity receiver.

atomiclama, 2018-02-05:
@mlyle You mentioned FEC. I was toying around with the idea of implementing something if I could find a simple library to use. Unfortunately my brief searches didn't turn up anything usable so I put it on the back burner.
Do you have any experience with FEC? Know of any libraries or snippets of code that would be useful? 
Cheers.

artemen, 2018-02-15:
I make my own versions of broversity which shows much better than the 6ch Rx range, if interested I can ship my PCB with schematics for self assembly, I have 2-3 on hand.

mikerc2017, 2018-02-15:
artemen
how far range that you get with version diversity ? and which is rf swich that you used ?

artemen, 2018-02-15:
Well, as always range varies with platform, location and style of flying: I consistetnly fly 3-4miles out in US suburban with my 450 size tricopter, that is with frsky telem enabled (never bothered turning that off) and that is under 50meters above takoff point.

 Now, on my 2meter raptor with RX on an wing I have yet to find my limit :)  with sufficient altitude - the furthest i've gone was just shy of 24miles in Sacramento valley, with more or less flat surface, for video link quality had to go as high as 400meters above takoff, but mostly 100-120 RC was happy. 

The switch i am using is the same as in broversity made by SKY, however decapling caps I use are 30uf (have to double check my schematics) the original broversity schematics doesn't have their value so had to do some research and calculations.

artemen, 2018-02-15:
@csurf  though the antenna switch is obsolete for its intended spectrum/datarate (wifi/cellular) its actually very good for 433mhz low datarate application as it has excellent SWR below 600mhz and more than sufficient switching speed for our application.  And as an added bonus the old new stock is half the price of   its successor. as i was told by a bulk seller, they have more than 40k instock only with that one seller not even considering others.

mikerc2017, 2018-02-15:
@artemen
awesome
 . do you have any video record flights with it ? 
not image with two anten mono . go to far .
Great

madhunm, 2018-02-15:
@artemen

Where are you sourcing the switches from? I'm unable to find any.

artemen, 2018-02-15:
@madhunm
Lol, the last time I was looking for them it was a yer ago ;) still have quite a few on hand... Will write to the seller I got my last batch from no way they just threw them all away ....

madhunm, 2018-02-15:
@artemen

What are your thoughts on respinning the board?

With parts that are available...

artemen, 2018-02-15:
@mikerc2017
monopoles were on for programming purposes- easier to put on desk than with dipoles, and you know what happens to rfm modules without any antennas.  Though simple dipoles are sufficient for 1-2 miles in direct line of sight, thought you loose telemetry at about 1mile mark. These results I saw on a xugongv2 which is half aluminum half CF.  That frame pretty much requires diversity as it blocks RF pretty potently as antenna placement places are very limited.

As for the 24mile video ... I work with companies which have faa333 and i wouldn't want to jeopardize this relationship.  I am not selling these boards, (simply put no time to assemble these boards at a price that will make any sense to anyone). so just ask for one, populate and see which mileage you get (I have two of a bit older design than on a picture unpopulated)  PM your address if you are in the States, and I'll mail you one.

artemen, 2018-02-15:
@madhunm 
It will be relatively easy to rework PCB for https://www.mouser.com/ds/2/472/SKYA21012_203021B-875861.pdf 

control logic is the same, footprint and pinout are different, up to 32dbm RF power handling and OK isolation rating of 22-24db for our frequencies. 

or this one (better): 
https://www.mouser.com/ds/2/196/Infineon-BGSA12GN10-DS-v02_10-EN-1225430.pdf
same conrol logic, needs clean 3.3v, which is present via onboard LP5907.  It has better insertion loss (less than 0.25db under 500mhz), up to 39dbm power handling and 30db isolation between RF ports.

mikerc2017, 2018-02-16:
@Arteman 
i like a rf switch 6 pin, smaller than orginal rx diversity brotronic . 
even i will try use a rf swith availble  on rfm22 pcb 

i flight with mono anten on Rx 100mw but only 400m with telemetry
 Tx 1w. Maybe this  is time to move to divesity

mikerc2017, 2018-02-16:
@arteman
which anten that you used on modul tx ?

madhunm, 2018-02-16:
@artenen

The chips you looked to are tiny! I mean, I'm sure they will disappear if you breathe on them sideways...

I'm partial to the skyworks chip; let me know your thoughts.

Thanks,

mikerc2017, 2018-02-20:
if use diversity for rx with modul 1w .How is it ? 
i saw on tbs rx used two modul sx1272 for diversity . So which is diversity better ?  diversity use Rf switch or diversity two modul ?

artemen, 2018-02-20:
I've made a design for it a few years ago, but came to conclusion that there are no real benefits.  if you really need a bidirectional link, than just use commercial 900mhx WAN equipment for true single frequency rc/video/telemetry link. the prices are less than dragonlink for ubiquity equipment, just need antenna tracker  for ground part. 

for hobby, one more high power transmitter on a vehicle which is already cramped up with equipment is just asking  for trouble, diversity or not. 

I also have a smallish very basic diversity : 

[image]
[image]

mikerc2017, 2018-02-20:
thanks artemen 

design pcb good

jack113113, 2018-03-05:
hello  artemen

version diversity brotronic  also will work with modul 868mhz ?

artemen, 2018-03-05:
@jack113113 
I am, not the correct person to ask, though I do not see a reason why not as long as you compile the source properly.

jack113113, 2018-03-06:
thanks Artemen .  So i will try it

artemen, 2018-03-06:
Btw, all of the openLRS RX designs badly need filtering of power source just before the RFM22/23 here is a video of how the offthe shelf hawkeye 9ch RX, delux v1 TX and my own version of 1W TX affected by noise on power supply. The last one shows sensitivity below -114db while delux v1 at -100db and 9ch Rx at about -108db. In the cvideo there is a short gain in sensitivity of delux v1 when I accidentally touched antenna with my hand. https://photos.app.goo.gl/X3Y9YHd3bG67J29x1

jack113113, 2018-03-06:
thank for infor 

and i move to version 868mhz because   i want to use short anten on quad , hope Diversity work OK .
i wonder , Add a balun or filter input anten 868mhz will better ?

artemen, 2018-03-06:
The specific instance I mentioned is about inability of RFM22/23 along with onboard LDO to filter out noise on power line. RF side is actually sufficiently filtered and 50ohm matched, so no need for additional filtering.

jack113113, 2018-03-06:
thank   artemen

mikerc2017, 2018-03-14:
hi artemen 

do you have same problem  same me ?  warring sensor lost when use betaflight telemetry One Uart ? 

see video problem here https://www.youtube.com/watch?v=hT0vUbQawso

artemen, 2018-03-14:
Sorry i use frskyd and fly only APM hardware, so can't comment on multiprotocol.

mikerc2017, 2018-03-15:
artemen 
mean you only use  RSSI  and 2 analog  in frsky D ?? i wonder in RX diversity brotronic  , Have not 2 analog ???

artemen, 2018-03-15:
@mikerc2017 
i use frskyd output from PX4 family of FCs, it includes voltage, current, GPS, flight mode, GPS coord,  number of GPS sats, altitude, velocity and rssi.

mikerc2017, 2018-03-15:
thanks 
i do not know why my TX alway warring lost sensor , sometime wrong value From FC betaflight . Can not fix this problem in long time

geofrancis, 2018-03-16:
The sensor lost warning happens if a sensor is not updated after 5 seconds.

jack113113, 2018-06-01:
@artemen , i wonder about rf switch 
You  said  : this one (better):
https://www.mouser.com/ds/2/196/Infineon-BGSA12GN10-DS-v02_10-EN-1225430.pdf
same conrol logic, needs clean 3.3v, which is present via onboard LP5907. It has better insertion loss (less than 0.25db under 500mhz), **up to 39dbm power handling and 30db isolation between RF ports.**

what mean ?  RF switch increase sensitivy  ?

madhunm, 2019-12-27:
> I've made a design for it a few years ago, but came to conclusion that there are no real benefits. if you really need a bidirectional link, than just use commercial 900mhz WAN equipment for true single frequency rc/video/telemetry link. the prices are less than dragonlink for ubiquity equipment, just need antenna tracker for ground part.
> 
> for hobby, one more high power transmitter on a vehicle which is already cramped up with equipment is just asking for trouble, diversity or not.
> 
> I also have a smallish very basic diversity :
> 
> [image]
> [image]

@artemen I know this is a very old thread; considering this is still open, do you think you could share the diversity board and schematics?