openLRSng/openLRSng #167: Discussion: Port to STM32

atomiclama, 2018-01-24:
@csurf mentioned in #166 about plans to port to the STM32 platform so I though I'd open this issue to share the work I have done in this area and too see if there is any interest in using it.

So here goes a bit of brain dump ;-)

I have used Betaflight for the base code as this is what I'm familiar with, I have ported the rfm22 radio stack over to us the rx_spi model in betaflight. I'm currently using Nucleo dev boards with a hardwired rfm22 module on it. 
I have a virtual comport implementation that can route the serial on the FC to the serial on the TX so I can use the CLI MSP or any of the telemetries, this is working ok for small packets but MSP is a bit kack.
I have RC control working where the TX side uses the internal RCdata and transfers it over to the FC so it can use any of the betaflight supported RX, so Sbus iBus ppm pwm etc and be used to control the model. I'm currently building it into a quad to see what it feels like.

There is plenty more todo, I only work on this from time to time, so it has been slow progress.

The plan was to get it up and running on
https://www.banggood.com/OpenPilot-OPLINK-MINI-CC3D-REVO-Universal-Transceiver-TX-RX-Module-Integrating-Remote-Controller-p-1000081.html
And
https://www.banggood.com/OpenPilot-CC3D-Revolution-Revo-10DOF-STM32F4-Flight-Controller-Staight-Pin-p-1000068.html

mlyle, 2018-01-24:
Note that we have -- somewhat messy-- openlrsng receiver code running in dRonin.

mikeller, 2018-01-24:
@atomiclama: Would you consider opening a pull request to add the openLRSng SPI RX to the Betaflight codebase? This would be a great addition to Betaflight's growing list of supported SPI RX!

The 'RX passthrough' mode that you are proposing (if I understand you right) is another great idea, and I think we can make this into something that can be added to the official Betaflight code base as well.

atomiclama, 2018-01-24:
Yes that is my intention. But at the moment the code is just concept and needs some tidy. 

Also it is just the RFM22 level, the higher openlrs stack channel hopping etc is not implemented. 
Didn't like the way the RC data was mashed up to shorten the packet length, there are unused bytes in there that can be removed to achieve the same.

Guess I need to do a bit of a write up

csurf, 2018-01-24:
@atomiclama, 
which unused bytes in RC data packets are uou referring to? just curious...

atomiclama, 2018-01-24:
@csurf my mistake thought there was an saving in the config of register 0x33 and sync bytes. But my notes were wrong.

mikeller, 2018-01-24:
@atomiclama:

> Didn't like the way the RC data was mashed up to shorten the packet length

Is that the reduction to 4 bites resolution that you are talking about, or the fact that values that are outside of the [1000, 2000] range are encoded with less resolution? In my opinion, both of these are good approaches to maximise the bandwidth that is available for important data (i.e. the main control channels).

atomiclama, 2018-01-24:
Oh I can't remember the details ;-) it was a while back.

Yes agreed they are good methods to reduce the bandwidth.
I just wanted it simple to start with.

geofrancis, 2018-01-24:
there is a bigger issue this would fix other than more cpu/memory and that is the eprom corruption issues that effects most openlrs hardware since most of it runs the 328 @ 3.3v causing it to corupt the eprom. if it browns out or is power cycled fast,

mikeller, 2018-01-24:
To get back to the original topic of this issue:
I think porting to a different architecture is a worthwhile endeavour, and STM32 is a good candidate. BUT considering that 99% of the existing hardware, and pretty much all of the readily available new hardware currently is Atmel, we should be doing it by *adding* support for STM32, while keeping the existing hardware fully supported for the foreseeable future (even for new features).

geofrancis, 2018-01-24:
there is already a stm32 port for openlrsng but its very old and im not sure what condition its in, there was a 32bit module that dtfuhf was working on but it never really went anywhere, but it could be a good place to start.

https://github.com/openLRSng/openLRSng32/commits/v3.2

mlyle, 2018-01-24:
If one moved to STM32, it might also be worth moving away from RFM22 to some LoRa modules-- that are a couple decibels more sensitive for GFSK/FSK and also support LoRa modes.  There's a lot of SX127x-based modules e.g. RA-01.

geofrancis, 2018-01-24:
I would love to ditch rfm22/32bp modules, I cant count how many have failed over the years. I was looking at QLRS but because he is using the integrated mcu its very memory limited so he can only do basic telemetry.

csurf, 2018-01-24:
similar discussion going on here... #161 

perhaps we need another thread that just discusses RF modules

csurf, 2018-01-24:
RF module upgrade discussion here... #168

atomiclama, 2018-01-25:
> we should be doing it by adding support for STM32, while keeping the existing hardware fully supported for the foreseeable future (even for new features).

A decision for the project owners. 
But there comes a time where maintaining compatibility for old HW will hold the project back. Sometimes a clean break is required to give a project a kick.

mikeller, 2018-01-25:
> But there comes a time where maintaining compatibility for old HW will hold the project back. Sometimes a clean break is required to give a project a kick.

@atomiclama: From the discussions that were going on over the last few days I did not get the impression that there were a lot of enhancements / new features proposed that would go beyond what the existing hardware is capable of doing - potentially we'd have to look into doing different firmware 'flavours' to support different protocols, but that does not seem to be a prohibitive restriction to me, since most users will not have the requirement to switch between different protocols once the hardware has been wired up.
Yes, the point where Atmel MCUs are holding up progress and have to be ditched might come at some point in the future. But until then, putting effort into not shutting the door on users with existing hardware is worth extra effort in my opinion. Experience with the axing of STM32F1 support in Betaflight seems to prove this - users tend to be attached to hardware that is part of their builds, and resent being told they have to go and replace it to get access to new features.

atomiclama, 2018-01-25:
@mikeller 

> But until then, putting effort into not shutting the door on users with existing hardware is worth extra effort in my opinion.

That's great if the devs are willing to put the effort in. I don't know how you manage to do the work you are doing. Hats off to you. I can't even keep 1 quad in a flying state ;-)

mikeller, 2018-01-25:
@atomiclama: If resourcing is so much of a problem that there is only a choice between moving to an entirely new platform, or supporting your existing users, then I think the choice should be to support your existing users.

If users are forced to replace their hardware, a lot of them will potentially jump ship and go for one of the established and readily available long range systems from TBS or FrSky...

csurf, 2018-01-25:
My opinion is that, for AVR platforms, we try to clean-up and add as many practical features as possible before finally freezing & EOL'ing development for it. Existing AVR users will still be able to use old, stable code, but no further development efforts will be dedicated to it if it means having to 'jump through hoops' in order to make new features work on older hardware, and or if it risks compromising the functional stability ( same mentality as in betaflight/iNav development, where older STM platforms slowly get phased out).

For future development with an upgraded MCU (STM or whatever it ends up being), there are lots of 'bells & whistles' features that we can probably look into adding.

here's a non-definitive list (feel free to add more to wish list):
- LCD support
- full, robust telemetry support for MAVLINK, CRSF, LTM, etc.
- native bluetooth and/or wifi for configuration & telemetry
- 'advanced' locator beacon with support for GPS, morse, and beacon (I was already working on this for the AVR)
- come up with some kind of adaptive frequency-hopping strategy, where the RX can suggest to the TX that 'bad', noisy hop channels can be exchanged/replaced with 'good', low-noise channels.
- anything else???

jonathan2018, 2018-01-26:
s martport or fport ?

mikeller, 2018-01-26:
@jonathan2018: Can do both, they are 90% identical.

mlyle, 2018-01-26:
I think it's difficult to keep a unified code base with AVR and ARM, because one of the biggest upsides in moving to ARM is that the increased memory space allows a big improvement in code organization, cleanliness, and maintainability.

mikeller, 2018-01-26:
@mlyle: Not sure about that. If the code is properly modularised, re-using the modules for either multiple specific targets for Atmel or one target for ARM will be possible. In that respect, going multi-platform will be beneficial for the project. The addition of the SITL (x86) target to Betaflight can serve as an example of this.

mlyle, 2018-01-26:
To properly encapsulate and abstract, you need to waste resources to some extent.  AVR doesn't like you wasting resources like that.  ;)

mikeller, 2018-01-26:
@mlyle: I am not an expert in AVR. But gcc with LTO seems to be quite capable of optimising away inefficiencies introduced by source level portable code.

csurf, 2018-01-26:
@mlyle,
Interop with older gear won't really be possible unless we stick with the RFM, and the consensus seems to be growing that we should definitely be considering upgrading the RF module.

I think that there might be some basic concepts that we can carry over from the old code base to the newer platform, such as the general program layout / algorithms on how to handle the bind data, settings layout, order of operations, timing, etc.. 

Obviously, much of the AVR-specific routines will have to be discarded, and a lot of the code will need to be overhauled and abstracted. But, still, I don't think we'd be starting completely from scratch as we already have a 'working concept' that we can model it after.

mlyle, 2018-01-26:
@mikeller LTO's great and all.  But...

Most of the current code reads like a step or two over assembly.  With more resources, you might want to use something like an RTOS.   You'd want subsystems to have opaque handles so you could instantiate them multiple times, etc.  All of these things add overhead-- in memory, in cycles, etc.

mlyle, 2018-01-26:
@csurf --- the SXxxxx stuff can do FSK/GFSK with similar bandwidths-- and a couple decibels more sensitivity and a few decibels better rejection of adjacent interference.  I think a good first move would be to do *that*, and then LoRa specific modes could be added.

mikeller, 2018-01-26:
@mlyle: Have you ever looked into the overheads that are required to run an RTOS? Have you considered how much benefit you'd get out of using an RTOS for something that is as single-tasked as a data forwarder that a TX / RX is? Based on these considerations, and the decision **not** to go with an RTOS in Betaflight, I am not sure if that is something to aim for with openLRSng.

@csurf: I can see this is going into the direction of 'replace the MCU, replace the RFM'. As a consequence, the resulting hardware will not be compatible with all existing hardware, neither by being able to be used toghether, nor by using the same code base.
This is fine, and probably the way to go if you want to come up with the next generation open source LRS. But I have to ask if this should really be 'openLRSng', or an entirely new project (potentially re-using good bits of openLRSng).
I am in the game here to give owners of the current generation of openLRSng hardware the best firmware that is feasible with this hardware - and I think there still is a way to go until we are there. If you want to go the way of completely new hardware, then I am wondering if it wouldn't be better to start a new project or a fork? This isn't resentment or anything, just the opinion that doing both in parallel, and trading ideas and code is probably a better approach than trying to shoehorn both into the same project.

csurf, 2018-01-26:
@mlyle 
yeah, FSK with a more modern RF module is the plan, which is basically what we're discussing that over here... #168 

@mikeller 
I totally agree with all of that, which is why I think it's a good idea just to get a lot of discussions and hopefully more user participation. Honestly, the reason I wanted to get into working on this project was because I was an avid fan/user of olrsng, and was disappointed to see it stagnate and basically start to die. This is why I'm excited to see people taking interest in it again.

I think the goal should be to try to pick up where kha left off and try to clean things up and build it up as much as possible within the limits of the hardware. If we can add a few more features and/or try to improve on some of the existing features, then I would consider that a success...

I agree that, in the long-term, firmware based on new hardware and a newer codeset would be better served by having its own repo & a new name / new identity. However, I think that we're still a ways off from achieving that goal. When that time comes, I would hope that we could have some momentum built up here (in terms of user participation and good workflow) that would could carry over to a new project. 

I think an analogy might be how cleanflight picked up where baseflight left off, and people then got excited about working on it, and betaflight then emerged (the rest is history). It would be great to see a similar bit  of progression with openLRSng (and beyond...).

mlyle, 2018-01-26:
@csurf --- so then if you're doing FSK with more modern hardware, if you maintain bandwidths & deviation & frame format it should interop.

@mikeller --- I don't think "betaflight does it" instantly proves good practices or good architecture--- out in industry you don't find many things that choose to maintain all their hardware abstraction and tasking infrastructure themselves anymore (it's not 1995 anymore).  And, if you write something without any kind of real hardware abstraction, you're stuck with the parts you initially selected forever (e.g. once these STM parts are no longer the best choice, you'll be stuck in the same shitty situation you are today).

I maintain dRonin, which is a flight controller firmware that among other thigns, runs an openlrs receiver *as a task*.  It also does things like bitbang video overlays *as a task*.  An RTOS + what OpenLRS does + many other things will run on a low-end STM easily, so a better question is why *not* run an RTOS?  It lets you hit critical timing for the important stuff.

In fact, with a small amount of cleanup to the OpenLRS parts of the dR codebase, it'd be easy to just pop a few modules together (ppm receiver, sbus receiver, openlrs transceiver for an olrsng transmitter) and have the functionality that is in this codebase.  It might be worth considering this path.

mlyle, 2018-01-26:
Just one more thought-- openlrs hardware is no longer available commercially.  I think efforts to "build up" the current codebase require convincing someone to make the hardware again-- which seems difficult.

mlyle, 2018-01-26:
Oh, and another last thing--- https://github.com/d-ronin/dronin-hw/tree/master/pipxtreme/TauLinkModule  Here's RFM22b+STM32F1+Bluetooth module, in a JR module footprint.  So an open design exists now that could be used as a basis for development.

csurf, 2018-01-26:
@mlyle 
yes, this is true in a sense, but not necessarily "out of the box". 

The way it's currently done is to use the RFM in 'packet handling' mode, which means that all of the packet handling ops (construction/packaging, buffering, reception, parsing, etc.) are all offloaded to the RF module, and it's understood that each line of RF module/IC has its own way of doing things (this is the way I understood it at least). 

From my understanding, if you choose to opt out of using the built-in packet handler, then you'd have to do all of the packet management manually/via the main MCU. Obviously, if you have enough hardware resources to spare on packet processing, then this should definitely be possible. However, it's not ideal as it would take away horsepower that can be used for other things. 

So, the ideal setup is to simply use the RF module's built-in packet handler (assuming it has one) and save resources, However, it's not impossible to do it manually...

mlyle, 2018-01-26:
@csurf SX1276 has a flexible packet handler --- with your choice of sync word handling, data whitening, etc.  see 4.2.13, it can do preamble receive and arbitrary sync word detection.

csurf, 2018-01-26:
@mlyle 
cool... yeah, I actually have two RFM69W modules sitting on my desk waiting for me to mess around with. I just haven't had the time/patience to sit down, dig through the datasheet, and try to get something working with FSK (many examples for LoRa, haven't found anything good for FSK).

geofrancis, 2018-01-26:
As for compatibility, it really just needs the Tx module to retain comptibility with the mega based receivers since most people just have a single module but many receivers.so if you break compaitbility with receivers it means people will have to keep 2 modules, but if your new module had a "legacy mode" that could work with older receivers like frsky do with D8 and D16 receivers. then you could keep your old receivers and still upgrade your module and future receivers.

csurf, 2018-01-26:
@everyone
in terms of selecting an STM32 chip, what's the best choice in terms of value vs. resources? I'm not much of an STM guy, so feedback & help would be appreciated...

geofrancis, 2018-01-26:
The f3 is far more flexable with its uarts than a f1 or f4,

mlyle, 2018-01-27:
I would recommend we stick with the taulinkmodule hardware from @peabody124 for prototyping, since it already exists and is known to work--- we could be up and running very quickly.

geofrancis, 2018-01-27:
yea they are cheap and easy to get hold of.

rodizio1, 2018-01-27:
Please consider using Ti1 CC1310 modules,  there are complete small ready-to-use modules available with 30dbm PA for 5-15$:
https://www.aliexpress.com/wholesale?catId=0&SearchText=cc1310

TI CC1310 infos:
http://www.ti.com/product/CC1310
http://www.ti.com/product/CC1310/technicaldocuments

TI PDF comparing Lora with narrowband:
http://www.akermann.bg/files/Semnrs/H2016/Module3.4_100km%20TI%20NarrowBand%20vs%20LORA.pdf

Vidoes:
http://www.youtube.com/watch?v=7zLUYswu3Ek

TI CC1120 rangetest (with lower data rate though...)
https://www.youtube.com/watch?v=wgqtEu5PfAw

TI CC120 rangetest 2
https://www.youtube.com/watch?v=zMcRYLHCsw0

TI CC1120 application notes with urban area non-LOS range test:
http://www.ti.com/lit/an/swra482/swra482.pdf

Achieving optimal radio range PDF (with detailed test-descriptions, also an indoor test through 12 stories):
http://www.ti.com/lit/an/swra479a/swra479a.pdf

Infos on channel hopping:
http://www.ti.com/lit/an/swra482/swra482.pdf

mikeller, 2018-01-27:
@mlyle: My comment wasn't meant as 'Betaflight does it'. But there were some considerations made about RTOS, and in the end the decision was that the overhead and performance cost did not justify the benefits gained. For something that is a lot simpler than a flight controller, using RTOS would introduce even more overhead compared to the performance requirements to do the actual work. But I guess you are right, if you start out by overspecifying the hardware requirements (which is cheap these days), then you can easily afford to use RTOS.

> Just one more thought-- openlrs hardware is no longer available commercially. I think efforts to "build up" the current codebase require convincing someone to make the hardware again-- which seems difficult.

I think you might be wrong there (and that's just the result of 20 seconds of googling):
https://hobbyking.com/en_us/orangerx-open-lrs-433mhz-transmitter-1w-jr-turnigy-compatible.html?___store=en_us
https://www.fpvmodel.com/dtf-uhf-deluxetx-transmitter-openlrs-uhf-system-jr-turnigy-taranis-compatible-with-9ch-receiver-and-6ch-receiver_g467.html
https://www.banggood.com/Wolfbox-1000mW-1W-433MHz-UHF-Transmitter-Tx-100mW-Receiver-Rx-Compatible-with-X9D-X12S-9XR-TH9X-p-1051206.html

> Oh, and another last thing--- https://github.com/d-ronin/dronin-hw/tree/master/pipxtreme/TauLinkModule Here's RFM22b+STM32F1+Bluetooth module, in a JR module footprint. So an open design exists now that could be used as a basis for development.

That looks interesting. Google came up empty for a vendor for these modules. Do you have a source where they can be ordered?

@geofrancis:

> The f3 is far more flexable with its uarts than a f1 or f4

Good point. The pain of having to deal with unidirectional / bidirectional / inverted UARTs externally should be avoided right from the get-go for new hardware.

geofrancis, 2018-01-28:
> Good point. The pain of having to deal with unidirectional / bidirectional / inverted UARTs externally should be avoided right from the get-go for new hardware.

with multiple UARTS like on a f3 board you could do a lot of things, specifically serial receiver with telemetry and you could even have a another port for a gps or telemetry converter or LEDS or VTX control,  so something in that direction might be the way to go

Timvrakas, 2018-01-28:
The TauLink hardware checks all the boxes for me. I hope to have custom SiLabs based radio done soon, which could replace the rfm22 with only small software changes. PlatformIO support is also great for STM.

On the other hand you might consider an ESP-8266 as the MCU and also a WiFi interface. Personally, WiFi always beats Bluetooth. Or if you wan't more power, the ESP32 is a faster MCU, with WiFi and Bluetooth.

atomiclama, 2018-01-29:
This module also exists at the moment and would be good to get up and running.

https://www.banggood.com/OpenPilot-OPLINK-MINI-CC3D-REVO-Universal-Transceiver-TX-RX-Module-Integrating-Remote-Controller-p-1000081.html

atomiclama, 2018-01-29:
Can I vote for any of the STM range that has the HW encryption IP built in.

mlyle, 2018-01-30:
@atomiclama -- I would vote against, because A) ST has different SKUs in part because it creates export control problems, etc, and export control can be difficult for small manufacturers and retailers to deal with, B) GCM AES is like 27 cycles/bit on STM32, so for any reasonable throughput it is going to be a small part of the system, and C) lots of the world can't use encrypted communciations.

Note that -authenticating- messages doesn't violate amateur radio regs in most of the world, and HMAC with SHA-256 is pretty cheap.

mlyle, 2018-01-30:
Note that oplink mini and the taulinkmodule are the same thing (roughly) and run the same firmware.  TLM just has integrated bluetooth + fits in a JR module slot.

atomiclama, 2018-01-30:
@mlyle Fair enough on the encryption.

As to the module. Can the TLM be bought as an assembled module? Where from?

mlyle, 2018-01-30:
@atomiclama No, there's no commercial availability of TLM.  As you point out, oplink mini is the same hardware and can be bought and stuck in a module bay, and there's design files for TLM up.  I might be willing to make a few TLM for people (or alternatively, designing a quick TLM+some widely available SX12xx module) if there's real interest in moving forward on this.

mikerc2017, 2018-02-15:
Any think about arduino M0 ? It is arm 32bit from atmel

jnb81, 2018-04-20:
considering STM32s' I think it should be carefully chosen (as well as hw layout)
https://quadmeup.com/stm32-f1-and-f3-versus-lrs-433mhz-interference-issue/

mlyle, 2018-04-20:
@jnb81 --- the oplink hardware we're talking about has been around for 5 years, with pretty good results.  It uses good bypassing and puts the RFM module on the opposite side of the board than the micro (so there's reasonable shielding from ground planes.  Also, I don't think an interferer 15-20dB over the noise floor at very close range is worth getting too worried about.

jonathan2018, 2018-06-25:
@csurf  any idea with a Atmege328bp ? 

Two more 16-bit timer/counters
Two of each USART, SPI, and I2C serial instead of one of each
Ten PWM channels instead of six