openLRSng/openLRSng #168: RF module upgrade discussion

csurf, 2018-01-24:
Let's get the conversation started here on topics related to upgrading to a newer, better RF module...

I just recently received a couple of HopeRF RF69W 433 LoRa modules. Anyone have any experience with these? These are driven by the semtech SX1231 RF IC.

My goal is to figure out how to do FSK with them, but there isn't much info available on this yet outside of datasheet. Most people use these modules for LoRa modulation, so there's lots of info on how to do LoRa but not so much on the other modulation schemes.

fiveangle, 2018-01-25:
Yes, it's quite a bad datasheet (almost as bad as the SX1278) but the info is there on p63 in the register table under _RegDataModul_ http://www.hoperf.com/upload/rfchip/RF69-V1.2.pdf

mikerc2017, 2018-01-25:
Lora (i.e. Crossfire and any other LRS based on Lora) doesn't work properly when there is interference.
https://www.rcgroups.com/forums/showthread.php?2037442-Ultimate-LRS-40-km-APM-Telemetry-RC-link-for-54-%28using-2-OrangeRX-1W-TX%29/page850

fiveangle, 2018-01-25:
https://www.rcgroups.com/forums/showpost.php?p=39048980&postcount=12755

atomiclama, 2018-01-25:
Lots of reading to do. Thanks for all the links and info.

atomiclama, 2018-01-25:
Another link for the pot of information. This one for a library to use the module. May be able to get some information as to how to configure it.

https://github.com/LowPowerLab/RFM69

Linjieqiang, 2018-01-25:
Maybe we can use STM32F0 and LORA RX1279 to make this project again.

rodizio1, 2018-01-27:
Lora doesn't really work well when there is interference (see all the threads about the crossfire for example).

The Si4331 is completely outdated, since the IoT hype, manufacturers have improved sensitivity and interference-immunity quite a lot.

I'd either use something with a TI CC1310 or with Silabs si4464/63/62/61, these are much newer and better chips.

TI PDF comparing Lora with narrowband:
http://www.akermann.bg/files/Semnrs/H2016/Module3.4_100km%20TI%20NarrowBand%20vs%20LORA.pdf

Interference test against Lora:
http://www.youtube.com/watch?v=7zLUYswu3Ek

csurf, 2018-01-27:
@rodizio1 
good info...

FYI, we don't want to necessarily use LoRa modulation. We'd still use FSK for RC control/telemetry. The consideration of LoRa modules is due to the fact that they use more modern RF IC's that can also do FSK (in order to move away from RFM). So, even though we're considering moving to LoRa-capable modules, we wouldn't necessarily modulate with LoRa.

rodizio1, 2018-01-27:
Well, LoRa's "main feature" is the wideband Lora modulation, not sure how good the narrowband modulation modes work on them compared to TI's or Silabs' modules which are optimized/specialized for narrowband.

csurf, 2018-01-27:
> not sure how good the narrowband modulation modes work on them compared to TI's or Silabs' modules which are optimized/specialized for narrowband.

This is a big question which we are trying to ascertain. The consensus is that we definitely want/need to move to a newer RF module, preferably something with better sensitivity and improved datarates if possible. The LoRa-capable modules seemed to be the most popular options, and it has been stated that both TBS and FrSky use them (I can't personally confirm this, but this is what other have claimed.

mikeller, 2018-01-29:
I was just able to confirm for FrSky:

[image]

rodizio1, 2018-01-29:
Crossfire is also using Semtech, not sure which exact chip from the top of my head. It was developed by a swiss company (http://www.miromico.ch/) for TBS.

jnb81, 2018-04-18:
what about SI4463? with -126dBm sensitivity
rx should gain 50% in reception comparing with current hw.

MicroMontis, 2018-04-19:
I vote for SI4463 module. It should be pretty easy to implement support for it. 
Moreover there is confirmation that SI4432(RFM22B) should be able to transmit/receive packets to/from 
SI4463 (compatibility):
https://www.silabs.com/community/wireless/bluetooth/forum.topic.html/si4432_and_si4463pa-K9Tk
In this case we could mix the modules like RFM23 for TX and SI4463 for RX.

Timvrakas, 2018-04-19:
I third the Si4463, and I'm doing hardware loss tests with Si4463 + TCXO prototypes I can share if anyone wants. Sticking with the SiLabs SPI interface vastly simplifies code changes, as the RFM22 was just an older chip of the same family.

Rusty105, 2018-04-19:
@jnb81 

Assuming the noise floor is as low.

dothien113, 2018-04-20:
video test SI4432  500mw TX and 100mw RX , it cheap 

https://www.youtube.com/watch?v=yw4sfvSYF9k&feature=youtu.be

JamesCordell, 2018-04-30:
@csurf Is the LoRa CSS modulation not appropriate for rc controllers as opposed to FSK? Or has this not yet been studied.

JamesCordell, 2018-05-02:
I have found the answer to my question. http://www.akermann.bg/files/Semnrs/H2016/Module3.4_100km%20TI%20NarrowBand%20vs%20LORA.pdf

Would the rfm98w and rmf98pw be a good candidate for a rx/tx setup?