openLRSng/openLRSng #172: RFC: finder beacon upgrades

csurf, 2018-01-26:
here's a "request for comments" on adding some features & functionality to the locator beacon. I was hoping I could get some feedback:

how do the following ideas sound?
- a 'beacon decay' setting to slowly increase the beacon interval over a period of time, in order to save battery and not transmit as often
- GPS beacon that can read GPS over serial and transmit the coordinates in Morse code (I already have a prototype of this feature in another github repo)
- custom beacon tunes (in addition to the fixed tone beacon and 'close encounters of the 3rd kind' tune)
- voice beacon for GPS coords (this takes up a lot of flash space, so might not be feasible on ATMega328 with the RX code also loaded on it, but we can try)
- periodic callsign via morse code (to comply with HAM reg's)
- any other ideas???

geofrancis, 2018-01-26:
You could get the gps information from the telemetry screen.

could the module transmit the gps coordinates in morse for them to be picked up on the handset module somenow, I know the RFM22b 100mw modules are used on cube sats at very low data rates.

csurf, 2018-01-26:
@geofrancis 
yeah, I actually use a simple smartphone app to decode the coordinates.

here's an old demo video showing the GPS Morse beacon:
https://www.youtube.com/watch?v=bE2lZfsWmL8

In that setup, I've set it up to transmit the the lat & lon in alternating transmissions, along with a callsign at the end ('CSRF' in this case).

The 'good/bad' tone at the beginning is meant to indicate if the GPS module still has an actual 3D lock, or if it's transmitting the 'stale', last-known coordinates. This is meant to give you an idea as to where the beacon/plane actually is (ex., if it's at the actual location of the coordinates, or if you should be searching somewhere in the vicinity based on 'dead reckoning'). A high, rising tone sequence indicates a good GPS lock, while a low, descending sequence means 'bad GPS'. 

I also added an 'offset' feature that adds preconfigured offsets to the coordinates so that others won't be able to find the beacon if they don't know the offsets (i.e., the coordinates in the video aren't my actual position; they've been offset by some amount that only I know).

FYI, the Morse transmit rate is somewhat slow in that example. I had another video where I sped it up substantially, and the Morse code decoder app will still able to keep up with it and reliably parse it. Unfortunately, I must have deleted that vid, so I'll have to make another.

csurf, 2018-01-26:
here's a demo video of the GPS voice beacon...
https://www.youtube.com/watch?v=k_s6RlvcKfo

closed 2018-02-19 by csurf

jnb81, 2019-01-13:
Hi, I was just gonna open new "issue" about gps voice beacon, can you please share more info? How to make it done?
I'm very interested into this project. 
On my current build I use open lrs (3.9.0 rc2) on orange RX, Omnibus F4 PRO on iNAV, and or of course GPS module. |I always has been looking into:
https://tbeacon.org/?page=6&language=en&session=t1nopvv4igatm9cmmo0ffrtql1
But when you look at HW almost everything is already on my plane.  
I love open lrs multi! I can fly 10+km and have 70% on orange 0.1W HW :)
https://youtu.be/Dz7UQQBEoPM
Open lrs multi deserves more attention!
I'd appropriate if you can give me some guidance how to implement gps voice beacon on my current setup.

jnb81, 2019-01-13:
OK after some reading I guess it's no way to have RX module as standard receiver and at the same time as gps beacon (probably due to HW (memory) shortage only simple beacon is supported), it's either gps beacon or receiver.
Still super useful.
Is the voice already implemented?
https://github.com/csurf/openLRSngBeacon-GPS
I guess I have to compile openLRSngBeacon-GPS.ino and load hex file into receiver unit?