openLRSng/openLRSng #184: Review RFM register settings

csurf, 2018-02-02:
For those that have experience working with the RFM22B and/or have good experience with general RF concepts, I think it might be worthwhile to revisit & discuss the various register settings that are currently being used to configure the RFM module within the openLRSng code.

here are links to the relevant code:
https://github.com/openLRSng/openLRSng/blob/master/openLRSng/RFM.h#L229
https://github.com/openLRSng/openLRSng/blob/master/openLRSng/RFM.h#L185
https://github.com/openLRSng/openLRSng/blob/master/openLRSng/binding.h#L161

link to datasheets stored in the wiki repo:
https://github.com/openLRSng/openLRSngWiki/tree/master/datasheets

All of the current RFM settings were initially setup by kha, and I'm not sure if they were carried over from some previous version of openLRS, or if he came up with them by himself. There wasn't much documentation/comments within the code (I tried to add comments for what I could decipher), so we'd have to dig through the datasheets to figure out what the various register settings are meant to achieve.

Basically, I think we should try to get a better understanding on why these settings are being used, and also if they perhaps should be tweaked for our application in order to try to achieve better performance .  Perhaps we can improve on the datarates and/or sensitivity by adjusting & testing different register settings?

atomiclama, 2018-02-02:
Is the Hope chip a copy of the Si chip ?

Found this a while back but never got round to comparing the register results or evaluating which was better. From code or from calculated.

[Si443x-Register-Settings_RevB1.zip](https://github.com/openLRSng/openLRSng/files/1688532/Si443x-Register-Settings_RevB1.zip)

I'll add it her as it maybe of help to someone.

csurf, 2018-02-02:
@atomiclama, yeah, it's the same chip. It's the Si4432 I *believe...

FYI, that spreadsheet should also be in the wiki folder that I linked to above, but thanks for posting it anyway.

atomiclama, 2018-02-03:
Opps ! should have looked in there first.

closed 2018-02-19 by csurf