openLRSng/openLRSng #187: PPM scaling issue for SBUS input

csurf, 2018-02-05:
@mikeller,
I just revisited a setup I have that takes SBUS input from an FrSky receive (x8r), feeds it into an olrs TX, and outputs PPM on the RX side. The scaling appears to be broken, as I'm getting rollover on the lower-end of the stick range...

example:
I setup PPM debugging on the RX side to output the raw PPM values coming in over the air (prior to scaling on RX side). Moving a control stick to the bottom of its range causes the output to jump from a minimum value of '12' to '1023'. It appears there must be a problem with the scaling of the FrSky SBUS signal going into the TX, and it's causing the uint16 values to roll over.

Since you were the last one to work on the SBUS-PPM conversion stuff in TX.h, I figured I'd ask you first...

csurf, 2018-02-05:
@mikeller 
I see that you ended up using a different formula for the sbus to usec scaling:
https://github.com/openLRSng/openLRSng/blob/master/openLRSng/TX.h#L699

Originally, I believe we were using the following formula which worked for FrSky SBUS (not sure about the actual Futaba version though):
```
uint16_t value = ((input * 5) >> 3) + 880;
```

mikeller, 2018-02-08:
Yes. This was done to make it symmetrical with https://github.com/openLRSng/openLRSng/blob/2d07c20b72c00911b69c7aa27e008068c871f347/openLRSng/serialPPM.h#L53-L58 which was designed to match how SBus is received in Betaflight / cleanflight. I never tested the TX side, so if the scaling of SBus on TX and RX side is different, then `SBUSToPpm` is probably bogus as it is.

PR #190 (2018-02-09): fix SBUS input scaling for TX (using FrSky scaling)
- fixes SBUS scaling to match FrSky SBUS scaling, fixes #187
- code clean-up for serialPPM.h
closed 2018-02-09 by csurf