openLRSng/openLRSng #188: Feature Request: "Emergency mode" / switchable data rates

csurf, 2018-02-05:
Per a discussion with @geofrancis, here's a proposal for an "emergency mode" where we can manually drop the data rate in the air in order to increase senstivity & range at the expense of a lower data rate.

here are the key points on the RX side:
- add user-configurable settings to enable an 'emergency mode' which drops the data rate down to a lower value (4800 or perhaps lower) after entering failsafe
- emergency mode can be initiated either immediately upon entering failsafe, or after a configured time-out. This is in order to give the RX a chance to relink at the higher data rate before finally giving up and dropping down to the emergency data rate
- add a means of disabling emergency mode & going back to the original data rate. For non-telemetry setups, this could be via an RC channel value, or perhaps we could come up with a more creative means of signaling to the RX that it should try to go back up to the higher rate

on the TX side:
- add a means of manually switching emergency mode on & off, either via an RC channel, a physical switch, long-pressing the bind button, or a multiprotocol command.
- add a way to signal to the RX that it needs to try to go back to the normal data rate when conditions improve and e-mode is no longer needed. For telemetry setups, we could come up with a way to send it in the packet metadata & have the RX ACK the change before changing over. For non-telemetry setups, we'd have to come up with some other way to signal the RX (toggle an RC channel or something)

on both sides, e-mode would require:
- change the packet structure & transmit rate so that channel payloads & update rates are broken up
- instead of sending all of the channels in one packet, we would:
  - send the first four control channels in a single packet at the highest update rate possible (~20Hz?)
  - occassionally, send the AUX channels in separate, alternating 4-channel packets at a lower rate (~5Hz?)
  example: 
    - AUX packet 1: channels 4 - 8
    - AUX packet 2: channels 8 - 12
    - AUX packet 3: channels 12 - 16

more feedback & ideas needed

mlyle, 2018-02-06:
This is dangerous.  It just provides more ways to get out of sync.

If you want to do rate negotiation, it needs telemetry, and you need a continual consensus operation that sets "when the deadline to go slow is" and where the difference in these counters can be definitively bounded.

geofrancis, 2018-02-06:
its doesnt have to be that complicated, its not rate negociation, this would only drop to low speed if manually switched or the RX is already in failsafe.

mlyle, 2018-02-06:
@geofrancis If you do it the way specified, getting out of failsafe requires realizing you're in failsafe and flipping the emergency mode switch.  So a small blip in signal that would self-recover becomes something that requires operator action to fix.  (And it becomes something an operator can make worse up to that timeout, where the RX has actually received a frame recently and after the operator flips the switch you need to wait the timeout again before recovering).

This seems not good.

csurf, 2018-02-06:
@mlyle 

'This would be a feature for advanced users who understand the risks involved, their aircraft, and their range limits. Obviously, you'd also need to have your RTH / failsafe measures in place beforehand.

This wouldn't be meant for dealing with small failsafe glitches. It would be used after substantial glitching / packet loss in order to manually kick the TX down to the lower rate. 

If you're on the edge of range and on the verge of going into failsafe, you can simply force the TX into e-mode, which would consequently force the RX into failsafe as well. Then, you just wait for the RX to e-mode timeout to expire. In other words, after the RX times out trying to listen for packets in 'normal mode' (higher data rate), it'll drop down into e-mode and should start catching packets at the lower data rate. At that point, you can regain control and decide what to do: keeping flying for further range, or return home under manual control. Then, as soon as you feel confident you're within strong signal range, you can return to 'normal mode', with the higher data & refresh rates.

@geofrancis how does this sound???

a possible setup:
- e-mode could have two  RX settings: 
  - e-mode failsafe timeout
  - e-mode listen interval 
- e-mode timeout value sets the time period to wait after failsafe kicks in before switching over to the lower data rate and listening for packets. Ex., a value of '10 seconds' would mean that it would switch over to e-mode 10 seconds after failsafe engages
- e-mode listen interval could be the amount of time that the RX should listen for e-mode packets before switching back to 'normal mode' to listen for packets. At that point, it would start listening for normal mode packets for the same amount of time. So, it would alternate between listening for e-mode and normal mode packets for 'x' amount of time.

example:
e-mode failsafe timeout: 3 seconds
e-mode interval: 2.5 seconds

- after failsafe kicks, in, the RX will wait 5 seconds before trying to go into e-mode for first time.
- after the 5 second timeout, the RX will listen for e-mode packets for 2.5 seconds
- after 2.5 seconds expire and no e-mode packets received, it'll switch back to normal mode
- the RX will then listen for normal-mode packets for 2.5 seconds
- if no normal-mode packets are received, it'll switch back to listening for e-mode packets for 2.5 sec
- so, it'll basically alternate between listening in each mode for the 'e-mode interval' duration
- this might work out great because it allows us to manually force a transition from the TX side and have the RX adapt to either, without having to worry about sending telemetry commands, getting an acknowledgement from the RX, sync'ing both sides, etc.

thoughts?

mlyle, 2018-02-06:
@csurf -- it's just more to go wrong.  You could select "slow" and not know that a frame had gotten through and need to wait for the timeout, etc.  In turn it interacts with the channel-hopping resynchronization, and makes things pretty non-deterministic even if you "know what you're doing"

There's safe ways to do this with a telemetry channel-- you can run a consensus operation between tx and rx.  Rx telemetry can say the number of frames from now that it is going to go into slow mode.  Tx can be sending a number of frames from now to go into slow mode, and make sure it never sends a value higher than 3 past what it heard from the rx last time, and never more than an absolute number (30?).  Then at most there will be 3 frames where one side is in slow mode and the other side isn't, and with a good link the timeout can be bumped to a good number to not give up too early.

Or without a telemetry channel, you could send fast frames too saying "i'm in slow mode", I guess.  Though I don't really like this scheme.

csurf, 2018-02-07:
@mlyle 
well, although I don't necessarily agree this will be a problem, I also can't rule it out yet as we're still in the brainstorming phase and don't actually have anything to test with hehe.

A big first step will be to try to build a working prototype. My proposed setup will work for non-telemetry configurations, and it'll be simpler to implement, so I think it'll be a good first step. Afterwards, we can look at how to set things up over telemetry.

The first big task will be to get the e-mode packet configuration & data rate setup. This will require new code to handle the different packet payloads and refresh rates. Once that's built, we can start looking at how to switch into e-mode.

geofrancis, 2018-02-07:
i would think if the receiver loses signal for signal for 1 second and normally that would failsafe,

but with e mode after that second of failsafe on normal rate it would try the emode rate and if it still doesnt get anythign for a second it failsafes. while its in failsafe it will try for a second on the high rate before a second on the low rate untill it gets a signal where it will stay until it looses it and repeates the 1 second scans,

the effect would only be adding 1 second to the reconnect time if its been out of contact for more than 1 second or any set amount of time

csurf, 2018-02-07:
@geofrancis 
yup, I think we're on same page then. The RX will simply keep trying at the two different rates, switching back & forth. If it doesn't catch a packet in either mode, then it's screwed anyways, and will simply stay in failsafe (and RTH/land/crash/whatever).

mlyle, 2018-02-07:
Right now the reconnect logic is based on the Rx slowing channel switch
rate and hoping the tx advances in phase and matches the channel.   If you
toggle back and forth at 1Hz there is double the state that has to align to
reconnect.  Already, reconnect is not particularly fast in marginal
conditions.

There are better ways to do this :/

mlyle, 2018-02-07:
I am on my phone now but one example of a better protocol--- designate an
"always slow", "always fast" channel in the hop sequence, and have a
rotating "opposite speed" channel.  Carry the mode information in the
frame.  Ensure that the slow and fast hop rates are relatively prime in a
fashion that guarantees speed alignment.

This is robust against multiple failures-- desync + rate unknown + any
individual channels blocked, and is fast to obtain sync when no channels
are blocked.

csurf, 2018-02-07:
if you can come up with a proof of concept, then I guess we could look at it, but on the surface it does seem a bit more complicated to be honest.

In any case, we'd still need to figure out the basics of the e-mode refresh & data rate setup, so there's a lot of work to do beforehand.

mlyle, 2018-02-07:
> if you can come up with a proof of concept, then I guess we could look at it, but on the surface it does seem a bit more complicated to be honest.

Sometimes the most trivial things do not work well or are dangerous.

fiveangle, 2018-02-07:
> Sometimes the most trivial things do not work well or are dangerous.

…and you will never know unless you try.

Pointing out possible issues and proposing alternatives is absolutely what brainstorming is about. But there is nothing more toxic than pure fear-mongering without substance during in the brainstorming phase.

I like your idea about the synchronization of a slow and fast (far and near) stream. It might be worth exploring, but it certainly seems far more complex of a solution than that proposed to auto-drop on failsafe, and ripe for unintended consequences that are manifold. A proof of concept like suggested would really make a lot of sense here because it doesn’t really pass the K.I.S.S. sniff test.  But nobody will poo-poo it before it gets out of the gate, and I hope you agree that other ideas deserve that same courtesy.

mlyle, 2018-02-07:
I've already proposed things that are more reliable.  Adding a mechanism where the user has to guess what state the remote radio in is no bueno, which was the original proposal.

As to the newer proposal:

The OpenLRS resynchronization mechanism was clearly carefully thought out.  The RX steps through channels in a very slow fashion, while the TX continues to hop quickly.  The hop rates are chosen so that even if a channel is blocked, that the RX & TX will end up on a common channel relatively quickly.  I would characterize its performance as *adequate*.

Now we're proposing a system where the RX will spend half of its time listening in a mode where it will miss TX frames.  There's some *requirements* to make sure this works well-- that all of the frequencies of changes (RX channel change rate, RX speed, TX channel change rate) are guaranteed to produce appropriate overlaps.  And even with a perfect set of parameters, performance in the normal resync condition is guaranteed to be worse than the current OpenLRS system.  With non-ideal parameters, you can get "unlucky" and never resynchronize.

Namely, you need to make sure that prime factors are not shared in resulting times, and that things are "long" enough that you're not likely to switch mid-message.  If we're going to go through that work to come up with something that will provably resync, why not make something *good*?  It's not much harder.

mlyle, 2018-02-07:
(The absolute best option, IMO,  would be to not use the rfm22 packet handler other than the start of frame detection, and always send a "fast mode" preamble before the slow mode preamble, and being able to shave the duplicate bits down and demodulate slow mode packets in fast mode.  But this is a fiddly, annoying thing to build, so I've not proposed this).

csurf, 2018-02-07:
... getting back setting up the e-mode data rate...

@geofrancis & others
To send 4 channel blocks, we can do a fixed-length 6-byte packet:
- 1 byte header/packet metadata
- 5 bytes for 4 channels (10 bits per channel)

assuming a 13-byte overhead
- preamble: 5 bytes (we assume no diversity here)
- sync word: 2 bytes
- header: 4 bytes
- CRC: 2 bytes
that's a total of 19 bytes
* NOTE: we save a header byte by doing fixed-length packets here

using the formula found here (also needs to be tested/verified)...
https://github.com/openLRSng/openLRSng/blob/master/openLRSng/common.h#L67

that should give us an update rate of around ~29Hz (~34458us packet interval) for uplink-only / no telemetry. That seems pretty decent to me if we can split the update rates for control & AUX channels...
- control block: ~10-15Hz
- AUX blocks: alternating @ ~5Hz ???

Enabling telemetry will lower the update rate, as you have to add the packet interval for the telemetry packet to the total interval. Maybe we can do a 'tiny telemetry' packet of only 2-3 bytes? thoughts???

csurf, 2018-02-07:
another thought...
perhaps we could experiment with going with a 2-byte header instead of 4 bytes, and a 1-byte sync word. That'll save us 3 bytes (every byte savings counts at this data rate). 

The 4-byte header is used for transmitting the 32-bit 'RF magic word' for matching the packet source & destination (TX and RX). I don't see why we couldn't get by with a 2-byte (or even a 1-byte) 'RF magic' header in this scenario.

feedback needed...

atomiclama, 2018-02-07:
I'm currently running with 2 byte header and 1 byte sync without problems, but this is land based range testing. I don't have any real world flight experience with it yet. 

I think the magic header is only an issue if you are flying with others using openlrs so could be decreased if the users are careful.

mlyle, 2018-02-07:
You guys uh... do what you want.  I'll go do what I think makes sense somewhere else.

mlyle, 2018-02-07:
Just for clarity, though:

> This, btw, would fundamentally cause issues with the idea of mixing speeds and trying to designate certain channels for a particular speed (that's how I understood it anyway).

The idea is, if you have 6 channels, 1 would be "always slow", 1 would be "always fast".  1 would be round-robining to be "the opposite of the TX's speed switch".

The worst case scenario would be that fast doesn't work, and the "always slow" channel is blocked-- re-acquisition works then still, but is slower.

Fast mode hop would look like

```
FSFFFS
FFSFFS
FFFSFS
FFFFSS
FSFFFS
...
```

Slow mode hop would look something like:

```
FFSSSS
FSFSSS
FSSFSS
FSSSFS
FFSSSS
...
```

Alternatively you could decide on a rule that doesn't allow two "wrong" direction channels in a row so that you don't have excessive loss in a row (in emergency mode) and so that you don't pay too much of an update rate penalty (normal mode).

Transmitter also only transitions to a mode when it reaches a channel that would be sent at that speed.

I don't really like this scheme compared to the "deadline" mechanism, but it is guaranteed to reacquire quick, provides feedback that fast-mode is working again, captures even if the user switch is in the wrong position, etc.  The downside is the loss of the fast frames while in the slow mode.

The upside of the "deadline" scheme is that it keeps sync really well, can be used for general rate negotiation (e.g. nothing says you have to have just 2 speeds), etc.  Downside is it requires telemetry.

csurf, 2018-02-07:
@mlyle 
ok, if you're confident & adamant about this strategy, we can give it a try. However, there's still a lot of work that needs to be done before we get to that point. I still need to figure out the data rate & packet structure stuff.

csurf, 2018-02-07:
@atomiclama 
Good to know, thanks for the info...
At least we know that there shouldn't be any major problems by running a lighter header configuration.

fiveangle, 2018-02-07:
One thing that would really make it a lot easier to figure out the optimal protocol would be captures of actual radio communication during various usage scenarios (even just racing miniquad and high-channel fixed-wing both with and w/o telem would likely be sufficient to cover most obvious real-world scenarios).  With that, it would be relatively simple to characterize the data patterns then plug 
 various model proposals discussed here over the top to see what the resulting behavior using the proposed spec would look like ("fast" ch latency, "slow" ch latency, sync reacquisition time, etc).

It appears OpenTX supports [s.bus mirror out the "PC" UART in the battery bay](https://github.com/opentx/opentx/wiki/Taranis-I-O-ports) so an openlogger hanging off of it should be all that's needed.  Does anyone think this would be of value ?  I think we can all make reasonable educated guesses but it's often surprising what you find where the rubber meets the road.  Or perhaps someone knows of captures like this that already exist ?

> I'm currently running with 2 byte header and 1 byte sync without problems

I haven't dug into the code to review but is there a reason why sync and header must be on byte boundaries?  256 values for sync bit seems like it might be a bit thin (pun intended).  Assuming 34458us packet interval and potentially 6s between frame reacquisition (worst case), that's about 175.  Of course 512 seems like more healthy breathing room but I agreed that 65536 is just gratuitous.  Packing/unpacking to steal a bit from header seems like an easy way to reduce over-air transport load unless there is a concern that packing/unpacking would require too much overhead for 8-bit AVR ?  Or does sync byte need to be on byte boundaries to maintain accurate timing or something like that ?

csurf, 2018-02-08:
@fiveangle 
not sure about the first part of your comment, someone else might want to chime in.

for the 2nd part, byte boundaries are how the RFM does it. Based on the info in the link below, you can't split it into bits, just full bytes. 

https://www.silabs.com/community/wireless/proprietary/knowledge-base.entry.html/2014/12/09/ezradiopro_and_sync-zc1N

sync word:
I'm not a pro on this stuff, but I understand the sync word to basically just be a delimiter to establish where packet header ends and the transition to data begins. A longer sync word gives better assurance against false packet detection.

also found this:
https://e2e.ti.com/support/rf__digital_radio/etc_rf/f/228/p/267578/937697#937697

From the looks of it, shaving a byte off the sync word might not be worth it if it means we're going to be dropping packets more frequently due to false positives. However, this would need to be tested as it might not even be a problem in our application.

FYI, you might want to take the packet interval timing stuff with a grain of salt at this point. All of the calculations need to be checked, tested, and verified (or updated). This all just stuff that the original author came up with, but there isn't much info in the comments as to how he came up with it.

mlyle, 2018-02-08:
> I'm not a pro on this stuff, but I understand the sync word to basically just be a delimiter to establish where packet header ends and the transition to data begins. A longer sync word gives better assurance against false packet detection.

@csurf, there is a preamble.  Basically, the receiver needs to lock onto the transmitted packet, and needs a balanced set of 1's and 0's for its PLL to find the exact center frequency to maximize signal to noise (the center frequency is the best thing to compare against for 1/0, because then noise is less likely to push the detected signal over the boundary).  After this long string of alternating 1's and 0's, there is a sync word, which indicates the beginning of real data-- so that large amounts of unqualified/false data are not sent to the microcontroller and so that things properly align with byte boundaries.

csurf, 2018-02-08:
@mlyle 
yeah, that's kinda how I understood it. The combination of preamble + sync word are used as a means to lock/settle the RX onto the signal frequency (preamble), and then verify that it's actually a real packet and not a false positive (sync word). If I got this wrong, please correct me :)

mlyle, 2018-02-08:
While I've not been over the code with a fine-toothed comb, the hop-algorithm and choice of constants are very well done.  There are lots and lots of things that you could do better / within more modern constraints, but when you're constrained by RFM22 style modules, relatively slow channel changes, and ordinary FSK... it's not bad.  Deviation/occupied bandwidth is chosen relatively well, data rates are sane, etc.

csurf, 2018-02-08:
my comment about the packet interval timing is specifically aimed at this:
https://github.com/openLRSng/openLRSng/blob/master/openLRSng/common.h#L68

I feel that we owe it to ourselves to at least test & verify that empirical formula. I'm not saying it's wrong, just that we need to figure out how it came about (test & verify).

csurf, 2018-02-08:
what would people think about lowering the RC channel resolution to 8-bits for e-mode? 1 byte per channel, 256 steps for the entire control range, so ~4usec per 'stick step.' 

I know this sounds terrible, but maybe it could be acceptable for emergency control? It would reduce the packet payload by one byte (4 bytes instead of 5), and, like I said, every byte counts for what we're trying to achieve... However, I realize this might not be worth it. thoughts?

fiveangle, 2018-02-08:
I'd be worried about the aux channels if someone configured one too close to transition points on the other end and the rounding from lower resolution "moves" the point triggering transition to a different mode.  It could be mitigated by jumping through hoops (keep last known "non-e" state then fit low rez scale so it hits that point) but yeah, seems like not worth the risk/effort.  Although 256 is still quite a lot of points 🤷‍♂️

mlyle, 2018-02-08:
If you actually send an aux channel as a switch, it's 4 values IIRC already.  Which for a 3 position switch effectively makes it "random" whether you get pos 2 or 3 in the middle.

mlyle, 2018-02-08:
Another approach to low resolution is to dither and average.  You can reflect big moves instantly, and for fine position / small moves improve precision with subsequent frames.

Just add a random number between -1/512 and 1/512 to each value sent.  On the receiver, if the value differs by more than 1 LSB take it as the new value; if not, run the current value through an IIR.

csurf, 2018-02-08:
this brings up the point of how to deal with switch channels in e-mode...

let's take a channel scheme like 4+4 (4 full channels, 4 switch channels). The switch channels currently only take up two bits, so 4 channels are stuffed into 1 byte. 

So, the TX uplink packet size for '4+4' would be 7 bytes:
- 1 byte: header/meta
- 5 bytes: 4 full RC channels (10-bit resolution)
- 1 byte: 4 switch channels (2-bit resolution)

If we drop into e-mode and use the scheme of sending only 4CH/5 bytes per packet, with alternating packets for each channel group, then we'd need to upscale the 4 switch channels from 2-bits each/1-byte total, to 10-bits each/5-bytes total. Programatically, this wouldn't be a problem, but it does seem like a waste of bandwidth having to upscale & send channel data that's already been scaled down & compacted. Still, I don't see any other practical way to deal with this, other than to keep things consistent and have a steady flow of 5-byte payload packets... thoughts?

csurf, 2018-02-08:
@mlyle 
that dithering algo sounds interesting. Can you elaborate a bit on it? Do you think it can be implemented in this setup without too much overhead/overload for the AVR's?

fiveangle, 2018-02-08:
I completely forgot about dithering !  You're pulling out the dusty synapses @mlyle :)

@csurf - since the sender has access to the higher resolution input, and has knowledge of the receiver algorithm (for example, running average of last 5 points received), TX can craft the values sent so their average at the other end more closely targets the actual input data.  It's an old and valuable trick that I think has been lost the last 20 yrs or so due to most bandwidth models dramatically improving.  In fact, with the increased power of something like STM32, similar could be done with standard modes, resulting in a 2x-4x of perceived resolution, at the expense of minor amount of indeterminism, which doesn't really matter when there's a human on the other end of the sticks.

geofrancis, 2018-02-08:
the problem with lowering channel resolution is that things like mode switches can be triggered inadvertantly like fiveangle said, thats why i would recommend lowering the update speed rather than the resolution. something like 1234+5, 1234+6, 1234+7 etc

mlyle, 2018-02-08:
> the problem with lowering channel resolution is that things like mode switches can be triggered inadvertantly 

Remember we're usually measuring PPM in an asynchronous noisy environment.  If the mode switch is right on the threshold, then there's a real risk just ordinary dither is going to make it wrong.  The real answer is "don't do that".

That is, we're already +/- 1 count for asynchronicity, plus figure +/- an additional 1-2 for noise.  Vs. the 3 counts of uncertainty when chopping resolution.

@csurf, re: dither.  You just add a random (or ramping) number to what you send before shifting right to get what to send.  if the 10 bit value is exactly 113, you'd add between 0 and 3 to it each time, yielding numbers in the range of 113..116.  When shifted right, they yield:

```
113: 28
114: 28
115: 28
116: 29
```

So 75% of the time, it's 28, 25% of the time, it's 29.

The other side demodulating in turn receives the first frame, say it's 28.  That's different from the old value, so it sets the output to 28 << 2 = 112.  Next it receives the next frame, which is 28  It's close enough that we use averaging.   We add 112 to the output and divide by two, and still end up with 112.  Next time we get a 29, add 116 and divide by 2, and get 114.  Next time we get a 28, add 112 and divide by 2, and get 113.

csurf, 2018-02-09:
> The problem with lowering channel resolution is that things like mode switches can be triggered inadvertantly like fiveangle said, thats why i would recommend lowering the update speed rather than the resolution. packets with something like channel 1234+5, 1234+6, 1234+7 etc so it would only use the same bandwidth as 5 channels for 16.

ok, the more I think of it, I kinda prefer this strategy.
It adds another byte to the payload, but avoids the complexity of having to deal with splitting up the packets for different channel groups.

The only issue is that it puts us back to a 7-byte payload, which is the same as the current payload size for the '4+4' channel scheme for 4800bps. So, there's no real bandwidth savings or data rate improvement, but at least we can stuff in every single channel at the lowest data rate.

With telemetry enabled, the refresh rate will will be pretty low, and the sticks will start to feel really slow & rough. Perhaps we could look at shrinking the telemetry packet size from 9 bytes to like 5 bytes?

geofrancis, 2018-02-09:
i suppose you dont need upload at all in that mode and all you would need is gps and possibly voltage.

csurf, 2018-02-09:
@geofrancis 
yeah, the down-link packet is what slows it down though...

With telemetry, the packet interval has to be increased on both ends (TX & RX) in order to allow some extra time for the RX to be able to send down its telemetry packet & also allow time for the TX to receive and process it. 

With telemetry disabled, you don't have to worry about a down-link packet. The TX is free to just blast out packets as fast as possible (well, at a rate where the RX can keep up at least). 

If we shrink the telemetry packet a bit, we reduce the time that needs to be added to the overall packet interval, and we get a slightly better refresh rate.

The up-link telemetry packets (if there are any) are simply mixed into the stream of outgoing packets from the TX, so it really doesn't affect the packet timing, other than perhaps slowing down the RC refresh rate a bit when serial data packets are sent instead of RC packets.

csurf, 2018-02-09:
I don't personally use telemetry, so I wouldn't be the one to judge this, but would it be tolerable to have a telemetry rate that's sending down data in 4-byte or 5-byte chunks?

geofrancis, 2018-02-09:
I was thinking that this would be useful for finding a lost model if it still transmitted back gps coordinates,

geofrancis, 2018-02-09:
i remember i had seen this years ago, this should give you an idea on the level of range increase by dropping the data rates.

csurf, 2018-02-09:
@geofrancis 
cool info.

If we manage to make e-mode work, I might just consider flying in it all the time for longer-range flights hehe. Perhaps it'll end up being known more as an LR (long-range) mode instead of soley as a 'last-resort / oh sh*t!' mode...

Here's the packet structure I'm considering:
- 7-byte total payload
- 1 byte header / meta / & partial 5th channel data (2bits)
  - [1 bit: TX sync]
  - [1 bit: RX sync]
  - [4 bits: AUX channel number (0 - 10 => ch5 - ch16)]
  - [2 bits: AUX channel LSB's]
- 5 bytes RC data for RC channels 1-4, 10 bits per channel
- 1 byte for AUX channel MSB's
- data up-link from TX will be disabled
- AUX channel will be sent in alternating transmissions, per @geofrancis's idea
  - 1st transmission: 1,2,3,4,5
  - 2nd transmission: 1,2,3,4,6
  - 3rd transmssion: 1,2,3,4,7
  - etc., etc., etc., ...