openLRSng/openLRSng #191: Discussion: Signal Noise Ratio

atomiclama, 2018-02-07:


csurf, 2018-02-07:
@atomiclama 
I actually set something like this up a while back. I used it for debugging & checking which hop channels were 'bad' / noisy when flying in a certain area. It would simply output a bunch of link stats in CSV format to the serial port on the RX, where I'd have an openLog connected to store the output to SD card. It would simply log when packets were received or dropped, and when the RX went into failsafe. It also did 'SNR' by sampling RSSI during the 'quiet time' period as you stated. 

The way I used it was to analyze the log file after a flight and see what hop channels were dropping packets & why. You could usually more/less detect parts of the frequency band that were being hit with local interference. I would then reconfigure my hop channels and move them around to avoid those parts of the frequency band that were causing problems. 

One idea I had was to keep this information stored in memory and keep something like a simple running average of the link state, and then log the stats to flash memory so that they can be analyzed later via configurator. Something that indicates which hop channels / frequencies are performing poorly and which are performing best might be useful.

If&when we move to new hardware, we would certainly have more options in terms of logging. For now, our options are pretty limited.

What would also be awesome would be to have GPS input into the RX as a means of logging position vs. link state.  That way, you could then map out areas / flight zones that were causing issues. Maybe in the future we can do something like this as well...

csurf, 2018-02-07:
... In terms of your original question... Are you referring to sending the RX SNR over telemetry? If so, we might be able to do that. It appears that there is one available byte left in the telemetry packet that isn't being used, so we could probably use it to store SNR, but it would be an average SNR over the whole band / all of the hop channels, so I'm not sure how useful it would be. For example, if you have 6 hop channels, with 5 having really good SNR and one hop channel with really bad SNR, the averaged SNR value would be lower, and it wouldn't really be a true representation of the link state.

csurf, 2018-02-08:
@atomiclama 
looking at the telemetry packet structure here...
https://github.com/openLRSng/openLRSng/blob/master/openLRSng/RX.h#L84

you can see that 2nd byte in downlink packet is using 2 bytes for AFC values, which I don't even know why we're sending down (doesn't seem like very useful data other than for testing). 

Perhaps we could consider doing away with the AFC register values, and use those two bytes for something else? Maybe 1 byte for 'average' SNR and another for 'minimum/worst' SNR? thoughts?

atomiclama, 2018-02-08:
@csurf Some good ideas there, I like the one about logging position and link state, made me laugh as I have just removed the EE from my FC so I hack on a RFM22 module.

The original way I was thinking was to provide alternatives to RSSI and give the user the ability to select which one they want to use. e.g. pass through to an OSD whilst FPV or back over telemetry.

I'm not experienced enough to gauge which metric would be best, RSSI, link quality or SNR.

I'll take a look at your code thanks.

csurf, 2018-02-08:
When using pure RSSI, you'll get an idea of how your signal strength is gradually dropping, while link quality will basically work as a 'packet-loss indicator'. However, the two concepts can be very different/separate in some cases.

From my experience, the most practical thing to monitor is hop channel performance, as that's what will normally get you when flying at range. 

What usually happens is that the plane will fly over an area with localized noise spikes from a source that's near the plane (nearby mountain-top UHF repeaters in my case). This will knock out one (or a few) of your hop channels, so, you'll start dropping packets on those bad hops. Your 'combined RSSI (link quality + RSSI)' will immediately drop to 50%, which will start to freak you out. 

However, the rest of your hop channels might still be 'clean' and showing strong RSSI (and fairly strong SNR). This is where having more hop channels can help you, as you can continue to fly with the majority of your hop channels intact (even though you're consistently losing packets on the bad hop channels).

This is why my strategy is to try to log my link stats and then go back and see which hop channels were giving my problems. I'll then move them away from the problem frequencies, go out & fly, check the logs, adjust as necessary, rinse, lather, repeat...

hope this makes sense.

fiveangle, 2018-02-09:
> When using pure RSSI, you'll get an idea of how your signal strength is gradually dropping, while link quality will basically work as a 'packet-loss indicator'. However, the two concepts can be very different/separate in some cases.

I think true radio "link quality" offers no more to users that "RSSI".  For any users, ideally, they want a 0-100 range dimensionless value reported to them that drops from 100 (or 99) to 0 linearly as their ability to control their craft is diminished with control completely lost at "0" (failsafe timer engaged or expired).  This is the ideal goal that I think should drive any change from the antiquated reporting of basic "RSSI".

It could be as simple as staged range where 0-10 is set to "10" if for example "e-mode" is not engaged, 11-20 is set to "20" if low-power packets in alternating power mode are not detected as "lost", and 31-99 generally following RSSI, SNR, or any combo that we collectively feel is representative, with any weightings within the range that we feel should degrade the reported CLQ accordingly (percentage of hop channels deemed unusable, as @csurf is exploring for example, etc).

csurf, 2018-02-09:
@fiveangle 
I agree that we need a better metric for evaluating the link. Unfortunately, we're kinda limited by bandwidth and 8-bit math. The stacking algo sounds cool, but I don't see we could easily implement it with the AVR. However, if you aready have an idea of how to set it up in code, please share...

The way link quality is currently done in code is based on using bit-wise math & creating somewhat of a running average via a '2-byte fifo', setting bits in a 16-bit variable on&off whenever a packet is received or lost. Since it has a fixed scale, I don't think it effectively relates to how good/bad the link really is because it doesn't take into account the total number of hops versus the number of hops that are being hit by noise. However, it terms of bitwise kung-fu, it's pretty cool.

RSSI is just an 8-bit value that comes directly from the RFM & that's already in a 0-255 scale, so nothing more needs to be done, other than to smooth it out with a running average. Oh, and it also gets scaled for PWM/PPM output. 

SNR would require an extra RSSI read from the RFM during the 'quiet time' in between packets, and then also smoothed out to get an average.

The 'combined RSSI' value stacks RSSI on top of link quality and stuff it into one byte; ~128 bits for each value. So, if any packets are lost, RSSI immediately goes "out the window" and the only thing that the value reflects is packet loss. The logic here is that "if you're already losing packets for whatever reason, then you've got bigger problems then worrying about gradual RSSI degradation." RSSI is no longer deemed relevant if you're about to lose control due to packet loss (which may or may not be true as I explained above).

If we had tons of bandwidth & resources to play with, I'd like to use a 32-bit 'hop channel health' variable that uses bit flags to indicate hop channels, with each bit representing the state of a hop channel (32 hop channels max, bit value of 1 =>'good', 0=> 'bad'). This data could be sent to an OSD or parsed via serial telemetry, and you could track how each hop channel performed over the duration of a flight. This is just a dream at this point, but I think something like this could be pretty useful.

Anyway, back to the problem at hand...
more ideas are definitely welcome, as long as we can make them more/less work within our current 8-bit / limited bandwidth infrastructure.
this is what we've currently got:
- For telemetry, we have ~3 bytes to play with if we get rid of the 2-byte AFC fields
- For PPM injection, we currently have the three options of setting up the output:
  - RSSI on 'x' channel
  - linkQ on 'x' channel
  - combined RSSI+linkQ on 'x' channel
  - RSSI on 'x' channel, linkQ on 'x+1' channel
for PWM output, we've currently only got RSSI setup for direct output (for directly driving a servo or something)
- LBEEP for packet loss (beep every time we lose a packet), which basically turns hardware PWM on & off whenever a packet is lost & received respectively to produce an audible tone

artemen, 2018-02-15:
Hey guys, can we change hop channels in real time (based on a well thought-through protocol) if certain number of  them get hit?

csurf, 2018-02-15:
@artemen 
This has always been on my personal wish list; dynamic / adaptive frequency hopping algorithm that adjusts the hop channels in flight to frequencies with the lowest RSSI. I haven't been brave or clever enough yet to come up with an algo that will achieve this though hehe.

In basic terms, it would require telemetry and a scanning strategy on the RX-side that checks RSSI on all available channels to determine which frequencies are "clear" and have the lowest noise levels. Then, the RX would have to signal to the TX that it wishes to change a specific hop channel from 'x' to 'y'. The TX would need to ACK the suggested change and the two start using the new frequency.  It would have to do this for all channels; constantly monitoring RSSI on both the assigned and 'free'/available hop channels, and then make decisions on the fly. It might get pretty complicated very fast, but maybe it's possible with some cleverness & creativity.

I guess we should open up a separate issue for this feature request...