openLRSng/openLRSng #192: PWM control of RFM23bp power amplifier

geofrancis, 2018-02-09:
Many years ago I built a module that could hotswitch between 3.3v and 5v going to the PA allowing me some control over the power output of the module, it basically involved a switch going between the stock 3.3v pad and a separate 6v regulator with some diodes and large capacitor to smooth out the transition. The first time I tried switching the module without the capacitor it would just reset but adding the capacitor slowed the transition enough that it would switch reliably.

now that got me thinking, there are plenty of spare pins on the Tx module that could be used to run a basic pwm voltage controller using a regulator, transistor, capacitor and some resistors so it could modulate its own output directly by changing the voltage to the PA without my crude switching device or reducing the power into the amplifier like the software currently does.

csurf, 2018-02-09:
This is more of a hardware design issue, but we could definitely add code to support it. Someone would need to breadboard it and test it out though.

There might be some pretty cool things that can be done with this.

One idea might be to gradually ramp up/down of the power output depending on what the RX's RSSI value (received over telemetry). i.e., we could try to keep the RSSI at a decent level, and adjust the transmit power up & down as necessary just enough to keep the link solid while also keeping transmit power at a minimum.

Another cool thing might be to monitor the RF module's temperature via the built-in temp sensor, and then kill the power if it starts getting too hot (we should already be monitoring it for emergency/alert purposes, but it never got done).

csurf, 2018-02-12:
@geofrancis 
would you be willing & able to mock this up hardware-wise?

goahead0, 2018-03-08:
According to hoperfs datasheet the rfm23bps PA is theoretically capable of power levels of +1 to +30 dBm. Practically there seem to be severe limitations on finely adjusting the tx power. There are just three bits in reg 6d which deals with the PA. 

The data sheet is unclear, they talk about setting the power level in 3dB steps, but the table provided only lists +28/29/30dBm. It looks like the reduced energy efficiency prompted hoperf to restrict the usable power range.

A first step requiring no hardware modifications would be to set the PA power in the limited range possible according to the temperature value delivered by the internal sensor.

The remaining 5 bits of register 6d are unexplained in the data sheet and it looks like they are also not used in openLRSng. The module already has the hardware to control the power level. Before adding hw-pwm to control the module supply voltage externally, it might be good to make sure it really is necessary.

I'll ask hoperf if they can clarify the issue if no one already has done so.

atomiclama, 2018-03-08:
I have one of these modules on the way and was planning on doing some current measurements at various register settings and VCC levels. I'll post up any measurements I get.

goahead0, 2018-03-09:
According to hoperf's technical support the rfm23bp is based on a si4431 + external PA. The power settings in register 6d allow to set the power for the si4431 in 3dBm steps from -8dBm to +13dBm. The PA adds roughly 20dBm to this (depending on supply voltage). So the output power can roughly be set from a minimum of about 12dBm up to 33dBm for a supply voltage of 5V in an ideal situation.

The graphics below illustrates the PA power and current consumption for a voltage range of 3.6V - 9V depending on the input power:

[image]

artemen, 2018-03-19:
I've recently decided to do the [Narpat's](https://www.rcgroups.com/forums/showpost.php?p=33899703&postcount=3691) mod and power PA separately from the Si4331, as expected at 9v we do get around 2w of output from the old rfm23bp, also if done properly heat is not excessive and PA runs for more than 24hrs without frying. this was demonstrated by Narpat a year ago. What's not expected is that even with about than 33dbm output, packet loss rate didn't change much, this made me think that either RX or TX or both do not get locked properly onto the exact frequency. If so, would this be hardware fault or code fault?

mikerc2017, 2018-03-19:
@artemen 

after you mod 9v for PA , range better ? i try mod 6v for Modul on TX , but seem result not change

artemen, 2018-03-19:
Range is marginally better, i would say i've notced ine spot where I woild loose rc link almost 2km away at about 2-3 meters anove ground it was behind trees and a small building, i have solid control there now, though loose about half the packets.  Haven't tested in long range yet.

mikerc2017, 2018-03-19:
seem still not yet perfect, if RX have a LNA  ,maybe range will better instead mod Highvolt for modul ?

goahead0, 2018-03-23:
The module already contains an LNA + programmable amplifier with automatic gain control for the rx. Lora's higher sensitivity comes at the price of a much reduced data rate. I would concentrate on good antennas. 

2km away, 2-3m above ground and behind multiple obstacles you can't seriously expect a much better result.

artemen, 2018-03-23:
@goahead0 
 how about 4km and 10 meters above ground on a multie?  ;)

goahead0, 2018-03-23:
Too many unknown factors involved for me to compare and not to speculate.

artemen, 2018-03-23:
@goahead0 
can't agree more with you, so many videos of DL flying 7-8 km only meters above the ground, but never mention that ground station is elevated or if they are flying uphill. 

 4170meters out 12meters above takeoff point about 7-8 meters above ground on a flat as pancake terrain with some trees but mostly grass fields in a suburban area. nagoya on dtfuhf delux TXv1 and diy diploes on a diy diversity running broversity code.

mikerc2017, 2018-03-24:
distance  2m  above ground. i  only can move to 500m .All use dipole anten. modul 1w .
@artemen   i real want to see your video.

artemen, 2018-03-24:
@mikerc2017 
a year ago new place not exactly two meters but low enough to be behind multiple trees ans a few houses: 
https://www.youtube.com/watch?v=9lWgp0LnbVs  

one of the first flights of the season, the end of this February  
https://youtu.be/iHCLbrwrBTI?t=380 at this frame abpout 9-10meters above takeoff point,  3.9-4km out. watch from the begining to get the scale, video ends at 4.4km

mikerc2017, 2018-03-25:
Thanks artemen 

and everyone : i find a modul 500mw with support  **LNA**  , i will order and test  and report