openLRSng/openLRSng #194: Setup binary build job with travisCI?

csurf, 2018-02-12:
@fiveangle 
can we set something up with travisCI to automtically build & upload all of the binary images to the repo? I'm thinking of something like a 'nightly builds' directory that we can occasionally dump the binary images into (after a PR is merged). Would something like this be possible to setup?

mikeller, 2018-02-12:
I don't think the repository is the right place to push builds, or binaries in general. But we could look into pushing them into something like a Google drive share or similar. But in general I don't think travis CI supports extraction of build artefacts, we'd have to look into using Jenkins or similar.

csurf, 2018-02-12:
@mikeller 
ok, whatever you guys think is the best way to deal with this.
How are builds managed for betaflight?

mikeller, 2018-02-12:
Betaflight uses Jenkins for artefact generation, running on hardware / connectivity provided by members. I've also started experimenting with a free open source account for TeamCity. This could be another option for artefact generation for openLRSng.

fiveangle, 2018-02-14:
I can have Travis upload resulting binaries whatever filesharing service we want (like transfer.sh) but the URL would be burried in the Travis build log.   I agree that uploading binaries to the repo probably isn't the best option.  What would be really nice is upload to some filesharing service, then auto-comment to the commit that triggered the build with the URL to that commits built binaries, but that would probably take some funky hoops as I'm not sure how the comment mechanism coulid be done.

Let me know if you want me to pursue.

mikeller, 2018-02-14:
@fiveangle: If travis allows this as part of their free offering, that would be a great option. I had a quick look at transfer.sh, and they seem to be having a 14 day expiry on the uploaded files, which is probably not what we want, because it would mean that there won't be any installable artefacts available if nothing has been merged into master for more than two weeks.
Are there options to use something like Google Drive, or dropbox, or similar, with no automatic expiry of files?

fiveangle, 2018-02-15:
Google drive looks to have some Oauth hoops to jump through but not impossible. I’ll put it on my todo list.