openLRSng/openLRSng #195: Remove 8MHz support?

csurf, 2018-02-12:
I'm a bit confused by the 8MHz support in the code. Both the makefile and code have options for supporting an 8MHz XTAL. However, I don't even think that hardware with 8MHz oscillators exist "in the wild." 

In fact, it looks like only "RX type 8" is setup for the option an 8MHz build. However, hardware.h even puts out a hardware warning for hardware type 8 if the CPU clock isn't set to 16MHz, which contradicts the idea of it being meant to also run 8MHz.

Can we simply clean up the makefile & code and remove the 8MHz support?

geofrancis, 2018-02-13:
I think it was more of an experiment to run with the internal clock so it could be made with a 328 pu

csurf, 2018-02-13:
ok, so then it should be safe to remove?

geofrancis, 2018-02-13:
yes as far as i know no hardware ever came of it.

csurf, 2018-02-13:
ok. thanks for the info.
removing & closing...

closed 2018-02-13 by csurf