openLRSng/openLRSng #197: Atmega32u4 Rx possible?

ameeuw, 2018-02-23:
Hey everybody - really nice to see some activity within this repo again. I am getting my old OrangeRx gear up again and now tried to also build my own light-weight receiver. So far I have tried to build one with a 3.3v 8MHz arduino mini pro but was rather unsuccessful due to the timing issues with the 8MHz running chip...

Now I stumbled across the "beetle" board - a quite small and nice square board with a atmega32u4, broken out SPI and two onboard LEDs. The version I linked below also features a 3.3v regulator. I guess one could even saw off the USB-connector if in need for a really tiny receiver.

[AliExpress Link](https://www.aliexpress.com/item/Beetle-Virtual-Keyboard-BadUSB-Pro-Micro-ATMEGA32U4-Module-Mini-Development-Expansion-Board-For-Arduino-Leonardo-R3/32847715254.html)

My question about this is: How big is the effort to enable receiver code for the 32u4? The DTFUHF/HawkEye deluxe TX (TX_BOARD_TYPE 6) also uses the 32u4 - is this just a matter of pin redefinition and adjusting the switching routines?

mikeller, 2018-02-24:
@ameeuw: That's a neat board. I think it shouldn't be too hard, and giving it a go can't hurt. Look into `openLRSng/hardware.h` to find the pin definitions.

ameeuw, 2018-02-25:
I looked into it a little and made a testing board using an Arduino Pro Micro. So far I have been getting the TX-code to compile and work an RF22b.
I tried to just copy over the pin definitions like I did for my own diy receiver but it seems like none of the switching and timing logic of the receiver in `openLRSng/RX.h` can be directly taken over to the leonardo.
Is somebody firm with atmega32u4 hardware and its functions? I fear I do not have the knowledge to go over the RX code and adapt it.

ameeuw, 2018-02-28:
I actually stumbled across a maybe even easier solution...
[OpenLog (github link)](https://github.com/sparkfun/OpenLog) is an open source datalogger and is very cheaply available:

[AliExpress Link](https://www.aliexpress.com/item/1pcs-Openlog-Serial-Data-Logger-Open-Source-Data-Recorder-ATmega328-Support-Micro-SD-for-arduino/32838430172.html)

It is a board with an ATmega328 that runs on 3.3v with 16MHz - has SPI with CS and another pin (for nIRQ) broken out to the SD-card-holder. I guess if you desolder the sdcard this would be rather suitable.